raspberrypi/linux #4209: No packets sending through 6lowpan

nvayalil, 2021-03-12:
**Describe the bug**
Connected two Raspberry Pi W using 6lowpan and ping is working find with global address at both devices. However, it seems that no programs can send data through the interface, I tried python scrips and mosquitto server etc. 

**To reproduce**
Client:
# Enable bluetooth_6lowpan module
modprobe bluetooth_6lowpan
echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
# Start advertising
hciconfig hci0 leadv

Server:
modprobe bluetooth_6lowpan
echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
hcitool lescan
# Connect to the client with bluetooth mac address discovered
echo "connect <mac addr> 1" > /sys/kernel/debug/bluetooth/6lowpan_control
#Ping with link local address
ping6 -I bt0 <link local address of the client>
# Add gloabal address (in client)
ip addr add fd00::2/64 dev bt0
# Add global address (in server)
ip addr add fd00::1/64 dev bt0

# Ping the global address, server to client, works fine
ping6 fd00::2
# Ping the global address, client to server, works fine
ping6 fd00::1

# Try to connect client to server using a program, such as python scripts given below, No data is sending through the 6lowpan, observer with a tshark, or similar.

#tcp client program
#!/usr/bin/env python3

import socket
import struct

SERVER_ADDR = 'fd00::1'
SERVER_PORT = 9001
sequence_number = 1
sock = socket.socket(socket.AF_INET6, socket.SOCK_STREAM)
print('send connect request')
sock.connect((SERVER_ADDR, SERVER_PORT))

while sequence_number < 101 :
    data = struct.pack("!I", sequence_number)
    data +=  'Ping'
    sock.sendto(data, (SERVER_ADDR, SERVER_PORT, 0, 0))
    recv_data = sock.recvfrom(8)
    print(recv_data)
    sequence_number+= 1

sock.close()
del sock
# tcp server

#!/usr/bin/env python3

import socket

SERVER_ADDR = ''  
SERVER_PORT = 9001 

with socket.socket(socket.AF_INET6, socket.SOCK_STREAM) as s:
    server_addr = (SERVER_ADDR, SERVER_PORT, 0, 0)
    s.bind(server_addr)
    s.listen(1)
    conn, addr = s.accept()
    with conn:
        while True:
            data = conn.recv(128)
            if not data:
                print('break')
                break
            conn.sendall(data)
s.close()
del s

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 
PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)? 
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2
* Which firmware version (`vcgencmd version`)?
Feb 25 2021 12:12:09
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux

pelwell, 2021-03-12:
Running tshark on the _sender_ do you see any packets?

nvayalil, 2021-03-12:
Yes, but not leaving the sender, and sender tries to re-transmit again and again.
 
   14 347.454058556      fd00::2 ? fd00::1      TCP 96 40052 ? 9001 [SYN] Seq=0 Win=64660 Len=0 MSS=1220 SACK_PERM=1 TSval=1505630761 TSecr=0 WS=64
   15 348.400000887      fd00::2 ? fd00::1      TCP 96 [TCP Retransmission] 40052 ? 9001 [SYN] Seq=0 Win=64660 Len=0 MSS=1220 SACK_PERM=1 TSval=1505631804 TSecr=0 WS=64
   16 350.480060418      fd00::2 ? fd00::1      TCP 96 [TCP Retransmission] 40052 ? 9001 [SYN] Seq=0 Win=64660 Len=0 MSS=1220 SACK_PERM=1 TSval=1505633884 TSecr=0 WS=64

PAguirre-genrod, 2021-05-14:
Hi, this is the first time I'm contributing so please let me know if I should have open another issue or if this is not the way of adding info, but I´m facing the same error and I can add a few things.
I've been trying to establish an MQTT connection using BLE 6LoWPAN with a Nordic device as a node and a Raspberry Pi as master. I made [this post](https://devzone.nordicsemi.com/f/nordic-q-a/73698/making-mqtt-publisher-example-work-on-raspberry-pi-4) explaining what my first problem was and why I ended up having this problem.

In my application the end node inits a TCP connection and the Raspberry Pi sends the corresponding packets for it to be able to open a TCP socket, I use a sniffer so I can corroborate that those packets are indeed going to the air. The problem begins when the MQTT broker (mosquitto) tries to send a CONNACK packet, when that happens you can see the TCP packets corresponding to that CONNACK packet and the retransmissions when running tshark BUT not when capturing from btmon, for what I suppose there's a problem delivering TCP packets to BlueZ stack.
Pinging also works fine.

I have been using this application with Jessie and Stretch, this problem appears when trying to update to a Buster kernel.

In the post I linked at first you can get full captures of what I'm saying, but I'll leave an image:

[image]

Hope this can help to detect what the problem is and if there's anything else I can help with I'll be glad to do it.

### System

- Which model of Raspberry Pi?

  Raspberry Pi 3B and Raspberry Pi 4B

- Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2

Which firmware version (`vcgencmd version`)?
Feb 25 2021 12:10:40
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux