raspberrypi/linux #4210: usb 1-1.3: device descriptor read/64, error -32 (at boot with rpi4 arm64 OS)

mgrouch, 2021-03-12:
Started getting after today's update

[    2.352506] usb 1-1.2: Manufacturer: SEM
[    2.361967] input: SEM USB Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:1A2C:2124.0001/input/input0
[    2.421182] hid-generic 0003:1A2C:2124.0001: input,hidraw0: USB HID v1.10 Keyboard [SEM USB Keyboard] on usb-0000:01:00.0-1.2/input0
[    2.427351] input: SEM USB Keyboard Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:1A2C:2124.0002/input/input1
[    2.484646] input: SEM USB Keyboard System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:1A2C:2124.0002/input/input2
[    2.484967] hid-generic 0003:1A2C:2124.0002: input,hidraw1: USB HID v1.10 Device [SEM USB Keyboard] on usb-0000:01:00.0-1.2/input1
[    2.568269] usb 1-1.3: new low-speed USB device number 4 using xhci_hcd
[    2.604270] sd 0:0:0:0: [sda] 976773168 512-byte logical blocks: (500 GB/466 GiB)
[    2.604291] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    2.604563] sd 0:0:0:0: [sda] Write Protect is off
[    2.604580] sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
[    2.605073] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    2.605855] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[    2.621270]  sda: sda1 sda2
[    2.624586] sd 0:0:0:0: [sda] Attached SCSI disk
[    2.864495] usb 1-1.3: device descriptor read/64, error -32

My full dmesg

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.17-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000fc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 1949696
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000]   DMA32 zone: 12288 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91224 r8192 d31656 u131072
[    0.000000] pcpu-alloc: s91224 r8192 d31656 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1918976
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1024x600M@60 smsc95xx.macaddr=DC:A6:32:D5:46:6D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=bf32b703-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=1 splash quiet logo.nologo vt.global_cursor_default=1 plymouth.ignore-serial-consoles console=tty3 rd.systemd.show_status=false
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000000bc00000-0x000000000fc00000] (64MB)
[    0.000000] Memory: 7289456K/7798784K available (10944K kernel code, 1918K rwdata, 3912K rodata, 3648K init, 1240K bss, 247184K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36199 entries in 142 pages
[    0.000000] ftrace: allocated 142 pages with 4 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[... 401 lines cut ...]
[   44.036602] Bluetooth: HCI socket layer initialized
[   44.036614] Bluetooth: L2CAP socket layer initialized
[   44.036635] Bluetooth: SCO socket layer initialized
[   44.231415] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   45.753719] Bluetooth: HCI UART driver ver 2.3
[   45.753739] Bluetooth: HCI UART protocol H4 registered
[   45.753842] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   45.754053] Bluetooth: HCI UART protocol Broadcom registered
[   46.011944] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   46.011961] Bluetooth: BNEP filters: protocol multicast
[   46.011982] Bluetooth: BNEP socket layer initialized
[   49.488924] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   59.463936] broken atomic modeset userspace detected, disabling atomic
[   63.216137] Bluetooth: RFCOMM TTY layer initialized
[   63.216161] Bluetooth: RFCOMM socket layer initialized
[   63.216184] Bluetooth: RFCOMM ver 1.11
[   63.472784] input: Simulated Right Button as /devices/virtual/input/input9
[   65.397733] fuse: init (API version 7.32)
[   66.599978] rfkill: input handler disabled
[   83.710047] w1_master_driver w1_bus_master1: Attaching one wire slave 00.800000000000 crc 8c
[   83.714897] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
[  142.346777] w1_master_driver w1_bus_master1: Attaching one wire slave 00.400000000000 crc 46
[  142.351325] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  189.189539] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c00000000000 crc ca
[  189.193991] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.

pelwell, 2021-03-12:
You appear to have a large number of peripherals attached to your Pi - please list them, including all power supplies.

mgrouch, 2021-03-13:
I have keyboard, mouse, touchscreen in deskpi pro case with SSD drive. 
power supply is what came with deskpi pro.

mgrouch, 2021-03-13:
Almost same error happens on arm32 (only error code is different -71)

mgrouch, 2021-03-13:
5.10.11-v8+ 
Didn’t have that issue