raspberrypi/linux #4214: Bluetooth breaks built in WiFi on Pi 4

AleXSR700, 2021-03-14:
Hello everyone,

I am facing an issue with BT devices. I am using my Pi with a true BT mouse and a true BT keyboard.

The reason I do not want the pseudo BT/RF devices with dongles is the poor compatibility with other devices and the lack of usability with FireTV stick, Android phones etc.
So I specifically chose BT mouse and a BT keyboard (Logitech K480).

I can use Wifi, BT mouse and BT keyboard on any PC I own without issues. But on the RPi the keyboard keeps repeating keyyyyyyys.
I found other, older threads for RPi 3s, where people recommend external BT dongles or use of ethernet.
Neither is a viable option for me as I need the USB ports and am already using a wifi dongle for AP mode.

Using wifi + BT should be a pretty basic need after all. Is there a solution to this issue that does not require switching of hardware?
I don't mind if keyboard and mouse only worked one at a time as I do not type and move the mouse simultaneously. But I do need both the work without disconnecting/reconnecting them.
Maybe some kind of key input delay setting or something like that?

Any help would be really appreciated

pelwell, 2021-03-14:
I regularly use BT mouse and keyboard and WiFi simultaneously, but results will vary based on the channel the AP is using and what else is using the spectrum. HDMI (especially 2560x1440) and USB3 are also known to interfere in that bad.

I recommend using a channel in the 5GHz band or, failing that, one of the higher 2.4GHz channels - 11, for example. Also make sure you are running an updated image because improvements have been made to coexistence.

AleXSR700, 2021-03-15:
Well, that is unfortunately not an option.
Since this device is my travel media center, I cannot tell hotels etc. whoch channel to use or to activate 5 Ghz ;-)

Since, as I said, this seems to be a RasPi problem, surely there must be a solution there. If this were a physics problem, then my other devices would show the same behaviour.

max17048, 2021-03-15:
Maybe this references to #3727

Does this happens every time you connect to a 2.4GHz router or only on special routers? Which router and OS on the router you use private? My suspicions is, that the (my) router starts CTS after the BT-Broadcom-MAC appears once and this blocks the 2.4GHz-band.

AleXSR700, 2021-03-15:
I am using a Huawei E5576 4G router right now. As I am abroad, I have not tested on my home router. I did not have the BT keyboard at home yet.
But I can try a mobile hotspot set to 2.4 Ghz.

But the other issue was also never followed up on? And not resolved?

AleXSR700, 2021-03-16:
Tested with different router/hotspot. Same problem.
Also switched to higher channels. No change.

max17048, 2021-03-16:
Also here. On 4B, 3B, Zero W.

Switched on BT & 2.4GHz, the Raspi starts CTS-Flooding and block the WLAN-channel. Switch off BT, the CTS-Frames disappears. So the CTS-Frames come from Raspi, not from router.

W/o any further information, this remains mysterious.