raspberrypi/linux #4217: Reopen the issue until it is fixed upstream. Closing it makes it seem fixed. > > > This issue probably pertains to VL805 firmware or the Raspberry kernel and both of those are the vendor's responsibility to fix.

kashaiahyah85, 2021-03-15:
> > > This issue probably pertains to VL805 firmware or the Raspberry kernel and both of those are the vendor's responsibility to fix.
> > 
> > 
> > We have no access to the source of the VL805 firmware, so we are forced to rely on the manufacturers to fix any problems. Much of the code in the kernel comes from upstream as well, so again, whilst we will do our best to find issues, it's very difficult when it comes from somewhere else and you have no experience in that area.
> 
> I understand your position, but I still believe that we shouldn't be of the mentality "it's not our problem, case closed". Especially so when it's a non-issue on other architectures (x86_64) and on USB2.0 on the affected platform (Raspberry). One of the main reasons behind Raspberry Pi 4's attractiveness is the USB3.0 capabilities (so we're finally spared from the SD card's poor I/O and durability), sadly everything but that works flawlessly. At this point I would also like to point out that JMicron's JMS578 and JMS579 chipsets are very popular choices for the majority of USB-to-SATA/M.2 adapters and that is going to cause **a lot** of headaches. At the very least, if the team itself is not able to provide with a solution, the issue should be forwarded to whomever is responsible or can provide with a fix. Please keep this issue open and update us in light of new developments.

"Whilst we do our best to fix issues" does not mean "it's not our problem". We send test cases to VIA when repeatable bugs arise, and we also report and try to fix bug with upstream code when we can. Assuming we just sit on our hands and do nothing is plain rude.

_Originally posted by @JamesH65 in https://github.com/raspberrypi/linux/issues/3070#issuecomment-594541307_

P33M, 2021-03-15:
The original thread was locked because this issue tracker is not a tech support forum. We have a forum for that.

Are you reporting a bug in the Linux kernel?

kashaiahyah85, 2021-03-17:
> The original thread was locked because this issue tracker is not a tech support forum. We have a forum for that.

It is good form to pin known bugs, or ones that get reported frequently. In this case, the trouble is that it's a _perceived_ bug at the least, and a workaround being actively worked on at most. Because it's being worked on, it should be considered either a bug being fixed or a feature being worked in; in either case the bug tracker is the best place for it.

Pinning some sort of clue as to the state of the matter would certainly make things easier on people who are confused as of current. 

> Are you reporting a bug in the Linux kernel?

That's sort of a philosophical question, really. I don't control the philosophy of the project, though. And so I suppose not. I'll just close this, it's alright.

closed 2021-03-17 by kashaiahyah85