raspberrypi/linux #4221: I2C based kernel module unable to claim IRQ 0

timonsku, 2021-03-18:
I want to use the tca8418 keyboard driver.
https://github.com/torvalds/linux/blob/master/drivers/input/keyboard/tca8418_keypad.c

Though in dmesg I hit this error in the driver: https://github.com/torvalds/linux/blob/master/drivers/input/keyboard/tca8418_keypad.c#L332

```
[    5.537713] tca8418_keypad 5-0034: Unable to claim irq 0; error -22
[    5.537755] tca8418_keypad: probe of 5-0034 failed with error -22
```

The device shows up just fine when using i2cdetect. The issue seems to be that it can't claim that IRQ for Pin 19, though I don't even know where it gets this specific IRQ from and where to even debug this further.
I also tried GPIO18 and GPIO26 with the same outcome.

Here is the used dts: https://pastebin.com/YLVqPsbk

**Expected behaviour**
Kernel module should be able to finish its probe of the I2C port successfully and claim the IRQ.

**Actual behaviour**
Kernel module fails to load with the error describe above

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi CM4

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

* Which firmware version (`vcgencmd version`)?
Jan  8 2021 14:31:16
Copyright (c) 2012 Broadcom
version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

**Logs**
dmesg: https://pastebin.com/iuFWbqpz
config.txt: https://pastebin.com/vjj3qKVy

**Addtional context**

This is my documentation for how I compiled and enabled the module:

Clone and configure kernel with Pi defaults as described here but do not adjust the kernel version string
https://www.raspberrypi.org/documentation/linux/kernel/building.md

Then enable the TCA8418 driver with menuconfig as a module (use / to search for it)
https://www.raspberrypi.org/documentation/linux/kernel/configuring.md

```
# compile just the module
make scripts prepare modules_prepare
make -C . M=drivers/input/keyboard
# copy over to system modules
sudo cp drivers/input/keyboard/tca8418_keypad.ko /lib/modules/$(uname -r)/kernel/drivers/input/keyboard/
#update modules dependencies
sudo depmod

#compile dtoverlay
sudo dtc -@ -I dts -O dtb -o tca8418.dtbo tca8418.dts
#copy dtoverlay over to boot
sudo cp tca8418.dtbo /boot/overlays/

#Load dtoverlay
dtoverlay=i2c5,pins_10_11
dtoverlay=tca8418
```

pelwell, 2021-03-18:
I think the `0` irq is an indication that the I2C framework couldn't or didn't claim an interrupt for the device. Understanding why could be quite involved.

Start by removing the unnecessary parts of the overlay - your device isn't an `interrupt-controller`, and I'm fairly certain that you don't need the `irq-gpio` declaration - `of_irq_get` (called from `i2c_device_probe`) looks for `interrupts` and `interrupt-parent`, both of which you have.

You may be reluctant to build your own kernel, but it is likely to be the quickest way to understand what is going wrong. Follow our guide here: https://www.raspberrypi.org/documentation/linux/kernel/building.md
You'll need to use bcm2711_defconfig for the CM4. The first build will take quite a long time on the CM4 itself, but further builds will be much quicker.

Start by putting `#define DEBUG` at the top of file `drivers/of/irq.c` - that will enable the `pr_debug` output scattered through the file - and check the dmesg output. You can also add some similar lines (I would use `pr_err` so they stand out in dmesg) in `i2c_device_probe` (found in `drivers/i2c/i2c-core-base.c`).

timonsku, 2021-03-18:
I commented out the irq-gpio, interrupts etc. and it still throws that error. So that seems to be unrelated.

pelwell, 2021-03-18:
That's why I described them as unnecessary, rather than harmful.

timonsku, 2021-03-18:
Right yes, just wanted to report back. I'm currently building the kernel and see how that goes.

timonsku, 2021-03-31:
Ok this is weird. I compiled the kernel, added the debug statements. Re-enabled the defintion for the IRQ GPIO pin and now it just works :)
I highly doubt the debug statements changed anything so maybe just a Kernel bug in the current Raspbian? I'm now using 5.10.23-v7l+

pelwell, 2021-03-31:
The current `rpi-update` release is 5.10.25 - if that works (`sudo rpi-update`) then I think we can close this issue.

timonsku, 2021-04-05:
Updating to that kernel causes the driver to silently stop working, no errors related to it in dmesg.
Do I need to re-compile the module for every minor version?

pelwell, 2021-04-06:
Always recompile and reinstall modules when you rebuild the kernel.

timonsku, 2021-04-13:
I don't have access to the hw any more unfortunately but I would just regard this as fixed for now, its unlikely that the bug got fixed with some debug print statements.

closed 2021-04-13 by timonsku