raspberrypi/linux #4223: MMAL regression: Unable to handle kernel paging request in video_usercopy

FD-, 2021-03-18:
It seems some change between the 5.4 kernel and the 5.10.23 kernel introduced a regression in the MMAL that leads to a kernel oops on a Raspberry Pi Zero W when running yavta or a custom application that obtains data from a TC358743 and encodes it using MMAL.

**To reproduce**
1. Make sure you're running a recent 5.10.23 kernel (I obtained it via rpi-update)
2. Compile yavta from https://github.com/FD-/yavta. This is almost like the version by @6by9, just with two minor changes
3. Set the edid: `v4l2-ctl --set-edid=file=1080P50EDID.txt --fix-edid-checksums`
4. Start yavta using 
`./yavta --capture=10000000 -s 1080x720 -n 3 --encode-to=- -m -T /dev/video0 | gst-launch-1.0 -v fdsrc ! tcpserversink port=5001 host=0.0.0.0`
5. For receiving, I use https://play.google.com/store/apps/details?id=ca.frozen.rpicameraviewer&hl=en&gl=US on an Android device, but any other player that can stream raw H264 via TCP should also work

**Expected behaviour**
This should H264-encode the HDMI input and stream it to the client. It should not crash.

**Actual behaviour**
After only a few minutes, a kernel oops occurs that breaks yavta until the system is rebooted. Exact crash log below.

**System**
* Which model of Raspberry Pi? PiZeroW
**Newest Raspberry Pi OS Lite from website + newest kernel from rpi-update**
* Which firmware version (`vcgencmd version`)?
**Newest Raspberry Pi OS Lite from website + newest kernel from rpi-update**
* Which kernel version (`uname -a`)?
**Linux raspberrypi 5.10.23+ #1406 Mon Mar 15 15:37:16 GMT 2021 armv6l GNU/Linux**

**Logs**
```
[  885.329566] 8<--- cut here ---
[  885.341888] Unable to handle kernel paging request at virtual address bf930b90
[  885.355823] pgd = cf485c50
[  885.365049] [bf930b90] *pgd=00000000
[  885.374773] Internal error: Oops: 80000005 [#1] ARM
[  885.386178] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes tc358743 cec 8021q garp stp llc cdc_ether 
r8152 brcmfmac brcmutil sha256_generic libsha256 i2c_mux_pinctrl i2c_mux cfg80211 raspberrypi_hwmon rfkill bcm2835_unicam v4l2_dv_timings v4l2_fwnode bcm28
35_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_m
emops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd videodev vc_sm_cma(C) mc uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[  885.460912] CPU: 0 PID: 25320 Comm: vc.ril.isp Tainted: G         C        5.10.23+ #1406
[  885.476772] Hardware name: BCM2835
[  885.487826] PC is at 0xbf930b90
[  885.499006] LR is at video_usercopy+0x490/0x558 [videodev]
[  885.511891] pc : [<bf930b90>]    lr : [<bf0f4b84>]    psr: 60000013
[  885.525575] sp : c4f9fe18  ip : c4f9fe18  fp : c4f9feec
[  885.538511] r10: 00000000  r9 : c0ba7028  r8 : c4f9fe3c
[  885.551430] r7 : 00000000  r6 : 00000000  r5 : c050560f  r4 : c044560f
[  885.565432] r3 : 00000000  r2 : bf116a60  r1 : 2a023ea1  r0 : c3dcd4e0
[  885.579341] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  885.593964] Control: 00c5387d  Table: 04f90008  DAC: 00000055
[  885.607431] Process vc.ril.isp (pid: 25320, stack limit = 0xa48db7d9)
[  885.621796] Stack: (0xc4f9fe18 to 0xc4fa0000)
[  885.634066] fe00:                                                       00000001 00000000
[  885.649950] fe20: 00000000 00000000 00000000 00000000 c4f50480 b4afbbc4 00000000 00000001
[  885.665875] fe40: 00000001 002a3000 00002003 00000001 00000000 00000000 00000000 00000000
[  885.681761] fe60: 00000000 00000000 00000000 00000000 00000000 00000000 00000001 002a3000
[  885.697621] fe80: 002a3000 00000000 00000000 c4f9fe98 c0098640 c009671c ffffffff c4f9fea8
[  885.713474] fea0: bf0de2d0 015d2054 00000128 00000000 c08c1fe8 00000000 c4f9fedc 2a023ea1
[  885.729367] fec0: c05ccdd8 bf0f4c4c c4f50481 c4f50480 00000000 c0ba7028 b4afbbc4 00000004
[  885.745352] fee0: c4f9fefc c4f9fef0 bf0f4c68 bf0f4700 c4f9ff14 c4f9ff00 bf0ec148 bf0f4c58
[  885.761410] ff00: c044560f c4f50481 c4f9ffa4 c4f9ff18 c0209664 bf0ec108 c4f9ff6c 00000000
[  885.777612] ff20: 00000000 00000189 c0ba7028 015c95a8 00000001 00000000 c4f9ffa4 c4f9ff48
[  885.793953] ff40: c0099238 c0098504 00000000 00000000 ffffffff c00083e4 c4f9e000 00000000
[  885.810373] ff60: c4f9ffa4 c4f9ff70 c007db3c c008448c 604fc8bd 2a023ea1 03e0b269 00000001
[  885.826842] ff80: be95a878 be95a988 00000036 c00083e4 c4f9e000 00000000 00000000 c4f9ffa8
[  885.843401] ffa0: c0008240 c02094c4 00000001 be95a878 00000004 c044560f b4afbbc4 00000001
[  885.859955] ffc0: 00000001 be95a878 be95a988 00000036 00000000 00000000 00000000 b4afbf9c
[  885.876557] ffe0: 0002b064 b4afbbb4 000151cc b6da651c 60000010 00000004 00000000 00000000
[  885.893185] Backtrace:
[  885.904552] [<bf0f46f4>] (video_usercopy [videodev]) from [<bf0f4c68>] (video_ioctl2+0x1c/0x24 [videodev])
[  885.923106]  r10:00000004 r9:b4afbbc4 r8:c0ba7028 r7:00000000 r6:c4f50480 r5:c4f50481
[  885.939786]  r4:bf0f4c4c
[  885.951517] [<bf0f4c4c>] (video_ioctl2 [videodev]) from [<bf0ec148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[  885.970038] [<bf0ec0fc>] (v4l2_ioctl [videodev]) from [<c0209664>] (sys_ioctl+0x1ac/0x88c)
[  885.987431]  r5:c4f50481 r4:c044560f
[  885.999925] [<c02094b8>] (sys_ioctl) from [<c0008240>] (ret_fast_syscall+0x0/0x28)
[  886.016173] Exception stack(0xc4f9ffa8 to 0xc4f9fff0)
[  886.029800] ffa0:                   00000001 be95a878 00000004 c044560f b4afbbc4 00000001
[  886.046752] ffc0: 00000001 be95a878 be95a988 00000036 00000000 00000000 00000000 b4afbf9c
[  886.063724] ffe0: 0002b064 b4afbbb4 000151cc b6da651c
[  886.077489]  r10:00000000 r9:c4f9e000 r8:c00083e4 r7:00000036 r6:be95a988 r5:be95a878
[  886.094176]  r4:00000001
[  886.105386] Code: bad PC value
[  886.117447] ---[ end trace 1ea6dff965dcedef ]---
```

6by9, 2021-03-18:
Please try disabling /dev/vcsm (change the permissions or similar) under 5.4 and see if the issue occurs there too.

5.10 has changed to using vcsm-cma by default, so it would be useful to be able to check whether it is that change or something else that is causing the issue.

FD-, 2021-03-19:
I have now tested disabling /dev/vcsm on the 5.4 kernel, but it doesn't seem to provoke the same crash.

Specifically, I changed the permissions and ownership of /dev/vcsm prior to starting yavta:
```
sudo chmod 000 /dev/vcsm
sudo chown root:root /dev/vcsm
```

Is there anything else you want me to test?

Equidamoid, 2022-05-23:
Any news?
Still getting it on `5.10.92` every ~2-3 hours when encoding 20fps 960x720 video stream using ffmpeg's `h264_v4l2m2m` (with [patched](https://forums.raspberrypi.com/viewtopic.php?p=1780625#p1780625) to make the output useful).

`/dev/vcsm` doesn't seem to exist for me, neither it is mentioned in dmesg.