raspberrypi/linux #4227: Issue in dmesg kernel 5.10.y

tuxd3v, 2021-03-20:
Hello Guys,
I am using Rasbian Lite on my rpi1v v1.0( 256MB RAM ),
And I compiled a 5.10.y kernel with the default defconfig..

I only have Camera V1 on my rpi, and no DSI display..  
But I get this on dmesg, I don't know what is the ISP??

```lua
[    9.036585] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    9.281735] bcm2835-isp bcm2835-isp: bcm2835_isp_probe: ril.isp returned 1 i/p (1 expected), 2 o/p (3 expected) ports
[    9.296972] bcm2835-isp bcm2835-isp: Unregister from media controller
[   10.227853] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
```

I tested this commands and all of them seems to wort nice, raspistill included:
```lua
vcgencmd measure_volts
vcgencmd measure_temp
vcgencmd pm_get_status
mmal_vc_diag stats
mmal_vc_diag camerainfo
mmal_vc_diag mmal-stats
vchiq_test -p 1
vchiq_test -f 1
vchiq_test -b 10
```

Can any one help to to figure out what this problem is?
Many thanks in advance,
Best Regards
tux

6by9, 2021-03-20:
ISP = Image Sensor Processor. The V4L2 driver is used by libcamera.

Your firmware is quite badly out of date. ```vcgencmd version``` would tell you just how old it is, but I'm expecting over a year. It's not offering the functionality that the kernel driver is needing, so the kernel driver aborts loading.
As long as you're not wanting to use libcamera (which will be quite slow on a Pi0 or 1, and really struggle with only 256MB of RAM) then there is no actual issue - it's just a warning message in an error handling path.

tuxd3v, 2021-03-20:
hello 6by9,
You right I updated to the last ones, and everything is working ok now!
Many thanks
tux

closed 2021-03-20 by tuxd3v