raspberrypi/linux #4228: 5.10.17-v8+ Kernel Panic while poweroff - Null Pointer Exception - with TPM-Module SLB 9670

LloydS42, 2021-03-20:
**Describe the bug**
Kernel Panic - Null Pointer Exception - System freeze

**To reproduce**
Reboot or Shutdown the System

**Expected behaviour**
Kernel upgrade in January 2021 from Kernel Version 5.4.83 to 5.10.17

**Actual behaviour**
Raspberry Pi4 with currently RaspianOS

**System**
* _Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW_ (`dmesg | grep Machine`)
Raspberry Pi 4 Model B Rev 1.2
* _Which OS and version (`cat /etc/rpi-issue`)?_
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4
* _Which firmware version (`vcgencmd version`)?_
Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)
* _Which kernel version (`uname -a`)?_
Linux blueberry 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

**Logs**
Take a look under additional content

**Additional context**
Find discusion and more details in forum: [Raspberrypi.org/forum](https://www.raspberrypi.org/forums/viewtopic.php?f=75&t=303564&p=1817702#p1817702)

paulenuta, 2021-03-22:
Same for 32-bit architecture `Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l`.

pelwell, 2021-03-22:
We don't have TPMs and don't really support them, but it looks as though the "ops" field of tpm_chip is NULL when tpm_clk_enable is called at the beginning of tpm_start.

I wonder if `tpm_del_char_device` and `tpm_class_shutdown` are being called from different threads? Try adding some debug output to log entry to and exit from those functions, along with `tpm_chip_unregister`.

LloydS42, 2021-03-22:
The system information described here comes from a 64bit kernel and userland. I have the image of: [downloads.raspberry.org](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/), which is the official RaspianOS 64Bit beta version.
After installation, the system was updated with apt update/upgrade, not anymore.
I have another SD card with the official RaspbinOS, where the 64bit kernel runs with a 32bit userland. The result is also the kernel panic. I'm sure the error has been coming from 5.4 to 5.10 since the kernel update, whether under 32bit or 64bit.

paulenuta, 2021-03-22:
> We don't have TPMs and don't really support them, but it looks as though the "ops" field of tpm_chip is NULL when tpm_clk_enable is called at the beginning of tpm_start.
>
> I wonder if [...]

Hi Phil, you can test on the custom board sent by me in May 2018, it has same TPM chip on SPI CE1.

pelwell, 2021-03-22:
That was several office move/reorganisations ago - I'm not finding it at the moment.

This looks like a simple shutdown problem in the upstream driver. Just add the debug I suggested.

If you want to try a quick fix, insert the following at the beginning of tpm_chip_start:
```
int tpm_chip_start(struct tpm_chip *chip)
{
	int ret;

	if (!chip->ops)
		return -EIO;
	tpm_clk_enable(chip);
```

paulenuta, 2021-03-22:
> int tpm_chip_start(struct tpm_chip *chip)

Are you refer [here](https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/drivers/char/tpm/tpm-chip.c#L100) ?
And recompile kernel?

pelwell, 2021-03-22:
That's the one - insert those two new lines before the call to `tpm_clk_enable`. It will result in a change to the tpm driver module, which will have to be installed, but our building instructions (https://www.raspberrypi.org/documentation/linux/kernel/building.md) explain how to do that.

paulenuta, 2021-03-22:
> Try adding some debug output to log entry to and exit from those functions, along with `tpm_chip_unregister`.

Any quick hand for doing this?

LloydS42, 2021-03-22:
Not yet. I have picked out a new sd card and will put on a fresh system image tonight. Compiling the kernel I will do on the RaspberryPi. Presumably it will run all night. First results, therefore, will not take place until tomorrow morning at the earliest.

pelwell, 2021-03-22:
Try this:
```diff
index ddaeceb7e109..0d55235f0822 100644
--- a/drivers/char/tpm/tpm-chip.c
+++ b/drivers/char/tpm/tpm-chip.c
@@ -101,6 +101,11 @@ int tpm_chip_start(struct tpm_chip *chip)
 {
        int ret;
 
+       if (!chip->ops) {
+               pr_err("%s: Avoiding NULL ops pointer\n", __func__);
+               return -EIO;
+       }
+
        tpm_clk_enable(chip);
 
        if (chip->locality == -1) {
@@ -295,6 +300,7 @@ static int tpm_class_shutdown(struct device *dev)
 {
        struct tpm_chip *chip = container_of(dev, struct tpm_chip, dev);
 
+       pr_err("%s start\n", __func__);
        down_write(&chip->ops_sem);
        if (chip->flags & TPM_CHIP_FLAG_TPM2) {
                if (!tpm_chip_start(chip)) {
@@ -304,6 +310,7 @@ static int tpm_class_shutdown(struct device *dev)
        }
        chip->ops = NULL;
        up_write(&chip->ops_sem);
+       pr_err("%s end\n", __func__);
 
        return 0;
 }
@@ -465,6 +472,7 @@ static int tpm_add_char_device(struct tpm_chip *chip)
 
 static void tpm_del_char_device(struct tpm_chip *chip)
 {
+       pr_err("%s start\n", __func__);
        cdev_device_del(&chip->cdev, &chip->dev);
 
        /* Make the chip unavailable. */
@@ -482,6 +490,7 @@ static void tpm_del_char_device(struct tpm_chip *chip)
        }
        chip->ops = NULL;
        up_write(&chip->ops_sem);
+       pr_err("%s end\n", __func__);
 }
 
 static void tpm_del_legacy_sysfs(struct tpm_chip *chip)
@@ -636,6 +645,7 @@ EXPORT_SYMBOL_GPL(tpm_chip_register);
  */
 void tpm_chip_unregister(struct tpm_chip *chip)
 {
+       pr_err("%s start\n", __func__);
        tpm_del_legacy_sysfs(chip);
        if (IS_ENABLED(CONFIG_HW_RANDOM_TPM))
                hwrng_unregister(&chip->hwrng);
@@ -643,5 +653,6 @@ void tpm_chip_unregister(struct tpm_chip *chip)
        if (chip->flags & TPM_CHIP_FLAG_TPM2)
                cdev_device_del(&chip->cdevs, &chip->devs);
        tpm_del_char_device(chip);
+       pr_err("%s end\n", __func__);
 }
 EXPORT_SYMBOL_GPL(tpm_chip_unregister);
```
I've used pr_err so they stand out in the coloured dmesg output to a console, but only the first is a real error.

pelwell, 2021-03-22:
Compiling a kernel on a Pi 4 shouldn't take more than about an hour.

paulenuta, 2021-03-22:
> 	if (!chip->ops)
> 		return -EIO;

Now I'm rebuilding with this, but on 3B+

paulenuta, 2021-03-22:
> That was several office move/reorganisations ago - I'm not finding it at the moment.
>
> This looks like a simple shutdown problem in the upstream driver. Just add the debug I suggested.
>
> If you want [...]

Except the kernel rebuild time on 3B+, this is truly quick fix :). No more hang on reeboot/poweroff.
I've started recompile with debug activated.

paulenuta, 2021-03-22:
Also now there is no more `tpm tpm0: A TPM error (256) occurred attempting the self test` on dmesg 
`dmesg | grep tpm
[   10.592971] tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 16)`.

paulenuta, 2021-03-22:
@pelwell How can we solve this upstream driver problem on 5.10.y and avoid custom kernel build?
Thanks a lot!

pelwell, 2021-03-22:
I would start by emailing the maintainers:
1. use git log on the driver file to find a commit ID _id_.
2. git format patch -1 _id_.  (creates a file named 0001-...)
3. scripts/get_maintainer.pl 0001-*
4. email some of the people in that list and the most appropriate looking email list, describing the problem and the successful workaround.

I would expect a fairly quick response since it seems like a straightforward bug to reproduce in a rarely used code path.

Once it is patched upstream we can back-port to our active branches, but any patch marked as a fix will get automatically ported to LTS kernels where relevant.

In the meantime we can put a patch into the downstream tree to

LloydS42, 2021-03-22:
@pelwell Thank you also from here. System now stop and restart without kernel panic.
@paulenuta I cannot confirm the tpm error 256 message that has also disappeared from dmesg.
<code>dmesg | grep tpm:</code>
```
3.114229 tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 22)
3.116715 tpm tpm0: A TPM error (256) occurred attempting the self test
3.116729 tpm tpm0: starting up the TPM manually
```
From my side, the issue can be closed.

paulenuta, 2021-03-23:
> dmesg | grep tpm

@LloydS42 did you check only after first reboot/poweroff?
In my case (32-bit) after second reboot I only see `tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 16)`.

paulenuta, 2021-03-23:
I only find this [commit](https://github.com/raspberrypi/linux/commit/1e5ac6300a07ceecfc70a893ebef3352be21e6f8#) related to the problem.

pelwell, 2021-03-23:
That's close, but it doesn't protect against the ops pointer being NULL.

paulenuta, 2021-03-23:
@pelwell this is the output with logging added:
`pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.10.25-v7l-TPM-log-Test+ #3 SMP Tue Mar 23 07:08:45 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~$ sudo poweroff
[  116.129054] 8<--- cut here ---
[  116.132491] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[  116.140982] pgd = b4a898b1
[  116.144067] [00000034] *pgd=03047003, *pmd=00000000
[  116.149349] Internal error: Oops: 206 [#1] SMP ARM
[  116.154512] Modules linked in: 8021q garp stp llc tpm_tis_spi tpm_tis_core tpm rtc_ds1307 spi                                                                                                                                             dev hid_logitech_hidpp sc16is7xx regmap_spi regmap_i2c joydev brcmfmac brcmutil hid_logitech_dj                                                                                                                                              sha256_generic cfg80211 rfkill raspberrypi_ts raspberrypi_hwmon i2c_bcm2835 bcm2835_codec(C) bcm                                                                                                                                             2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_c                                                                                                                                             ontig videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev mc vc_sm_cma(C) rpiv                                                                                                                                             id_mem rpi_backlight uio_pdrv_genirq backlight uio nvmem_rmem gpio_fan ip_tables x_tables ipv6
[  116.207253] CPU: 3 PID: 1 Comm: systemd-shutdow Tainted: G         C        5.10.25-v7l-TPM-l                                                                                                                                             og-Test+ #3
[  116.217682] Hardware name: BCM2711
[  116.221565] PC is at tpm_chip_start+0x1c/0xc0 [tpm]
[  116.226923] LR is at tpm_chip_unregister+0xcc/0x124 [tpm]
[  116.232793] pc : [<bf2bd07c>]    lr : [<bf2c39e4>]    psr: 20000013
[  116.239543] sp : c1909c48  ip : c1909c60  fp : c1909c5c
[  116.245254] r10: c1a6b814  r9 : c0e76954  r8 : c3398000
[  116.250965] r7 : c1205048  r6 : bf26d010  r5 : c32efbf8  r4 : c32ef800
[  116.257991] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : c32ef800
[  116.265023] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  116.272679] Control: 30c5383d  Table: 03c6ecc0  DAC: 55555555
[  116.278950] Process systemd-shutdow (pid: 1, stack limit = 0xb0e21de0)
[  116.286010] Stack: (0xc1909c48 to 0xc190a000)
[  116.290903] 9c40:                   c32ef800 c32efbf8 c1909c74 c1909c60 bf2c39e4 bf2bd06c
[  116.299634] 9c60: c32ef800 c3348800 c1909c8c c1909c78 bf26b160 bf2c3924 c3348800 c3348800
[  116.308365] 9c80: c1909ca4 c1909c90 c08aa324 bf26b14c c3348800 00000000 c1909cc4 c1909ca8
[  116.317108] 9ca0: c084687c c08aa2f8 c1a3be30 c3348800 c12f8fe0 c1205048 c1909cd4 c1909cc8
[  116.325852] 9cc0: c0846950 c084677c c1909cf4 c1909cd8 c0845068 c084693c c3348800 00000870
[  116.334590] 9ce0: 00000000 c1205048 c1909d44 c1909cf8 c08404dc c0844f98 c0b8326c 00000000
[  116.343317] 9d00: c07695a0 a0000013 c0e76954 c1a6b814 c1909d44 16375a27 c032ea38 c3348800
[  116.352035] 9d20: c1205048 00000000 c08aa614 c1356018 c0e76954 c1a6b814 c1909d5c c1909d48
[  116.360749] 9d40: c0840788 c0840358 c3348800 c1205048 c1909d74 c1909d60 c08aa610 c0840778
[  116.369456] 9d60: c3360394 c1205048 c1909d84 c1909d78 c08aa62c c08aa5d8 c1909db4 c1909d88
[  116.378157] 9d80: c083f9d4 c08aa620 c12f9038 c3360200 c3360394 16375a27 c3398000 00000000
[  116.386847] 9da0: 00000000 c1a6b854 c1909dd4 c1909db8 c08aaac8 c083f974 c3398380 c3398000
[  116.395525] 9dc0: 00000000 c1a6b854 c1909df4 c1909dd8 bf0915f4 c08aaa98 c1a6b800 c1a6b810
[  116.404209] 9de0: c1400604 c1a6b854 c1909e0c c1909df8 bf091648 bf0915cc c1a6ae10 c1a6b810
[  116.412893] 9e00: c1909e1c c1909e10 c0847998 bf091638 c1909e54 c1909e20 c0843a8c c0847978
[  116.421587] 9e20: c1909e54 c0e76944 c0247ab0 c1205048 4321fedc 00000000 fee1dead e6366a00
[  116.430282] 9e40: c120f08c 00000058 c1909e64 c1909e58 c024927c c0843914 c1909f94 c1909e68
[  116.438985] 9e60: c024940c c0249244 c1909ed4 c2d10f00 00000027 c1909f60 bef952ac 00000005
[  116.447701] 9e80: c1909ec4 c043b42c c043b278 c0492464 00000000 c1909ea0 c0725e8c c1205048
[  116.456416] 9ea0: c2d10f00 00000027 c1909ed4 c1909f60 bef952ac 00000005 c1909f54 c1909ec8
[  116.465142] 9ec0: c043b42c c0420550 c1909ed0 c1909ed4 00000000 00000005 00000000 00000027
[  116.473865] 9ee0: c1909eec 00000005 fffffff6 bef95270 00000004 bef95f78 00000010 bef95284
[  116.482592] 9f00: 00000005 bef9533c 0000000d b6de3fa8 00000001 00000000 ffffffff 00000000
[  116.491326] 9f20: 30c5387d c0201024 c1909f6c 16375a27 c1909f54 c1205048 c2d10f00 00000000
[  116.500060] 9f40: 00000092 c2d10f00 c1909f94 c1909f58 c043b598 c043b398 00000000 0000001e
[  116.508790] 9f60: 00000000 00000000 c032ea38 16375a27 00000000 00000000 00000000 00000058
[  116.517505] 9f80: c0200204 c1908000 c1909fa4 c1909f98 c0249510 c02492c4 00000000 c1909fa8
[  116.526214] 9fa0: c0200040 c0249504 00000000 00000000 fee1dead 28121969 4321fedc e6366a00
[  116.534934] 9fc0: 00000000 00000000 00000000 00000058 00000fff bef95c08 00000000 00436b80
[... cut ...]
[  116.869391] 9fe0: 00448e3c bef95b88 004334a8 b6f20a38
[  116.875052] Code: e52de004 e8bd4000 e5903410 e1a04000 (e5932034)
[  116.881778] ---[ end trace 44b25c1fd41f8e69 ]---
[  116.887044] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[  116.895345] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b                                                                                                                                              ]---
`

LloydS42, 2021-03-23:
> @LloydS42 did you check only after first reboot/poweroff?
> In my case (32-bit) after second reboot I only see `tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 16)`.

Yes. I have done several reboots (dmesg after restarting):
```
#3 (reboot)
3.113170 tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 22)
3.116578 tpm tpm0: A TPM error (256) occurred attempting the self test
3.116593 tpm tpm0: starting up the TPM manually

#4 (reboot)
3.177567 tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 22)
3.181897 tpm tpm0: A TPM error (256) occurred attempting the self test
3.181909 tpm tpm0: starting up the TPM manually

#5 (poweroff)
3.285419 tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 22)
3.287778 tpm tpm0: A TPM error (256) occurred attempting the self test
3.287794 tpm tpm0: starting up the TPM manually
```
Number 1 was inevitably a poweroff, because of kernel panic. The second a reboot.

PeterHuewe, 2021-03-23:
Hi,
I'll take care of this (upstream)
Peter

pelwell, 2021-03-23:
Thank you, Peter - I hope it isn't difficult to solve.

closed 2021-03-24 by LloydS42

felin-arch, 2021-09-04:
@PeterHuewe any news on this one? 

I would love this have this patched as well. How can I help?

paulenuta, 2022-02-17:
Seem to be fixed in `Linux raspberrypi 5.15.23-v7+ #1525 SMP Wed Feb 16 14:34:22 GMT 2022 armv7l [rpi-update]`