raspberrypi/linux #4229: CFExpress Cards

krishna36, 2021-03-21:
Hi All,

   I am trying to use Raspberry as a file transfer/back device.

I have SD cards, XQD cards and I am looking to transfer the files from XQD, SD Cards to SSD.

It's working fine, but recently I started using CFExpress card.

So when I try to do file transfer between CFExpress Cards and SSD I am getting I/O errors

Below is the error:- 
rsync: close failed on "/BlueSSD/PhotographyTrips/StGeorge3/01morning/20Mar2021_Time_22_11_PM/101NZ602/.KR6_4049.NEF.IUoSaY": Input/output error (5)
rsync error: error in file IO (code 11) at receiver.c(853) [receiver=3.1.3]
“The command rsync -av /Z6II/DCIM/ /BlueSSD/PhotographyTrips/StGeorge3/01morning/20Mar2021_Time_22_11_PM/ failed”

krishna36, 2021-03-21:
Has anyone testing file transfer with CFExpress card reader and SSD ?

krishna36, 2021-03-21:
I followed the steps mentioned in 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

Issue is resolved. 

I have a question, making these changes does that mean that we will have slower transfer rates ?

pelwell, 2021-03-21:
MSD is slower than UAS, except when UAS is broken in which case it is infinitely faster.