raspberrypi/linux #4231: USB not working on rpi 4 8GB

Lifferman, 2021-03-22:
Hi everyone,

i tried to make working my USB devices on my rpi 4 8gb.

It seems to be a problem with the VL805

`root@raspberrypi:/home/pi# rpi-eeprom-update`
`BCM2711 detected`
`VL805 firmware in bootloader EEPROM`
`Checking for updates in /lib/firmware/raspberrypi/bootloader/default`
`Use raspi-config to select either the default-production release or latest update.`
`BOOTLOADER: up-to-date`
`CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)`
` LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)`
`RELEASE: default`
`VL805: up-to-date`
`CURRENT: 00000000`
` LATEST: 00000000`

`root@raspberrypi:/home/pi# dmesg | grep xhci_hcd`
`[    1.414091] xhci_hcd 0000:01:00.0: xHCI Host Controller`
`[    1.414144] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1`
`[    1.914212] xhci_hcd 0000:01:00.0: can't setup: -110`
`[    1.914252] xhci_hcd 0000:01:00.0: USB bus 1 deregistered`
`[    1.914306] xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110`
`[    1.914358] xhci_hcd: probe of 0000:01:00.0 failed with error -110`

I search about an entire week about this but i can't find a fix... please help, i'm about to get crazy

timg236, 2021-03-22:
Does adding over_voltage=1 in config.txt help? If so, run the following commands then you should be able to remove that line from config.txt

"sudo apt update"
"sudo apt full-upgrade"

Lifferman, 2021-03-22:
i just tried adding `over_voltage=1` in config.txt, no change and already up-to-date
```
root@raspberrypi:/home/pi# sudo apt update && sudo apt full-upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

rma153, 2021-03-23:
Have you tried adding total_mem=3072 to config.txt? This is the workaround for the RAM limitation in the rpi 4 (4G and 8G models). It solves the dead USB ports issue.

timg236, 2021-03-23:
Assuming you are running Raspberry Pi OS (see questions in the new bug template) there is no need to put total_mem=3072 on any Pi. 
Since it's the 8GB and booting the bootloader is running ok.

Lifferman, 2021-03-23:
I tried, to be sure, to add `total_mem=3072`, it doesn't help..

i've tried 32 & 64 bit Raspbian, and ubuntu 64 bit, with no USB.

timg236, 2021-03-23:
Has it ever worked? 

It would be worth using the Raspberry Pi Imager to install the bootloader with default configuration. Select SD as the boot mode. This will try SD and then USB if no SD is found.

Next, boot without an SD card or USB and see if there are any PCIe or USB errors on the HDMI diagnostics screen. 

This would help identify whether there is a hardware failure.

Lifferman, 2021-03-23:
No, it never worked. i buy it 2 mouth ago, and i tried recently to use it (got no time early for it).
The reseller (kubii.fr) say it can't be a hardware problem...

i'm currently at work, i'll try it when i come home.

Lifferman, 2021-03-23:
> Has it ever worked?
>
> It would be worth using the Raspberry Pi Imager to install the bootloader with default configuration. Select SD as the boot mode. This will try SD and then USB if no SD is [...]

After i boot with SDcard, it show a green screen. If i start it without SD or USB (only a keybord plugged in), it show this
[image]

If i use Pi imager on a USB, and select the USB boot, it show this directly (without green screen):
[image]

timg236, 2021-03-23:
The green screen is recovery.bin indicating that it has finished updating the bootloader. If you flash an OS image to that SD card then it should boot as normal.

The logs show that USB is working because it is able to enumerate devices and find a mass storage device. The mass storage device appears to have a GPT partition but no OS.

What happens if you install Raspberry Pi OS latest to the SD card and boot that?

Lifferman, 2021-03-23:
IT'S WORKING!

I can't check with TV screen, because i got only one HDMI cable and it's used somewhere else rn.
But my keyboard is working, i can activate "lock num" key!

no error spotted:
```
root@raspberrypi:/home/pi# dmesg | grep xhci_hcd
[    1.184478] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.184513] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    1.187435] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0000003000000890
[    1.190071] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    1.190107] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    1.190131] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    1.552707] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    2.062764] usb 1-1.1: new low-speed USB device number 3 using xhci_hcd
[  106.793075] usb 1-1.2: new low-speed USB device number 4 using xhci_hcd
```
And it is detecting USB flash drive when `fdisk -l`:
```
Device     Start      End  Sectors  Size Type
/dev/sda1   2048 15665151 15663104  7.5G Microsoft basic data
```

A big THANKS for you all <3

closed 2021-03-23 by Lifferman

kurtn2005, 2023-03-13:
> IT'S WORKING!
>
> I can't check with TV screen, because i got only one HDMI cable and it's used somewhere else rn. But my keyboard is working, i can activate "lock num" key!
>
> no error spotted:
>
> ``` [...]

Hi @Lifferman ,

I have got the same issue, could you please tell how you fix it? I'm really appreciate it.

osmankiris, 2023-08-08:
Hi guys, @Lifferman @kurtn2005
I also have the same issue, could you please tell me how you fix it? I really appreciate it.
Best regards,

kurtn2005, 2023-08-17:
> Hi guys, @Lifferman @kurtn2005 I also have the same issue, could you please tell me how you fix it? I really appreciate it. Best regards,

My usb module was damaged due to some whatever I made, I found an tv set repair guy to fix.