raspberrypi/linux #4233: Full-KMS not working for composite output on Pi4

kFYatek, 2021-03-22:
There is a couple of things broken in the kernel that prevent composite output from working on Raspberry Pi 4 Model B.

These are actually pretty trivial to fix:
* [VEC's address](https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/arch/arm/boot/dts/bcm283x.dtsi#L493) is set to `0x7e806000` on all Pis. On Pi4, it is actually located at `0x7ec13000`. This needs to be fixed in the device tree.
* The `VEC_DAC_CONFIG` register is [configured to `0x460C000C`](https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/drivers/gpu/drm/vc4/vc4_vec.c#L454) by the VEC driver. This is also the value set by the firmware on Pi3 (I'm not sure about the older models). But on Pi4, the value set by the firmware is `0x61800000`, while `0x460C000C` breaks the image. I'm not sure how to interpret that; possibly consulting firmware devs would be a good idea.
  * Note; `0x61800000` produces a usable image on the Pi3 for me as well. But again, I have very little idea what that means, especially about the lower bits - the higher bits seem to configure some DC bias.
* The PixelValve for the VEC on Pi4 is [configured to use clock source 0](https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/drivers/gpu/drm/vc4/vc4_crtc.c#L994). This should actually be `PV_CONTROL_CLK_SELECT_VEC`, like [for Pi3](https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/drivers/gpu/drm/vc4/vc4_crtc.c#L940).

I applied this quick and dirty patch onto the current `rpi-5.10.y` branch:
```patch
diff --git a/drivers/gpu/drm/vc4/vc4_crtc.c b/drivers/gpu/drm/vc4/vc4_crtc.c
index 6d71c7e6979a..eb808e111638 100644
--- a/drivers/gpu/drm/vc4/vc4_crtc.c
+++ b/drivers/gpu/drm/vc4/vc4_crtc.c
@@ -999,7 +999,7 @@ static const struct vc4_pv_data bcm2711_pv3_data = {
        .fifo_depth = 64,
        .pixels_per_clock = 1,
        .encoder_types = {
-               [0] = VC4_ENCODER_TYPE_VEC,
+               [PV_CONTROL_CLK_SELECT_VEC] = VC4_ENCODER_TYPE_VEC,
        },
 };
 
diff --git a/drivers/gpu/drm/vc4/vc4_vec.c b/drivers/gpu/drm/vc4/vc4_vec.c
index bd5b8eb58b18..ec96b77a8d80 100644
--- a/drivers/gpu/drm/vc4/vc4_vec.c
+++ b/drivers/gpu/drm/vc4/vc4_vec.c
@@ -451,10 +451,7 @@ static void vc4_vec_encoder_enable(struct drm_encoder *encoder)
        VEC_WRITE(VEC_CONFIG2,
                  VEC_CONFIG2_UV_DIG_DIS | VEC_CONFIG2_RGB_DIG_DIS);
        VEC_WRITE(VEC_CONFIG3, VEC_CONFIG3_HORIZ_LEN_STD);
-       VEC_WRITE(VEC_DAC_CONFIG,
-                 VEC_DAC_CONFIG_DAC_CTRL(0xc) |
-                 VEC_DAC_CONFIG_DRIVER_CTRL(0xc) |
-                 VEC_DAC_CONFIG_LDO_BIAS_CTRL(0x46));
+       VEC_WRITE(VEC_DAC_CONFIG, 0x61800000);
 
        /* Mask all interrupts. */
        VEC_WRITE(VEC_MASK0, 0);
```
And also loaded this quick and dirty device tree overlay:
```dts
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,brcm2835-vec";

    fragment@0 {
        target = <&vec>;
        __overlay__ {
            reg = <0x7ec13000 0x1000>;
        };
    };
};
```
And I can confirm that then, `enable_tvout=1` and `dtoverlay=vc4-kms-v3d-pi4,composite=1` in `config.txt` result in usable picture in X on Pi4.

I would post a full-blown pull request, but I'm not familiar with device trees and such enough to know how to realize detection of Pi revision in the VEC driver elegantly and in line with best practices in the kernel. There are more problems with the Full-KMS VEC driver, so I might post pull requests for those later, but I don't feel confident enough to mess with device trees. Sorry about that. I could also just set `VEC_DAC_CONFIG` to `0x61800000` regardless of hardware, as that seems to work - but also I think that such change should be consulted with firmware devs (and/or non-public Broadcom documentation), because I really have no idea what I'm setting with that register ;)

6by9, 2021-03-26:
Thanks for investigating - the VEC has largely been ignored so far for Pi4 as it requires specific clock setup and a few other weird configs. When I looked at the full KMS driver it also seemed to be slightly broken for selecting the different output standards, so was building up the energy to try and tackle it as a whole. HDMI and DSI have been taking priority though.

The firmware does set VEC_DAC_CONFIG to 0x460c00c conditionally on it being Pi0-3. Pi4 it leaves it at the default value (it has forced a reset first via VEC_DAC_CONFIG bit 0).
The docs I have don't go into any detail as to what the  parts of that register actually do.
If we need different behaviours for 2835 vs 2711 then we add in a new compatible string for the 2711 version of the VEC - that's relatively easy.

DT changes, the definition of the vec ought to move from bcm283x.dtsi to bcm2835-common.dtsi (Pi0-3) and bcm2711.dtsi (Pi4). Changing the reg address without changing the node name is a little icky.

The change to bcm2711_pv3_data contradicts the docs I have that say the VEC is on output 0. Checking the firmware source there is a comment
> // Pi4 - must use VEC 02 on PixelValve 3 - the documentation is wrong !

So I think that confirms your change. If you want to provide a PR for that then the commit will be marked appropriately as your contribution. Please ensure it has a Signed-of-by tag as it will be upstreamed.

kFYatek, 2021-03-26:
I plan on posting a PR that fixes the output standards thing you mention, among other things, probably over the weekend.

I may do another one for this issue as well, but I don't promise anything - the device trees still seem a little bit intimidating to me ;)

BTW, when experimenting with VEC with the Full-KMS driver on Pi4, I also discovered that it produces a somewhat usable signal even without `enable_tvout=1`. In that configuration, it's just clocked about 0.8% slower than it should be (and messing with `clk_set_rate()` does not fix it), so it's no longer within the tolerances of standard NTSC or PAL, but might be actually still OK for many displays. It breaks the colour subcarrier decoding on typical displays, but it could be relatively easy to mitigate by setting `VEC_CONFIG1_CUSTOM_FREQ` and adjusting `VEC_FREQ3_2`/`VEC_FREQ1_0` accordingly. Do you think supporting such mode of operation would be worthwhile?

popcornmix, 2021-03-26:
> BTW, when experimenting with VEC with the Full-KMS driver on Pi4, I also discovered that it produces a somewhat usable signal even without enable_tvout=1. In that configuration, it's just clocked [...]

Can you test with latest rpi-udpate firmware?
The issue is due to limited PLLs. When you were testing enable_tvout=1 forced the core to 432MHz which allows VEC to get an accurate 108MHz clock. But we found that HDMI also ideally wants a 108MHz clock got audio.

Latest firmware has a reshuffle of PLLs which should mean accurate 108MHz is available with or without enable_tvout=1.

kFYatek, 2021-03-26:
Unfortunately, nothing changed. I ran `rpi-update` which updated my firmware to `edf2e9c318863999c97c50cdb74eee235ede3af5` and kernel to 5.10.25, I rebuilt my tweaked module version against it and without `enable_sdtv=1`, VEC is still running too slow - on a display the colours are garbled, and I measured the horizontal scanning frequency in NTSC mode to be 15609.57 Hz. With `enable_sdtv=1`, it's 15734.45 Hz, much closer to the specified 15734.[265734] Hz.

However, like I mentioned, if I enable `VEC_CONFIG1_CUSTOM_FREQ` and set `VEC_FREQ3_2`/`VEC_FREQ1_0` to `0x2235fe23` (or `0x2a5fa26f` for PAL; in any case, exactly 1.008x the theoretical value when running on proper 108 MHz clock), I get proper colours. There's slightly more artifacting than normal, as line-by-line subcarrier phase relationship is lost, but it's perfectly usable. I don't have a proper CRT to check, but I guess many displays might be OK with such 59.46 Hz (or 49.6 Hz for PAL) signal. Right now I can only check with a USB grabber, and it displays them perfectly.

popcornmix, 2021-03-29:
Looks like kernel side clock driver is using PLLD / 4 / 7 for vec (when clock hasn't been set up from `enable_tvout=1`) which gets:
```
pi@pi4d:~ $ vcgencmd measure_clock plld vec
frequency(100)=3000000000
frequency(10)=107143064
```
when it wants 108MHz.
You can get the right frequency with (using PLLC):
```
sudo busybox devmem 0xfe1010f8 32 0x5a000095
sudo busybox devmem 0xfe1010fc 32 0x5a006000
pi@pi4d:~ $ vcgencmd measure_clock pllc vec
frequency(100)=2592000000
frequency(10)=108000000
```
although that doesn't immediately get a picture on screen.

popcornmix, 2021-03-29:
I did find this description of DAC_CONFIG (for 2711, unsure if earlier chips are identical):
```
31:09 |   | BIAS_CTRL
07:00 |   | DAC_CTRL
```
With DAC_CTRL:
```
[7] analog self test - default = 0
[6] CRC disable pin, 0 = enable (default), 1 = disable
[5] static shuffle data control in 12-bit DAC decoder, 0 = enable (default), 1 = disable
[4] offset-binary format control, 0 = offset-binary (default), 1 = complimentary
[3] input data select control, 0 = i_dac_data, 1 = self-test input data
[2] clock edge select in level shifter block, 0 = positive (default), 1 = negative
[1] clock edge select in retiming block, 0 = positive (default), 1 = negative
[0] DAC power down, 0 = normal operation (default), 1 = power down mode
```

6by9, 2021-03-29:
Earlier versions have

31:24 LDO_BIAS_CTRL - LDO bias control bus 
23:16 DRIVER_CTRL - DAC driver control bus 
15:0 DAC_CTRL - DAC control bus 

Reset value 0 for all fields.

I couldn't find any detail for the bit usage within those fields.

kFYatek, 2021-03-30:
OK, so it seems that in the `0x460C000C` that's used for Pi3 and earlier, bit 3 "input data select control, 0 = i_dac_data, 1 = self-test input data" is set, which seems to be the main culprit of why there is no output with that setting.

This bit must have been unused on have a different meaning on earlier chips.

closed 2021-04-21 by pelwell

eugene-s-nesdev, 2021-07-24:
Ntemis comment:
> Report this to raspberry pi kernel repo so we can fix this in the future

I confirm this problem on RPi3B+ with lastest "Lakka-RPi3.aarch64-nightly-20210724-58dafbe.img" 
- i've installed fresh lastest build and tested it on 1920x1080 display via HDMI - all works good.
- added this lines to config, then plug RPI to CRT-TV via 3.5mm out:
```
hdmi_force_hotplug=0
hdmi_ignore_hotplug=1
sdtv_mode=16
```
Signal is present (i see deep dark-gray backgroung with some noise) but there are no picture.

Lakka 2.2.x/2.3.x works excellent on CRT with same settings

jayare5, 2021-07-25:
I was gonna comment here too :P 
I want to add I've also tried it like these combinations: 

A) config.txt (kms, doesn't work)
```
enable_tvout=1
dtoverlay=vc4-kms-v3d-pi4,composite=1
```

B) config.txt (fkms, doesn't work)
```
enable_tvout=1
dtoverlay=vc4-fkms-v3d-pi4,composite=1
```

C) config.txt and distroconfig.txt (kms, doesn't work)
```
enable_tvout=1
```
```
dtoverlay=vc4-kms-v3d-pi4,composite=1
```
D) config.txt and distroconfig.txt (fkms, works, then hangs)
```
enable_tvout=1
```
```
dtoverlay=vc4-fkms-v3d-pi4,composite=1
```

eugene-s-nesdev, 2021-07-25:
Why is this issue closed if the problem is not solved?

kFYatek, 2021-07-25:
@eugene-s-nesdev The problem that I had when reporting this issue has been solved by merging #4240 onto the main kernel development branch. AFAIK this has not been yet released in production kernel versions. If the current development versions exhibit a similar problem, then it is an unrelated issue that may have been introduced afterwards.

eugene-s-nesdev, 2021-07-25:
Thank you. Does this fix solve RPi3 tvout issues as well as RPi4? 
Was it tested on real hardware?

kFYatek, 2021-07-25:
I had no problems on Pi3 at that time. Yes, I tested the patch on real Pi4 and on Pi3 to make sure it didn't break anything there.

If you have issues on Pi3 then this is definitely a separate, unrelated issue.

popcornmix, 2021-07-25:
github closed the issue automatically as a PR was merged which contained the text "This fixes #4233"

Can users with the issue confirm if it is a problem using RPiOS with the latest rpi-update firmware/kernel?
Reports of it not working with Lakka-RPi3.aarch64-nightly-20210724-58dafbe are less useful as we don't know exactly what is contained in that image.

gouchi, 2021-08-04:
@popcornmix We are using [kernel commit from 21st May](https://github.com/raspberrypi/linux/commit/58f672b8da4a4094283a7fcb29dc63f3eb7ddb62) so the [fix](https://github.com/raspberrypi/linux/pull/4240/files) should be included with this [RPi firmware commit](https://github.com/raspberrypi/firmware/commit/fb0bfa6a669745578041e838fc73cc17e8c543c9). At last, we are using [those default configurations](https://github.com/libretro/Lakka-LibreELEC/tree/master/packages/tools/bcm2835-bootloader/files/3rdparty/bootloader) if needed.

But fair enough, I understand it is better to test with "vanilla" RPi Linux distribution RPiOS.

@eugene-s-nesdev @jayare5 will it possible to make a test with with RPiOS ? Thank you for your tests.

eugene-s-nesdev, 2021-08-04:
Where i can find lastest compiled version + updated kernel for RPi3 ?

eugene-s-nesdev, 2021-08-04:
> Ntemis comment:
> 
> > Report this to raspberry pi kernel repo so we can fix this in the future
> 
> I confirm this problem on RPi3B+ with lastest "Lakka-RPi3.aarch64-nightly-20210724-58dafbe.img"
> 
> * i've installed fresh lastest build and tested it on 1920x1080 display via HDMI - all works good.
> * added this lines to config, then plug RPI to CRT-TV via 3.5mm out:
> 
> ```
> hdmi_force_hotplug=0
> hdmi_ignore_hotplug=1
> sdtv_mode=16
> ```
> 
> Signal is present (i see deep dark-gray backgroung with some noise) but there are no picture.
> 
> Lakka 2.2.x works, 2.3.x hangs at flower, 3.x show blackscreen.

Strange. That lakka-nightly which i've tested had 5.10.50 kernel https://github.com/libretro/Lakka-LibreELEC/commit/eb4a4fe0a29de5afae4e92ee2ea7bb6cd9c22b8e
but still don't working. I need to test RPiOS

eugene-s-nesdev, 2021-08-06:
I've downloaded "2021-05-07-raspios-buster-armhf.img", tested it on LCD via HDMI, and then tried all SDTV-modes on CRT:

```
sdtv_mode=0          / Normal NTSC
sdtv_mode=2          / Normal PAL
sdtv_mode=16        / Progressive NTSC
sdtv_mode=18        / Progressive PAL
```

All modes works correct on CRT.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
```

6by9, 2021-08-06:
If it worked on 5.10.17 then you weren't using Full KMS (vc4-kms-v3d) as the patches weren't merged until 5.10.32.
apt has been updated in the last couple of days and should have 5.10.52 now.

Raspberry Pi OS defaults to vc4-fkms-v3d (aka Fake KMS), where the firmware is still responsible for the display pipeline configuration. You need to edit config.txt to change that.

gouchi, 2021-08-07:
@6by9 it should work using DRM/KMS only ? As on Lakka we don't use Xorg neither Wayland.

Thank you.

Ntemis, 2021-08-09:
@gouchi i think we have our answer on lakka

PR #4240 (2021-04-21): Fix for Full-KMS VEC on Pi4
This fixes #4233

I hope that I didn't mess up any formalities, as this is my first PR here.

I tested it on a Pi 3B and a Pi 4B, both happily display a composite output with Full-KMS now.