raspberrypi/linux #4235: Investigate adding support for Toshiba TC358749 HDMI to CSI2 bridge

6by9, 2021-03-25:
From https://github.com/6by9/raspi_tc358743/issues/2, TC358743 has been superceded by TC358749 which adds in a deinterlacer and scaler.

https://github.com/54shady/qop_kernel/tree/master/drivers/media/i2c includes a fixed up version of the driver to support both chips, although not using many of the extra features of the '749.

Look at pulling these changes in and upstreaming them.

binbinshare, 2021-03-25:
[image]
I see this 749 on the official website, which shows that it has been discontinued

binbinshare, 2021-03-25:
[image]
TC358840XBG is ok.

6by9, 2021-03-25:
Indeed it appears that TC358749 has been declared End Of Life (EOL).

Discussion on the forums at https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=307136&p=1840631
TC358840 has already had a driver submitted to mainline but needing review. 
https://patchwork.linuxtv.org/project/linux-media/patch/44d2d84d-f18d-9c80-977f-8a2b8675e70a@xs4all.nl/
Hans Verkuil (one of the V4L2 maintainers) is involved in this driver, and has an active kernel tree at https://git.linuxtv.org/hverkuil/media_tree.git/log/?h=tc358840

I see of no beneficial way to drive '840 with more than the 4 CSI2 data lanes on the Pi as it has one 2-lane and one 4-lane interface. The '840 will want the same number of lanes per interface, so that would mean both have to run in 2-lane mode. In which case why bother with 2 2-lane interfaces instead of 1 4-lane instance.
There may be other boards that can handle it, but not the Pi. Routing CSITX1 out to a second connector may be worthwhile, to support those boards.

binbinshare, 2021-04-25:
https://www.youtube.com/watch?v=Kp6YQF6duwg&amp;list=PLcc7zFstBLj8g5XBTOIDzJoNGT9VmiI0T
hello，I fixed the sound in the new board, and the resolution is up to 1080p60fps

ThomasVon2021, 2021-09-06:
hello:
I fixed the image conversion function of tc358743 I made before. This is the GitHub link：
https://github.com/ThomasVon2021/Raspberry-HDMI-CSI-I2C.

6by9, 2022-09-10:
Closing as unlikely to be done and the chip is EOL.

closed 2022-09-10 by 6by9