raspberrypi/linux #4236: Pi4 kernel 5.10.17-v8+ eth0: hw csum failure

Omoeba, 2021-03-26:
**Is this the right place for my bug report?**
I believe so.

**Describe the bug**
`dmesg` and the syslog are flooded with errors stating `eth0: hw csum failure`.

**To reproduce**
List the steps required to reproduce the issue.
1. Receive packets over the built-in ethernet port

**Expected behaviour**
The error doesn't occur.

**Actual behaviour**
The error occurs.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Pi4
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage4
```
* Which firmware version (`vcgencmd version`)?
```
Feb 25 2021 12:10:40
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux gateway 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
```
[  775.927484] eth0: hw csum failure
[  775.927503] skb len=72 headroom=144 headlen=72 tailroom=1960
               mac=(130,14) net=(144,40) trans=184
               shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
               csum(0xffff3a65 ip_summed=2 complete_sw=0 valid=0 level=0)
               hash(0x0 sw=0 l4=0) proto=0x86dd pkttype=2 iif=2
[  775.927508] dev name=eth0 feat=0x0x0000010000004829
[  775.927514] skb headroom: 00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  775.927518] skb headroom: 00000010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  775.927521] skb headroom: 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  775.927525] skb headroom: 00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  775.927528] skb headroom: 00000040: a0 7f 98 00 80 01 00 00 c5 a8 00 00 00 00 00 00
[  775.927531] skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  775.927535] skb headroom: 00000060: 83 80 8a f6 32 92 01 85 30 36 2c 30 31 34 64 35
[  775.927538] skb headroom: 00000070: 30 2c 36 37 3b 30 30 30 31 3d 74 73 31 2e 32 39
[  775.927541] skb headroom: 00000080: 00 00 33 33 00 00 00 01 84 8a 8d 32 c4 19 86 dd
[  775.927545] skb linear:   00000000: 60 00 00 00 00 20 3a ff fe 80 00 00 00 00 00 00
[  775.927548] skb linear:   00000010: 86 8a 8d ff fe 32 c4 19 ff 02 00 00 00 00 00 00
[  775.927551] skb linear:   00000020: 00 00 00 00 00 00 00 01 86 00 8d 15 40 c0 07 08
[  775.927555] skb linear:   00000030: 00 36 ee 80 00 00 00 00 01 01 84 8a 8d 32 c4 19
[  775.927558] skb linear:   00000040: 05 01 00 00 00 00 05 dc
[  775.927561] skb tailroom: 00000000: 7f e7 5d 8f 4e d6 80 de b7 e0 00 3d e4 70 bc ad
[  775.927565] skb tailroom: 00000010: 03 fe 00 f8 cf 28 86 4c 37 20 e1 2e 10 8a 6f 77
[  775.927568] skb tailroom: 00000020: b6 e9 10 3c 45 1c 02 e6 c0 3f 06 f8 82 0f e0 cd
[  775.927572] skb tailroom: 00000030: 3f 1f 16 10 fe fe 8a 00 4e 1e ee 1f f5 9f 34 71
[  775.927575] skb tailroom: 00000040: f1 9b bf 06 7f 62 0a ab a3 91 ab df b2 58 10 8b
[  775.927578] skb tailroom: 00000050: fe f9 00 03 92 06 26 a0 0f 08 55 7e fe 7f 94 00
[  775.927582] skb tailroom: 00000060: 55 fc 8b 90 05 0f f8 c3 16 d0 7c bb fe dc 72 7b
[  775.927585] skb tailroom: 00000070: 00 13 0f 30 40 f9 02 02 cc 0c 18 04 f8 03 18 ca
[  775.927589] skb tailroom: 00000080: a0 00 03 d0 1c 3f e0 a7 b2 1e c0 9f 47 b9 3b a9
[... 844 lines cut ...]
[  786.089038]  ipv6_rcv+0xd8/0x110 [ipv6]
[  786.089079]  __netif_receive_skb_one_core+0x60/0x88
[  786.089082]  __netif_receive_skb+0x20/0x78
[  786.089085]  netif_receive_skb+0x48/0x1f8
[  786.089089]  ifb_ri_tasklet+0x158/0x278 [ifb]
[  786.089126]  tasklet_action_common.isra.20+0x17c/0x1a0
[  786.089129]  tasklet_action+0x40/0x50
[  786.089132]  __do_softirq+0x1a8/0x510
[  786.089135]  irq_exit+0xe8/0x108
[  786.089138]  __handle_domain_irq+0xa0/0x110
[  786.089141]  gic_handle_irq+0xb0/0xf0
[  786.089143]  el1_irq+0xc4/0x180
[  786.089147]  arch_cpu_idle+0x18/0x28
[  786.089150]  default_idle_call+0x58/0x1d4
[  786.089154]  do_idle+0x25c/0x270
[  786.089157]  cpu_startup_entry+0x30/0x70
[  786.089160]  rest_init+0xd8/0xe8
[  786.089164]  arch_call_rest_init+0x18/0x24
[  786.089167]  start_kernel+0x544/0x578
```

**Additional context**
I appended `genet.skip_umac_reset=n` to my `/boot/cmdline.txt` as per #4184 and the issue kept occurring.
The only way to stop the flood is `sudo ethtool -K eth0 rx off`.
The issue also occurs in kernel versions `5.10.17-v7l+` and `5.10.11-v7l+`.  I haven't tested `5.10.11-v8+`.

Omoeba, 2021-04-20:
The issue persists in kernel 5.10.31-v8+

pelwell, 2021-04-20:
Try giving some information about your network architecture - this isn't affecting many people or GitHub would be flooded, so what might make your environment different?

pelwell, 2021-04-20:
And what is host 84:8a:8d:32:c4:19, that is sending IPV6 all-nodes multicasts? It appears to be a Cisco device.

6by9, 2021-04-20:
The online decoder I found when last looking at this says we have an ICMPv6 Router Advertisement.
https://hpd.gasmi.net/?data=6000000000203AFFFE80000000000000868A8DFFFE32C419FF02000000000000000000000000000186008D1540C007080036EE80000000000101848A8D32C41905010000000005DC&force=ipv6

And a couple of ICMPv6 Multicast Listener Query (130)
https://hpd.gasmi.net/?data=6000000000240001FE80000000000000868A8DFFFE32C419FF0200000000000000000000000000013A0005020000010082007FC02710000000000000000000000000000000000000027D0000&force=ipv6

No specific reason why these messages would flag invalid checksums.

It does look like there's something odd about your network though with
```IPv4: martian source 192.168.1.69 from 169.254.124.175, on dev eth1```
being logged several times in there. Are you bridging between eth0 and eth1 or similar?

Omoeba, 2021-04-21:
`84:8a:8d:32:c4:19` doesn't appear to be connected to my network anymore. I am indeed forwarding traffic between eth0 and eth1 with eth1 connected to the LAN and eth0 connected to the WAN. I'll try disconnecting eth1 completely and see if the issue persists.

pelwell, 2021-04-21:
It would be interesting to see if the problem recurs with traffic from another device, and if so what kind of traffic it is.

Omoeba, 2021-04-21:
The problem still occurs with eth1 completely disconnected. I'll try plugging eth0 into my pc instead of a SB8200 cable modem tomorrow and see if the issue persists.

Omoeba, 2021-04-25:
Plugging the pi directly into my pc caused the issue to disappear despite throwing every kind of traffic I can think of at it. I’ll try the 5.4 bcmgenet driver since I don’t recall the issue occurring prior to 5.10.

Omoeba, 2021-04-26:
The 5.4 bcmgenet driver failed to compile so I unfortunately couldn't test it. However, since I don't recall the issue occurring prior to 5.10, it's likely some change between 5.4 and 5.10 that caused the issue.

ralphrmartin, 2021-05-03:
See also #4184, and what resolved the issue for me. I'm not sure if it is relevant here.

Omoeba, 2021-07-13:
I've done some additional testing and the issue began at kernel version 5.6. No log flooding occurs when rx offloading is enabled on versions 5.4 and 5.5 while 5.6 immediately started flooding syslog with `csum failure` messages. I still have no idea which particular commit caused the issue but it's narrowed down significantly.

wulfy23, 2021-08-22:
what makes my environment different

1. openwrt with possible non-coherant testing patchset for initial support
2. sqm, ctinfo (packet steering to alt cpu)
3. eeprom 8 months old maybe

relation to external traffic (sender) would seem non related from here... i'd be looking at the packet steering element first if it is reproducible with your codebase...

wulfy23, 2021-08-28:
some logs
```
[root@dca632 /usbstick 45°]# dmesg | grep -A15 Hardware
[  243.210946] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  243.216798] Workqueue: mmc_complete mmc_blk_mq_complete_work
[  243.222456] Call trace:
[  243.224902]  dump_backtrace+0x0/0x16c
[  243.228562]  show_stack+0x18/0x30
[  243.231876]  dump_stack+0xd4/0x110
[  243.235273]  netdev_rx_csum_fault.part.0+0x48/0x58
[  243.240062]  netdev_rx_csum_fault+0x3c/0x40
[  243.244241]  __skb_checksum_complete+0xdc/0xe0
[  243.248684]  icmpv6_rcv+0xec/0x570
[  243.252082]  ip6_protocol_deliver_rcu+0xe8/0x504
[  243.256694]  ip6_input+0x98/0xbc
[  243.259917]  ipv6_rcv+0xf0/0x130
[  243.263140]  __netif_receive_skb_one_core+0x48/0x60
[  243.268013]  netif_receive_skb+0x58/0x11c
[  243.272019]  br_pass_frame_up+0x134/0x190
--
[  268.298922] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  268.304750] Call trace:
[  268.307206]  dump_backtrace+0x0/0x16c
[  268.310867]  show_stack+0x18/0x30
[  268.314182]  dump_stack+0xd4/0x110
[  268.317580]  netdev_rx_csum_fault.part.0+0x48/0x58
[  268.322371]  netdev_rx_csum_fault+0x3c/0x40
[  268.326552]  __skb_checksum_complete+0xdc/0xe0
[  268.330995]  icmpv6_rcv+0xec/0x570
[  268.334394]  ip6_protocol_deliver_rcu+0xe8/0x504
[  268.339007]  ip6_input+0x98/0xbc
[  268.342230]  ipv6_rcv+0xf0/0x130
[  268.345454]  __netif_receive_skb_one_core+0x48/0x60
[  268.350327]  netif_receive_skb+0x58/0x11c
[  268.354334]  br_pass_frame_up+0x134/0x190
[  268.358338]  br_handle_frame_finish+0x2e0/0x460
--
[  269.526687] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  269.532509] Call trace:
[  269.534954]  dump_backtrace+0x0/0x16c
[  269.538606]  show_stack+0x18/0x30
[  269.541913]  dump_stack+0xd4/0x110
[  269.545304]  netdev_rx_csum_fault.part.0+0x48/0x58
[  269.550086]  netdev_rx_csum_fault+0x3c/0x40
[  269.554260]  __skb_checksum_complete+0xdc/0xe0
[  269.558695]  icmpv6_rcv+0xec/0x570
[  269.562086]  ip6_protocol_deliver_rcu+0xe8/0x504
[  269.566692]  ip6_input+0x98/0xbc
[  269.569909]  ipv6_rcv+0xf0/0x130
[  269.573127]  __netif_receive_skb_one_core+0x48/0x60
[  269.577994]  netif_receive_skb+0x58/0x11c
[  269.581994]  br_pass_frame_up+0x134/0x190
[  269.585993]  br_handle_frame_finish+0x2e0/0x460
--
[  275.819060] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  275.824888] Call trace:
[  275.827344]  dump_backtrace+0x0/0x16c
[  275.831006]  show_stack+0x18/0x30
[  275.834322]  dump_stack+0xd4/0x110
[  275.837720]  netdev_rx_csum_fault.part.0+0x48/0x58
[  275.842510]  netdev_rx_csum_fault+0x3c/0x40
[  275.846691]  __skb_checksum_complete+0xdc/0xe0
[  275.851134]  icmpv6_rcv+0xec/0x570
[  275.854533]  ip6_protocol_deliver_rcu+0xe8/0x504
[  275.859145]  ip6_input+0x98/0xbc
[  275.862368]  ipv6_rcv+0xf0/0x130
[  275.865592]  __netif_receive_skb_one_core+0x48/0x60
[  275.870464]  netif_receive_skb+0x58/0x11c
[  275.874470]  br_pass_frame_up+0x134/0x190
[  275.878475]  br_handle_frame_finish+0x2e0/0x460
[... 79 lines cut ...]
[  338.959635]  ip6_input+0x98/0xbc
[  338.962853]  ipv6_rcv+0xf0/0x130
[  338.966070]  __netif_receive_skb_one_core+0x48/0x60
[  338.970937]  netif_receive_skb+0x58/0x11c
[  338.974937]  br_pass_frame_up+0x134/0x190
[  338.978936]  br_handle_frame_finish+0x2e0/0x460
--
[  360.441933] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  360.447755] Call trace:
[  360.450199]  dump_backtrace+0x0/0x16c
[  360.453852]  show_stack+0x18/0x30
[  360.457159]  dump_stack+0xd4/0x110
[  360.460550]  netdev_rx_csum_fault.part.0+0x48/0x58
[  360.465333]  netdev_rx_csum_fault+0x3c/0x40
[  360.469506]  __skb_checksum_complete+0xdc/0xe0
[  360.473942]  icmpv6_rcv+0xec/0x570
[  360.477334]  ip6_protocol_deliver_rcu+0xe8/0x504
[  360.481940]  ip6_input+0x98/0xbc
[  360.485157]  ipv6_rcv+0xf0/0x130
[  360.488374]  __netif_receive_skb_one_core+0x48/0x60
[  360.493241]  netif_receive_skb+0x58/0x11c
[  360.497240]  br_pass_frame_up+0x134/0x190
[  360.501239]  br_handle_frame_finish+0x2e0/0x460

```

wulfy23, 2021-09-10:
@Omoeba if you have an ipad or similar idevice on your network...

can you reproduce with all idevices disconnected?

openelectron, 2021-09-10:
Not sure if its helpful, but, I have a custom CM4 carrier board (with poorly tracked Ethernet) this causes the same problem. Moving the same CM4 module to the official carrier board fixes the problem.

My only workaround was to limit the ethernet to 100Mbit using;
ethtool -s eth0 speed 100 duplex full

Omoeba, 2021-09-23:
> @Omoeba if you have an ipad or similar idevice on your network...
> 
> can you reproduce with all idevices disconnected?

I do but the issue still occurs with eth1 completely disconnected and eth0 is the internet-facing interface

wulfy23, 2021-09-28:
```
ip6tables -t mangle -I INPUT -i br-lan \
	-p ipv6-icmp -m icmp6 --icmpv6-type 136 \
	-j DSCP --set-dscp 0x00
```

leezu, 2023-02-16:
Observed this when uploading TB volume data with rclone from a RPi4 to AWS S3. The RPi is directly connected via Ethernet to a Verizon Fios Optical Network Terminal exposing 1Gbit upstream connection. Transfer happened at around 400MBit/s.

```
[Thu Feb 16 02:44:30 2023] bcmgenet fd580000.ethernet eth0: hw csum failure
[Thu Feb 16 02:44:30 2023] skb len=40 headroom=144 headlen=40 tailroom=1992
                           mac=(130,14) net=(144,20) trans=164
                           shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
                           csum(0x3a5f28a1 ip_summed=2 complete_sw=0 valid=0 level=0)
                           hash(0x0 sw=0 l4=0) proto=0x0800 pkttype=0 iif=2
[Thu Feb 16 02:44:30 2023] dev name=eth0 feat=0x0000010000004829
[Thu Feb 16 02:44:30 2023] skb headroom: [REDACTED]
[Thu Feb 16 02:44:30 2023] CPU: 0 PID: 5799 Comm: rclone Tainted: G         C         6.1.7-v8+ #1
[Thu Feb 16 02:44:30 2023] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[Thu Feb 16 02:44:30 2023] Call trace:
[Thu Feb 16 02:44:30 2023]  dump_backtrace.part.0+0xec/0x100
[Thu Feb 16 02:44:30 2023]  show_stack+0x20/0x30
[Thu Feb 16 02:44:30 2023]  dump_stack_lvl+0x8c/0xb8
[Thu Feb 16 02:44:30 2023]  dump_stack+0x18/0x34
[Thu Feb 16 02:44:30 2023]  netdev_rx_csum_fault+0x68/0x70
[Thu Feb 16 02:44:30 2023]  __skb_checksum_complete+0x10c/0x114
[Thu Feb 16 02:44:30 2023]  nf_ip_checksum+0x84/0x160
[Thu Feb 16 02:44:30 2023]  nf_checksum+0x54/0x64
[Thu Feb 16 02:44:30 2023]  nf_conntrack_tcp_packet+0xb0c/0x1820 [nf_conntrack]
[Thu Feb 16 02:44:30 2023]  nf_conntrack_in+0xec/0x850 [nf_conntrack]
[Thu Feb 16 02:44:30 2023]  ipv4_conntrack_in+0x20/0x30 [nf_conntrack]
[Thu Feb 16 02:44:30 2023]  nf_hook_slow+0x54/0xf4
[Thu Feb 16 02:44:30 2023]  nf_hook_slow_list+0x88/0x11c
[Thu Feb 16 02:44:30 2023]  ip_sublist_rcv+0x1e4/0x1f0
[Thu Feb 16 02:44:30 2023]  ip_list_rcv+0xf8/0x19c
[Thu Feb 16 02:44:30 2023]  __netif_receive_skb_list_core+0x190/0x220
[Thu Feb 16 02:44:30 2023]  netif_receive_skb_list_internal+0x194/0x2b0
[Thu Feb 16 02:44:30 2023]  napi_complete_done+0x70/0x210
[Thu Feb 16 02:44:30 2023]  bcmgenet_rx_poll+0x3a0/0x43c
[Thu Feb 16 02:44:30 2023]  __napi_poll+0x40/0x214
[Thu Feb 16 02:44:30 2023]  net_rx_action+0x344/0x3c0
[Thu Feb 16 02:44:30 2023]  __do_softirq+0x198/0x4f0
[Thu Feb 16 02:44:30 2023]  ____do_softirq+0x18/0x24
[Thu Feb 16 02:44:30 2023]  call_on_irq_stack+0x2c/0x60
[Thu Feb 16 02:44:30 2023]  do_softirq_own_stack+0x24/0x3c
[Thu Feb 16 02:44:30 2023]  __irq_exit_rcu+0xd4/0x120
[Thu Feb 16 02:44:30 2023]  irq_exit_rcu+0x18/0x50
[Thu Feb 16 02:44:30 2023]  el0_interrupt+0x54/0x100
[Thu Feb 16 02:44:30 2023]  __el0_irq_handler_common+0x18/0x24
[Thu Feb 16 02:44:30 2023]  el0t_32_irq_handler+0x10/0x20
[Thu Feb 16 02:44:30 2023]  el0t_32_irq+0x190/0x194
```

wulfy23, 2023-03-09:
fwiw there was an interesting post in an openwrt  bug report regarding this afair fingering network "scaling" subsystems... which was my initial gut feeling on this due to prevalence.

i tend to agree with that, however it's occurrence or more importantly lack thereof across similar and dissimilar hardware is pointing me towards actual core/buffer??? (re)allocation  for the above as key here...

just a laymans theory with nothing to substantiate so grain of salt and all of that...