raspberrypi/linux #4239: Errors with Raspberry Pi reference 2021-03-04

mgrouch, 2021-03-27:
Bluetooth and others.
They weren't present in previous kernel build

[   85.988295] Bluetooth: hci0: command 0x0401 tx timeout
[  112.100292] Bluetooth: hci0: command 0x0401 tx timeout
[  133.554922] broken atomic modeset userspace detected, disabling atomic
[  136.358306] Bluetooth: RFCOMM TTY layer initialized
[  136.358333] Bluetooth: RFCOMM socket layer initialized
[  136.358356] Bluetooth: RFCOMM ver 1.11
[  136.644858] input: Simulated Right Button as /devices/virtual/input/input8
[  138.212353] Bluetooth: hci0: command 0x0401 tx timeout
[  138.368778] fuse: init (API version 7.32)
[  139.670267] rfkill: input handler disabled
[  140.228314] Bluetooth: hci0: command 0x0c24 tx timeout
[  142.244306] Bluetooth: hci0: command 0x0c52 tx timeout
[  144.470861] w1_master_driver w1_bus_master1: Attaching one wire slave 00.400000000000 crc 46
[  144.476945] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  149.392746] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  149.596071] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  164.324305] Bluetooth: hci0: command 0x0401 tx timeout
[  190.436896] Bluetooth: hci0: command 0x0401 tx timeout
[  191.238630] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c00000000000 crc ca
[  191.243492] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[  216.548300] Bluetooth: hci0: command 0x0401 tx timeout
[  242.660302] Bluetooth: hci0: command 0x0401 tx timeout
[  261.670601] w1_master_driver w1_bus_master1: Attaching one wire slave 00.200000000000 crc 23
[  261.676038] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[  268.772310] Bluetooth: hci0: command 0x0401 tx timeout
[  294.884303] Bluetooth: hci0: command 0x0401 tx timeout
[  320.322769] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
[  320.327668] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[  320.996295] Bluetooth: hci0: command 0x0401 tx timeout
[  323.646933] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  323.647323] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  323.650162] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  323.650172] ieee80211 phy0: brcmf_cfg80211_get_channel: chanspec failed (-110)
[  323.650186] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  323.650192] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-110)

Kernel:

[   85.988295] Bluetooth: hci0: command 0x0401 tx timeout
[  112.100292] Bluetooth: hci0: command 0x0401 tx timeout
[  133.554922] broken atomic modeset userspace detected, disabling atomic
[  136.358306] Bluetooth: RFCOMM TTY layer initialized
[  136.358333] Bluetooth: RFCOMM socket layer initialized
[  136.358356] Bluetooth: RFCOMM ver 1.11
[  136.644858] input: Simulated Right Button as /devices/virtual/input/input8
[  138.212353] Bluetooth: hci0: command 0x0401 tx timeout
[  138.368778] fuse: init (API version 7.32)
[  139.670267] rfkill: input handler disabled
[  140.228314] Bluetooth: hci0: command 0x0c24 tx timeout
[  142.244306] Bluetooth: hci0: command 0x0c52 tx timeout
[  144.470861] w1_master_driver w1_bus_master1: Attaching one wire slave 00.400000000000 crc 46
[  144.476945] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  149.392746] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  149.596071] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  164.324305] Bluetooth: hci0: command 0x0401 tx timeout
[  190.436896] Bluetooth: hci0: command 0x0401 tx timeout
[  191.238630] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c00000000000 crc ca
[  191.243492] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[  216.548300] Bluetooth: hci0: command 0x0401 tx timeout
[  242.660302] Bluetooth: hci0: command 0x0401 tx timeout
[  261.670601] w1_master_driver w1_bus_master1: Attaching one wire slave 00.200000000000 crc 23
[  261.676038] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[  268.772310] Bluetooth: hci0: command 0x0401 tx timeout
[  294.884303] Bluetooth: hci0: command 0x0401 tx timeout
[  320.322769] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
[  320.327668] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[  320.996295] Bluetooth: hci0: command 0x0401 tx timeout
[  323.646933] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  323.647323] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  323.650162] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  323.650172] ieee80211 phy0: brcmf_cfg80211_get_channel: chanspec failed (-110)
[  323.650186] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[  323.650192] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-110)

mgrouch, 2021-03-31:
Looks like power management related in latest broadcom firmware are breaking
bluetooth

mgrouch, 2021-03-31:
There are even more errors I thin related.
When I try to turn off bluetooth and wifi power saving
I get error:

[ 3320.800397] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[ 3320.800780] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[ 3320.837426] Bluetooth: hci0: command 0x0401 tx timeout

mgrouch, 2021-03-31:
I’ve turned off 1wire put a better power supply 5.2 volts 15 watts
And see still same Bluetooth errors
both on arm64 kernel and armhf. 
Somehow I think it points to Broadcom firmware

pelwell, 2021-03-31:
What is Bluetooth being used for in your system?

Please try adding `over_voltage=2` to your config.txt.

mgrouch, 2021-03-31:
I’ve tried over_voltage=2 same errors. 
I’ve tried only switching airplane mode back and force. Nothing is even connected to Bluetooth yet.

Matheritasiv, 2021-04-11:
My 3b+ had run into the same problem right after I upgraded package `linux-firmware` from version `20201218.646f159-1` to `20210208.b79d239-1`, when the kernel was 5.10.4-v8+. Here are booting logs
```
Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Linux version 5.10.4-v8 (wasiv@Sarch) (aarch64-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1 SMP PREEMPT Tue Jan 5 10:59:11 CST 2021
Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
efi: UEFI not found.
Reserved memory: created CMA memory pool at 0x000000001e800000, size 256 MiB
OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Zone ranges:
  DMA      [mem 0x0000000000000000-0x000000003bffffff]
  DMA32    empty
............
Bluetooth: Core ver 2.22
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO socket layer initialized
Bluetooth: HCI UART driver ver 2.3
Bluetooth: HCI UART protocol H4 registered
Bluetooth: HCI UART protocol Three-wire (H5) registered
Bluetooth: HCI UART protocol Broadcom registered
Bluetooth: hci0: command 0xfc18 tx timeout
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
usbcore: registered new interface driver brcmfmac
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
Bluetooth: hci0: BCM: failed to write update baudrate (-110)
Bluetooth: hci0: Failed to set baudrate
Bluetooth: hci0: command 0x0c03 tx timeout
Bluetooth: hci0: BCM: Reset failed (-110)
Bluetooth: hci0: hardware error 0x00
Bluetooth: hci0: command 0xfc18 tx timeout
Bluetooth: hci0: BCM: failed to write update baudrate (-110)
Bluetooth: hci0: Failed to set baudrate
Bluetooth: hci0: command 0x0c03 tx timeout
brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Bluetooth: BNEP filters: protocol multicast
Bluetooth: BNEP socket layer initialized
Bluetooth: hci0: BCM: Reset failed (-110)
............
```
and logs before that upgradation
```
Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Linux version 5.10.4-v8 (wasiv@Sarch) (aarch64-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1 SMP PREEMPT Tue Jan 5 10:59:11 CST 2021
Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
efi: UEFI not found.
Reserved memory: created CMA memory pool at 0x000000001e800000, size 256 MiB
OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Zone ranges:
  DMA      [mem 0x0000000000000000-0x000000003bffffff]
  DMA32    empty
............
Bluetooth: Core ver 2.22
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO socket layer initialized
Bluetooth: HCI UART driver ver 2.3
Bluetooth: HCI UART protocol H4 registered
Bluetooth: HCI UART protocol Three-wire (H5) registered
Bluetooth: HCI UART protocol Broadcom registered
Bluetooth: hci0: BCM: chip id 107
Bluetooth: hci0: BCM: features 0x2f
Bluetooth: hci0: BCM4345C0
Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
Bluetooth: hci0: BCM4345C0 UART 37.4 MHz wlbga_ref_iLNA_iTR_eLG 0139
Bluetooth: hci0: BCM4345C0 (003.001.025) build 0139
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
usbcore: registered new interface driver brcmfmac
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Bluetooth: BNEP filters: protocol multicast
Bluetooth: BNEP socket layer initialized
............
```
Now my kernel version is 5.10.17-v8+ and I have tried downgrading `linux-firmware` to the old version, but the problem remains.

mgrouch, 2021-09-24:
Found the cause:

Full story here: https://github.com/bareboat-necessities/lysmarine_gen/issues/120

closed 2021-09-24 by mgrouch