raspberrypi/linux #4243: DVB-C Terratec Cinergy HTC 00b2 - drxk & em28xx errors

ChrizZz90, 2021-03-28:
**Describe the bug**
Using a DVB-C/T Terratec Cinergy HTC USB device (0ccd:00b2) with TVHeadend is causing endless error messages and spams the logs. I have 2 sticks and I tested them on 2 Raspi4 (with different kernel versions), but the result is the same. It doesn't matter to install the firmware, using an external powered USB-Hub or change the power adapter. I tried the official one and another one with more power.
My goal is to use both on 1 Raspi4, but doing this now is causing new error messages in seconds and after some time the Raspi4 umounts all USB devices. Especially the umount behavior is critical for me because my goal is to have OMV running on the Raspi4, too. Maybe it is similar to #3631 

**To reproduce**
Install TVHeadend, plug in the USB DVB-C device (downloading the firmware doesn't change anything) and start the channel search or play a channel in TVHeadend. As soon as the DVB-C device is in use, errors appear.

**Actual behaviour**
How the syslog looks after plugging the device in:
`
Mar 29 00:43:55 raspberrypi kernel: [ 2195.187491] usb 1-1.1: new high-speed USB device number 5 using xhci_hcd
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498546] usb 1-1.1: config 1 interface 0 altsetting 1 endpoint 0x82 has invalid wMaxPacketSize 0
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498896] usb 1-1.1: New USB device found, idVendor=0ccd, idProduct=00b2, bcdDevice= 1.00
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498905] usb 1-1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498913] usb 1-1.1: Product: Cinergy HTC Stick
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498921] usb 1-1.1: Manufacturer: TERRATEC 
Mar 29 00:43:55 raspberrypi kernel: [ 2195.498928] usb 1-1.1: SerialNumber: 123456789ABCD
Mar 29 00:43:55 raspberrypi kernel: [ 2195.500116] em28xx 1-1.1:1.0: New device TERRATEC  Cinergy HTC Stick @ 480 Mbps (0ccd:00b2, interface 0, class 0)
Mar 29 00:43:55 raspberrypi kernel: [ 2195.500126] em28xx 1-1.1:1.0: Audio interface 0 found (Vendor Class)
Mar 29 00:43:55 raspberrypi kernel: [ 2195.500134] em28xx 1-1.1:1.0: Video interface 0 found: isoc
Mar 29 00:43:55 raspberrypi kernel: [ 2195.500141] em28xx 1-1.1:1.0: DVB interface 0 found: isoc
Mar 29 00:43:55 raspberrypi kernel: [ 2195.577540] em28xx 1-1.1:1.0: chip ID is em2884
Mar 29 00:43:55 raspberrypi kernel: [ 2195.653890] em28xx 1-1.1:1.0: EEPROM ID = 26 00 00 00, EEPROM hash = 0xfebadddc
Mar 29 00:43:55 raspberrypi kernel: [ 2195.653899] em28xx 1-1.1:1.0: EEPROM info:
Mar 29 00:43:55 raspberrypi kernel: [ 2195.653907] em28xx 1-1.1:1.0: 	microcode start address = 0x0004, boot configuration = 0x00
Mar 29 00:43:56 raspberrypi kernel: [ 2195.660309] em28xx 1-1.1:1.0: 	I2S audio, 5 sample rates
Mar 29 00:43:56 raspberrypi kernel: [ 2195.660317] em28xx 1-1.1:1.0: 	500mA max power
Mar 29 00:43:56 raspberrypi kernel: [ 2195.660326] em28xx 1-1.1:1.0: 	Table at offset 0x27, strings=0x249a, 0x1c6a, 0x1486
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727491] em28xx 1-1.1:1.0: Identified as Terratec Cinergy HTC Stick (card=82)
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727503] em28xx 1-1.1:1.0: Currently, V4L2 is not supported on this model
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727516] em28xx 1-1.1:1.0: dvb set to isoc mode.
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727586] em28xx 1-1.1:1.0: Binding audio extension
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727594] em28xx 1-1.1:1.0: em28xx-audio.c: Copyright (C) 2006 Markus Rechberger
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727602] em28xx 1-1.1:1.0: em28xx-audio.c: Copyright (C) 2007-2016 Mauro Carvalho Chehab
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727889] em28xx 1-1.1:1.0: Endpoint 0x83 high-speed on intf 0 alt 7 interval = 8, size 196
Mar 29 00:43:56 raspberrypi kernel: [ 2195.727897] em28xx 1-1.1:1.0: Number of URBs: 1, with 64 packets and 192 size
Mar 29 00:43:56 raspberrypi kernel: [ 2195.733755] em28xx 1-1.1:1.0: Audio extension successfully initialized
Mar 29 00:43:56 raspberrypi kernel: [ 2195.733770] em28xx 1-1.1:1.0: Binding DVB extension
Mar 29 00:43:56 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Mar 29 00:43:56 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Mar 29 00:43:56 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Mar 29 00:43:56 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Mar 29 00:43:56 raspberrypi systemd-udevd[10263]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 1' failed with exit code 99.
Mar 29 00:43:56 raspberrypi kernel: [ 2196.332715] drxk: status = 0x639260d9
Mar 29 00:43:56 raspberrypi kernel: [ 2196.332725] drxk: detected a drx-3926k, spin A3, xtal 20.250 MHz
Mar 29 00:44:00 raspberrypi kernel: [ 2200.341243] drxk: DRXK driver version 0.9.4300
Mar 29 00:44:00 raspberrypi kernel: [ 2200.372237] drxk: frontend initialized.
Mar 29 00:44:00 raspberrypi kernel: [ 2200.372257] tda18271 12-0060: creating new instance
Mar 29 00:44:00 raspberrypi kernel: [ 2200.374063] tda18271: TDA18271HD/C2 detected @ 12-0060
Mar 29 00:44:01 raspberrypi kernel: [ 2200.838664] dvbdev: DVB: registering new adapter (1-1.1:1.0)
Mar 29 00:44:01 raspberrypi kernel: [ 2200.838678] em28xx 1-1.1:1.0: DVB: registering adapter 0 frontend 0 (DRXK DVB-C DVB-T)...
Mar 29 00:44:01 raspberrypi kernel: [ 2200.838695] dvbdev: dvb_create_media_entity: media entity 'DRXK DVB-C DVB-T' registered.
Mar 29 00:44:01 raspberrypi kernel: [ 2200.840106] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843171] em28xx 1-1.1:1.0: DVB extension successfully initialized
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843195] em28xx 1-1.1:1.0: Registering input extension
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843436] Registered IR keymap rc-nec-terratec-cinergy-xs
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843683] rc rc0: Terratec Cinergy HTC Stick as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/rc/rc0
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843793] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
Mar 29 00:44:01 raspberrypi kernel: [ 2200.843890] input: Terratec Cinergy HTC Stick as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/rc/rc0/input1
Mar 29 00:44:01 raspberrypi kernel: [ 2200.844001] em28xx 1-1.1:1.0: Input extension successfully initialized
Mar 29 00:44:01 raspberrypi kernel: [ 2200.965624] tda18271: performing RF tracking filter calibration
Mar 29 00:44:01 raspberrypi systemd-udevd[10277]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Mar 29 00:44:03 raspberrypi kernel: [ 2203.257110] tda18271: RF tracking filter calibration complete
Mar 29 00:44:03 raspberrypi kernel: [ 2203.257182] em28xx 1-1.1:1.0: DVB: adapter 0 frontend 0 frequency 0 out of range (47000000..864000000)
Mar 29 00:44:03 raspberrypi tvheadend[538]: linuxdvb: adapter added /dev/dvb/adapter0
Mar 29 00:44:03 raspberrypi tvheadend[538]: linuxdvb: adapter 0 setting exclusive flag
`

But as soon as I use the device with TVHeadend:
`
[... 216 lines cut ...]
GPIO 33: level=1 fsel=7 alt=3 func=RXD0 pull=UP
GPIO 34: level=1 fsel=7 alt=3 func=SD1_CLK pull=NONE
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD pull=UP
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0 pull=UP
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1 pull=UP
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2 pull=UP
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3 pull=UP
GPIO 40: level=0 fsel=4 alt=0 func=PWM1_0 pull=NONE
GPIO 41: level=0 fsel=4 alt=0 func=PWM1_1 pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=0 func=INPUT pull=UP
GPIO 45: level=1 fsel=0 func=INPUT pull=UP
BANK2 (GPIO 46 to 53):
GPIO 46: level=0 fsel=0 func=INPUT pull=UP
GPIO 47: level=0 fsel=0 func=INPUT pull=UP
GPIO 48: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 49: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 50: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 51: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 52: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 53: level=0 fsel=0 func=INPUT pull=DOWN
`

I had to stop the output because it provides all of the errors again.

ChrizZz90, 2021-03-31:
@6by9 do you think your [comment 3 years ago](https://github.com/raspberrypi/linux/pull/2091#issuecomment-361538644) might solve my issue? I can't see what "pi_patches/rpi-4.14.y-hauppauge" is. Seems like the link to it doesn't work any more.

I also tried to build a new kernel based on linuxtv media build, but after that every other module like Docker, File System Quota or NFS didn't work any more.

ChrizZz90, 2021-04-11:
I am still trying to find a solution and did a test with the Hauppauge DualHD device. I get the same errors [like the guys in the support board](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=255742&sid=c5089af59f8c4c4928de1f17885d7f0d&start=25). It looks like Raspberry Pi (4?) has issues with DVB-C or DVB-T devices in general?!

nextlooper42, 2021-09-01:
Yes, something seems to be really broken in RPi4 codebase. I tried Hauppauge DualHD, OSMC DVB-T2/DVB-C TV dongle and random chinese dongles and **all of them** have corrupted output for no reason

Curiously this is NOT happening on RPi3B+ with exactly same setup (drivers also).

So problem is not in tuners itself, but with RPi4...

Max11223344, 2021-09-01:
exactly my problem ....
It still happens on Raspi 4 with updated version:  uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux.

And -of course (?) - no problems on Pi 3b+ ... :(

ChrizZz90, 2021-09-01:
well, I solved this issue just by testing lots of different devices. There are chips that work and I am using now this one: August DVB-T230 - I don't have errors, sometimes they appear and I have to plug the devices in again and I do a restart of my Pi4.

Oh, one thing I need to add: Yes it is a DVB-T device but DVB-C is also working ;)

Same answer as in #3631