raspberrypi/linux #4248: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2

mgrouch, 2021-03-31:
dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.17-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000fc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 1949696
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000]   DMA32 zone: 12288 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91224 r8192 d31656 u131072
[    0.000000] pcpu-alloc: s91224 r8192 d31656 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1918976
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1024x600M@60 smsc95xx.macaddr=DC:A6:32:D5:46:6D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=59cbf19f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=1 splash quiet logo.nologo vt.global_cursor_default=1 plymouth.ignore-serial-consoles console=tty3 rd.systemd.show_status=false
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000000bc00000-0x000000000fc00000] (64MB)
[    0.000000] Memory: 7289456K/7798784K available (10944K kernel code, 1918K rwdata, 3912K rodata, 3648K init, 1240K bss, 247184K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36199 entries in 142 pages
[    0.000000] ftrace: allocated 142 pages with 4 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3b8/0x578 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000007] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000244] Console: colour dummy device 80x25
[    0.000300] printk: console [tty1] enabled
[    0.000355] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000380] pid_max: default: 32768 minimum: 301
[    0.000511] LSM: Security Framework initializing
[    0.000751] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000830] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.002155] Disabling memory control group subsystem
[    0.004251] rcu: Hierarchical SRCU implementation.
[    0.005265] EFI services will not be available.
[    0.005745] smp: Bringing up secondary CPUs ...
[    0.006685] Detected PIPT I-cache on CPU1
[... 373 lines cut ...]
[  260.143572] w1_master_driver w1_bus_master1: Family 0 for 00.200000000000.23 is not registered.
[  286.180754] Bluetooth: hci0: command 0x0401 tx timeout
[  312.293044] Bluetooth: hci0: command 0x0401 tx timeout
[  318.779499] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
[  318.783912] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
[  338.405456] Bluetooth: hci0: command 0x0401 tx timeout
[  353.816183] w1_master_driver w1_bus_master1: Attaching one wire slave 00.600000000000 crc 65
[  353.820641] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
[  364.521843] Bluetooth: hci0: command 0x0401 tx timeout
[  390.400867] w1_master_driver w1_bus_master1: Attaching one wire slave 00.e00000000000 crc e9
[  390.405387] w1_master_driver w1_bus_master1: Family 0 for 00.e00000000000.e9 is not registered.
[  390.630152] Bluetooth: hci0: command 0x0401 tx timeout
[  416.742412] Bluetooth: hci0: command 0x0401 tx timeout
[  445.158741] Bluetooth: hci0: command 0x0401 tx timeout
[  448.673020] w1_master_driver w1_bus_master1: Attaching one wire slave 00.100000000000 crc 9d
[  448.677492] w1_master_driver w1_bus_master1: Family 0 for 00.100000000000.9d is not registered.
[  460.084344] v3d fec00000.v3d: MMU error from client L2T (0) at 0x2d61000, pte invalid
[  471.270862] Bluetooth: hci0: command 0x0401 tx timeout
[  484.441272] w1_master_driver w1_bus_master1: Attaching one wire slave 00.900000000000 crc 11
[  484.445790] w1_master_driver w1_bus_master1: Family 0 for 00.900000000000.11 is not registered.
[  497.383046] Bluetooth: hci0: command 0x0401 tx timeout
[  523.495233] Bluetooth: hci0: command 0x0401 tx timeout
[  549.607455] Bluetooth: hci0: command 0x0401 tx timeout
[  554.153866] w1_master_driver w1_bus_master1: Attaching one wire slave 00.500000000000 crc db
[  554.158309] w1_master_driver w1_bus_master1: Family 0 for 00.500000000000.db is not registered.

pelwell, 2021-03-31:
That's expected. What the log doesn't say (because reporting on success is regarded as too verbose) but could is:
```
Direct firmware load for brcm/brcmfmac43455-sdio.txt succeeded
```
I think your Bluetooth timeouts are caused by underpowering the Pi - is the SSD in a power enclosure? - or perhaps due to an interaction with onewire (which also looks unhappy).

mgrouch, 2021-03-31:
I've turned off 1wire. Still same. Looks like power management related in latest broadcom firmware are breaking
bluetooth

Y0ngg4n, 2022-10-01:
Have the same issue.

pelwell, 2022-10-01:
Which one? The reported issue is expected behaviour, bit an error.

Y0ngg4n, 2022-10-04:
I have the problem, that my raspberry pi seems to crash randomly. I checked the logs and this was in the logs.

pelwell, 2022-10-04:
When I say that `Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2` is _expected_ I mean that you _should_ see it. If you don't see that message then it is either that the kernel has been changed to not log that message or that `brcmfmac` has failed before it gets that far. In a way, _not_ seeing it is an error.

Googling the error message is a great problem solving technique, googling the unavoidable warning message isn't.

closed 2022-10-04 by pelwell

XECDesign, 2022-10-04:
Next time we update the firmware, would it make sense to add a symlink to avoid confusing people?

pelwell, 2022-10-04:
We'll need one for each platform, but it's not a bad idea.

Y0ngg4n, 2022-10-04:
@pelwell thank you :)