raspberrypi/linux #4253: Enable Extra file system support failed

Twilight-Dream-Of-Magic, 2021-03-31:
Hello everyone, I have hard problem
Mount failed: unknown filesystem type 'xfs' (custom build-ed gentoo linux system without use grub2 on the my Raspberry PI 4) and today rebuild-ed linux kernel to not using module
I need help

pelwell, 2021-03-31:
Your .config file must contain either `CONFIG_XFS_FS=y` or `CONFIG_XFS_FS=m`.

Twilight-Dream-Of-Magic, 2021-03-31:
> Your .config file must contain either `CONFIG_XFS_FS=y` or `CONFIG_XFS_FS=m`.
I have discord chat history, please to look up

pelwell, 2021-03-31:
You'll need to `sudo modprobe config` (or perhaps `sudo modprobe configs`) before `zcat /proc/config.gz` will work, which will allow you to check the running kernel configuration.

Twilight-Dream-Of-Magic, 2021-03-31:
> You'll need to `sudo modprobe config` (or perhaps `sudo modprobe configs`) before `zcat /proc/config.gz` will work, which will allow you to check the running kernel configuration.
```
twilight-dream-rpi64# sudo modprobe configs
modprobe: FATAL: Module configs not found in directory /lib/modules/5.10.16-v8+
twilight-dream-rpi64# sudo modprobe config 
modprobe: FATAL: Module config not found in directory /lib/modules/5.10.16-v8+
twilight-dream-rpi64#
```

Twilight-Dream-Of-Magic, 2021-03-31:
> ```
> twilight-dream-rpi64# sudo modprobe configs
> modprobe: FATAL: Module configs not found in directory /lib/modules/5.10.16-v8+
> twilight-dream-rpi64# sudo modprobe config
> modprobe: FATAL: Module [...]

Never not work

pelwell, 2021-03-31:
Your kernel config must not include the setting to enable the /proc entry.

Your are not reporting a bug in the kernel, and this isn't a help forum, so I'm closing the issue.

closed 2021-03-31 by pelwell