raspberrypi/linux #4254: Suggestion: Include MPTCP

the-loudspeaker, 2021-03-31:
[Multipath TCP](https://www.multipath-tcp.org/) has been a part of Linux kernel since 5.6. it is under development but in 5.10 it has matured very much and is quite usable. More information about it's progress can be found [here](https://github.com/multipath-tcp/mptcp_net-next/wiki).

It is disabled in Raspberry Pi kernel by default. I mean not enabled in kernel .config itself. Is there a reason for it? I see no harm in enabling it.

y0ug, 2022-09-21:
👍 I would love to see it enable by default. Rebuilding the kernel to just enable this feature is a long process.

matttbe, 2023-07-18:
MPTCP has been added without IPv6 support in #5487.

I guess this ticket can be closed too (I don't have the permission to do that)

closed 2023-07-18 by pelwell