raspberrypi/linux #4259: WiFi stops working as soon as USB 3.0 disk is connected via USB 3.0 hub

ast0815, 2021-04-03:
**Describe the bug**

Whenever I connect a USB 3.0 external disk to the Pi via a USB 3.0 USB hub, WiFi stops working.

**To reproduce**

* Start pinging `8.8.8.8`.
* Plug in USB hub to USB 3 socket in Pi.
* Plug USB disk into any of the ports of the hub.
* See pings no longer going through immediately.

**Expected behaviour**

I would expect the disk not to have any influence on the WiFi at all.

**System**

raspinfo: http://paste.debian.net/1192192/

**Additional info**

I have tried to set the "quirks" mode for the USB-SATA adapter, but that did not help. I also tried to upgrade to the newest kernel with `rpi-update` to no avail. When I plug in the drive directly into the USB 3.0 socket of the Pi, it works fine. It also works when I plug in a USB 2.0 memory stick into the hub, or when I connect the hub to a USB 2.0 plug of the Pi instead of the 3.0 ones.

I can get the WiFi to work again after unplugging the disk, by running `sudo wpa_cli -i wlan0 reconfigure`. When I try this with the disk still attached, the status of the adapter just flips back and forth between `SCANNING`, `ASSOCIATING`, and `DISCONNECTED`.

Here is the output of `lsusb` with everything attached:

```
Bus 002 Device 011: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 008: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 007: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 006: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 011: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 010: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

P33M, 2021-04-03:
Your hub has poor shielding, either in the cable between the Pi and the hub, or on the hub PCB itself.
https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html

You haven't specified the make or model of hub, which makes recommending workarounds a bit difficult. Is the hub upstream cable captive?

ast0815, 2021-04-03:
It is this hub: https://www.box.co.uk/ST103008U2C-StarTech.com-10-Port-USB-3.0-Hub-with-Ch_1876915.html

The cable can be detached.

ast0815, 2021-04-03:
Changing the cable did not seem to change anything, but keeping the hub further away from the Pi seems to "fix" this issue. What a pain.

Thanks a lot for the pointer!

dawigit, 2022-05-14:
### 1st Test: RPi4-bullseye wifi+usb3 (SanDisk 3.2Gen1 32GB)

connecting and mounting of an USB3 disk is working for me.

but after a `cd /mnt/sda` and a simple (or multiple) `ls` all the clients become 'disassociated':

`May 14 12:07:28 pigroap hostapd: wlan0: STA b8:27:eb:xx:xx:xx IEEE 802.11: disassociated`
`May 14 12:07:29 pigroap hostapd: wlan0: STA b8:27:eb:xx:xx:xx IEEE 802.11: disassociated`
`May 14 12:08:10 pigroap hostapd: wlan0: STA b8:27:eb:xx:xx:xx IEEE 802.11: disassociated`
`May 14 12:08:11 pigroap hostapd: wlan0: STA b8:27:eb:xx:xx:xx IEEE 802.11: disassociated`

it seems like after some amount of data has been transferred,  wifi stops working.
unmount / eject works, but has no effect on wifi. 
but when i REMOVE the USB device:

`May 14 12:30:16 pigroap hostapd: wlan0: STA b8:27:eb:xx:xx:xx RADIUS: starting accounting session 0F14253E0897CFD9`

all the clients become re-associated, ssh connections to those clients continue to work.

### 2nd Test: RPi4-bullseye wifi+usb3 (SanDisk 3.2Gen1 64GB)

i was not able to reproduce the wifi failure.

from /var/log/syslog:
(32gb)
`May 14 13:28:00 pigroap kernel: [174678.843386] sda: detected capacity change from 120176640 to 0`
`May 14 13:28:03 pigroap kernel: [174682.116100] usb 2-1: USB disconnect, device number 6`
`May 14 13:28:20 pigroap kernel: [174698.740193] usb 2-1: new SuperSpeed USB device number 7 using xhci_hcd`
`May 14 13:28:20 pigroap kernel: [174698.765866] usb 2-1: New USB device found, idVendor=0781, idProduct=5583, bcdDevice=1.00`
`May 14 13:28:20 pigroap kernel: [174698.765904] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`May 14 13:28:20 pigroap kernel: [174698.765920] usb 2-1: Product:  SanDisk 3.2Gen1`
`May 14 13:28:20 pigroap kernel: [174698.765935] usb 2-1: Manufacturer:  USB`
`May 14 13:28:20 pigroap kernel: [174698.765949] usb 2-1: SerialNumber: 0101d37ee66f17355e95b5f89f3d0916e81757af4897a01f60c70bae98027ac0da1900000000000000000000f8dc6
0faff0620008355810713aa874b`

...
(64gb)
`May 14 13:32:13 pigroap kernel: [174931.615723] usb 2-1: new SuperSpeed USB device number 8 using xhci_hcd`
`May 14 13:32:13 pigroap kernel: [174931.637453] usb 2-1: New USB device found, idVendor=0781, idProduct=5591, bcdDevice= 1.00`
`May 14 13:32:13 pigroap kernel: [174931.637489] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`May 14 13:32:13 pigroap kernel: [174931.637506] usb 2-1: Product:  SanDisk 3.2Gen1`
`May 14 13:32:13 pigroap kernel: [174931.637520] usb 2-1: Manufacturer:  USB`
`May 14 13:32:13 pigroap kernel: [174931.637533] usb 2-1: SerialNumber: 0101cbcd2eff717cc3de8f39e6ab106c438dc68f2d2ba9cd5479fc6d5eb1584b377800000000000000000000fee86
9440017250091558107cf28ad11`

### 3rd Test: RPi4 net-install to USB Disk (the 64GB media mentioned above), wifi
net-install was successful.
os is working, no interference with usb3/wifi

### 4th Test: like the 1st, but with another SanDisk 32GB usb drive
the same here: data transfers from that device result in wifi disassociations.
after physically un-pluging the usb media everything works fine again.

### conclusion
This makes 'what install media are you using?' an important (1st) question (forums/support)
when it comes to problems regarding wifi.
But who supects the installation media or even data transfer from
an usb3 device responsible for wifi disconnects?