raspberrypi/linux #4263: Memory leak on lxpanel

rin67630, 2021-04-07:
**Describe the bug**
The panel of the current version (and previous ones) of Raspberry Pi suffers from a memory leak.
The system runs further but becomes 
**To reproduce**
Just let Raspbery Pi run for a long time, after a few months (almost) no menu function will be callable.
The system runs further but becomes inoperable, can't even call Terminal

**Expected behaviour**
Call e.g terminal

**Actual behaviour**
Got out of memory message "Failed to fork (Cannot allocate memory)"
cf also https://www.raspberrypi.org/forums/viewtopic.php?t=267015

**System**
Cannot load any new task on that system ( working else normally, for what is loaded)
especiallly not Terminal to call reports.

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
All models 
* Which OS and version (`cat /etc/rpi-issue`)?
e.g. 
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch
but also the current version with updates upgrades from 6 weeks ago.

* Which firmware version (`vcgencmd version`)?
e.g.
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
e.g 4.19.66-v7+ #1253

**Logs**
cannot call terminal, hence no log

**Additional context**
N.B. All systems are remote, running VNC with team connection.

# I did find a workaround: *depress the bluetooth button* on the panel. 
This reloads the panel and everything gets cleared!
So could this help to find the culprit?

lionello, 2022-08-27:
I'm running into this too. @rin67630 what do you mean with "depress the bluetooth button"?

JanDiederich, 2023-08-08:
I get this with a self-written software after a few hours. After 10 hours the 4 GiB of my Raspberry Pi 4 are nearly **full**!
I suspected at first my own software, but it uses barely any memory, and if it has a leak it's tiny. lxpanel memory usage _explodes_.

rin67630, 2023-10-04:
> I'm running into this too. @rin67630 what do you mean with "depress the bluetooth button"?
Sorry for the late answer.
In the panel there is a Bluetooth field. If you click on it it fixes the memory leak.

galmok, 2024-06-08:
Restarting lxpanel is the only solution right now, but due to security, it cannot be done with "export DISPLAY=:0 && lxpanelctl restart":

    root@raspberrypi:~# DISPLAY=:0 lxpanelctl restart
    Invalid MIT-MAGIC-COOKIE-1 keyCant connect to display: :0

We need a solution to this problem. 1-2 weeks of of idle time and lxpanel consumes all memory and crashes the OS.

galmok, 2024-07-23:
I upgraded my RPi OS to the latest (there is a guide somewhere) and now the lxpanel is ok (no more huge memory consumption).

Master-Code-Programmer, 2024-07-23:
> I upgraded my RPi OS to the latest (there is a guide somewhere) and now the lxpanel is ok (no more huge memory consumption).

Can confirm. I completely reinstalled Raspberry PI OS in its latest version, and the bug is gone.

closed 2024-07-23 by pelwell

rin67630, 2024-07-24:
The last version?
Do you mean bullseye with wayland?
That does not work on the weakest Raspberries, those freqently used remotely.
I don't consider it solved.

galmok, 2024-07-24:
Just check the version I am running and it says "12 (bookworm)" in /etc/os-release.

rin67630, 2024-07-28:
Yes, I meant Bookworm with Wayland, which has a completely different windowing system. 
So the bug remains for the weakest systems (with the least memory), those running headless and remote, supported by VNC.
Please reopen that issue.

JamesH65, 2026-02-16:
> > Does it work ok on Trixie with Wayland? Is it just Bookworm with X11 that shows the problem?
> 
> Never tested Trixie. Should not happen though.

We know it shouldn't happen, but IIRC it was fixed on Trixie. Since Wayland is the supported backend on Trixie, if it works on Trixie and Wayland then it has already been fixed. We are unlikely to back port to the older OS and even less likely to fix anything X11 related, so to get the fix you need Trixie and (probably) Wayland.