raspberrypi/linux #4264: raspberrypi-kernel package lacks adfs module

ssb22, 2021-04-07:
As NOOBS ships RISC OS, some Raspberry Pi users will be dual-booting RISC OS with Raspbian, and may want to access their RISC OS files from Raspbian. This may especially be the case on the Pi Zero which lacks network support in RISC OS (unless you have a supported chipset in your USB-Ethernet adapter, and some common chipsets like Realtek are NOT supported) so accessing RISC OS partitions is the only way to get files out of it.

The RISC OS boot partition is FAT-formatted and can be mounted from Raspbian. However, RISC OS has problems writing FAT entries to the SD card, frequently corrupting the FAT directory when it attempts to write to this partition. I can therefore use the FAT partition to get files _into_ RISC OS, but once inside RISC OS we must treat that partition as read-only (to avoid corruption) and it is therefore unsuitable for getting files _out_ of RISC OS.

The Linux kernel does have a filesystem module called "adfs" which should be able to read RISC OS partitions (read-write support is marked as dangerous, but read-only would at least be sufficient for getting files out of RISC OS without risking FAT corruption). However, the "adfs" module appears to be missing on the Raspberry Pi. Are we expected to compile a custom kernel if we want this module, or would it be possible to enable it on the default build?

Thanks.

HinTak, 2021-04-14:
If you get the matching kernel source, you should be able to build the adfs as an out-of-tree kernel module (while formally it is in-tree); or convert it to dkms, for the matter.

ssb22, 2021-05-24:
Well I built a kernel on the Pi Zero [as documented at raspberrypi.org](https://www.raspberrypi.org/documentation/linux/kernel/building.md) (I had to `git clone` over `ssh` as there were issues cloning via `https` from the Pi Zero, and the build took over 24 hours—I should probably have disabled unused modules—but it otherwise went well) and I'd put `CONFIG_ADFS_FS=m` into `arch/arm/configs/bcmrpi_defconfig` **but** the module does not work to read files from the installation of RISC OS on the Pi.

`dmesg` says `can't find an ADFS filesystem on dev mmcblk0p7.`

So, either RISC OS on the Pi is using some other filesystem instead of ADFS, or it's using a version of ADFS that the Linux kernel module does not recognise.  Either way, turning on this module is not useful so I'm closing this ticket.

ssb22, 2021-05-24:
Apparently RISC OS on the Pi is using a filesystem it calls SDFS and it is not supported by the Linux kernel at all.

closed 2021-05-24 by ssb22