raspberrypi/linux #4269: The bcm2835 V4L2 camera driver's ISO_SENSITIVITY setting doesn't accept the ISO values specified by V4L2. It takes only values in the range [0,4] but should accept ISO values multiplied by 1000. E.g. to set ISO 100 a value 100*1000 should be accepted by the driver.

joakimoh, 2021-04-11:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

6by9, 2021-04-11:
I think you're misinterpreting the spec.

https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/ext-ctrls-camera.html?highlight=v4l2_cid_iso_sensitivity
```
V4L2_CID_ISO_SENSITIVITY (integer menu)
```
That "menu" bit is very important.
bcm2835-camera enumerates a menu with values of 0 (auto), 100000, 200000, 400000 and 800000 (see https://elixir.bootlin.com/linux/latest/source/drivers/staging/vc04_services/bcm2835-camera/controls.c#L54).
The values of those menu items are the ISO*1000, not the index values.

```v4l2-ctl --list-ctrls-menu``` will enumerate all the controls with all the menu values on a running system. I don't have a Pi with camera connected to quote the actual output.

Note that the spec also says that 
```
The effect of setting this control while the V4L2_CID_ISO_SENSITIVITY_AUTO control is set to a value other than V4L2_CID_ISO_SENSITIVITY_MANUAL is undefined, drivers should ignore such requests.
```
so you must also set V4L2_CID_ISO_SENSITIVITY_AUTO to MANUAL if you wish V4L2_CID_ISO_SENSITIVITY to do anything.

joakimoh, 2021-04-11:
That explains a lot. Thanks for clarifying this!

BTW: the ISO setting doesn’t seem to have any effect (when manual mode is selected) unless you re-start the capturing after changing it. I saw there was already an issue on this but it doesn’t seem to be resolved,or?

Skickat från min iPad

6by9, 2021-04-12:
Closing as not actually an issue.

closed 2021-04-12 by 6by9