raspberrypi/linux #4270: Wrong Hardware line in /proc/cpuinfo on Raspberry Pi 3B+

markshep, 2021-04-11:
Running `grep ^Hardware /proc/cpuinfo` on a Raspberry Pi 3 Model B+ outputs the following:

    Hardware        : BCM2835

However that device contains the BCM2837B0 not the BCM2835.

System info:

    # cat /etc/rpi-issue
    Raspberry Pi reference 2018-06-27
    Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage2
    # vcgencmd version
    Aug 15 2019 12:06:42 
    Copyright (c) 2012 Broadcom
    version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)
    # uname -a
    Linux rpi-3b-plus 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

pelwell, 2021-04-11:
It's the name of the family of processors, not the specific device name.

JamesH65, 2021-04-11:
Check out the FAQ for the reasons why...

https://www.raspberrypi.org/documentation/faqs/#pi-software-cpuinfo

closed 2021-04-11 by JamesH65