raspberrypi/linux #4271: pi4 hard crash with FKMS and 4k60

alanbork, 2021-04-11:
rpi-update'ing to the latest firmware (acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31), I can get a hard crash (have to unplug the power to reset) using the official modetest:

modetest -s 89:4096x2160-60

much older versions of the firmware (January?) did not crash with this command, but the problem was observed with the current firmware of last week (since rpi-update has no log I can't say which exact release that was).

modetest -s 89:4096x2160-50 works with the current firmware.

it's a shame because this new firmware is better otherwise - less tearing.

alanbork, 2021-04-11:
also, I'm getting this in dmesg:
  408.159513] raspberrypi-firmware soc:firmware: Request 0x00048015 returned status 0x80000001
[  408.159536] ------------[ cut here ]------------
[  408.159640] WARNING: CPU: 1 PID: 652 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:403 vc4_plane_set_blank+0x15c/0x168 [vc4]
[  408.159656] vc4_plane_set_blank: firmware call failed. Please update your firmware
[  408.159670] Modules linked in: 8021q garp stp llc brcmfmac brcmutil sha256_generic cfg80211 rfkill v3d gpu_sched raspberrypi_hwmon vc4 cec i2c_bcm2835 drm_kms_helper drm bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) drm_panel_orientation_quirks videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_soc_core mc vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
[  408.160376] CPU: 1 PID: 652 Comm: modetest Tainted: G        WC        5.10.27-v7l+ #1409
[  408.160386] Hardware name: BCM2711
[  408.160396] Backtrace:
[  408.160433] [<c0b78f58>] (dump_backtrace) from [<c0b792ec>] (show_stack+0x20/0x24)
[  408.160448]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6abc
[  408.160465] [<c0b792cc>] (show_stack) from [<c0b7d67c>] (dump_stack+0xcc/0xf8)
[  408.160484] [<c0b7d5b0>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[  408.160499]  r10:c3fd7400 r9:00000009 r8:bf388148 r7:00000193 r6:00000009 r5:bf388148
[  408.160510]  r4:bf3ac8bc r3:c1205094
[  408.160526] [<c0220ab0>] (__warn) from [<c0b79a90>] (warn_slowpath_fmt+0xa4/0xd8)
[  408.160538]  r7:00000193 r6:bf3ac8bc r5:c1205048 r4:bf3ac884
[  408.160623] [<c0b799f0>] (warn_slowpath_fmt) from [<bf388148>] (vc4_plane_set_blank+0x15c/0x168 [vc4])
[  408.160638]  r9:00000000 r8:00000000 r7:c31cc000 r6:c329d620 r5:c1205048 r4:ffffffea
[  408.160788] [<bf387fec>] (vc4_plane_set_blank [vc4]) from [<bf3882dc>] (vc4_plane_atomic_update+0x34/0x38 [vc4])
[  408.160801]  r8:c329ae00 r7:bf3a7b08 r6:c329d440 r5:00000000 r4:c3fd03c0
[  408.160974] [<bf3882a8>] (vc4_plane_atomic_update [vc4]) from [<bf354eb0>] (drm_atomic_helper_commit_planes+0xe8/0x28c [drm_kms_helper])
[  408.161146] [<bf354dc8>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf38a9a0>] (vc4_atomic_complete_commit+0x1b0/0x680 [vc4])
[  408.161161]  r10:c31cc000 r9:00000000 r8:c31cc500 r7:00000000 r6:c31cc000 r5:c3fd03c0
[  408.161171]  r4:00000001
[  408.161314] [<bf38a7f0>] (vc4_atomic_complete_commit [vc4]) from [<bf38afec>] (vc4_atomic_commit+0x17c/0x1cc [vc4])
[  408.161329]  r10:c31cc000 r9:00000000 r8:c31cc500 r7:00000000 r6:00000000 r5:c31cc000
[  408.161339]  r4:c3fd03c0 r3:00000001
[  408.161610] [<bf38ae70>] (vc4_atomic_commit [vc4]) from [<bf2da928>] (drm_atomic_commit+0x54/0x60 [drm])
[  408.161623]  r9:c2fc311c r8:00000001 r7:c329d440 r6:c31cc000 r5:c3fd03c0 r4:00000000
[  408.162013] [<bf2da8d4>] (drm_atomic_commit [drm]) from [<bf2f0f98>] (drm_client_modeset_commit_atomic+0x208/0x248 [drm])
[  408.162026]  r7:c329d440 r6:00000001 r5:c31cc1c4 r4:c3fd03c0
[  408.162412] [<bf2f0d90>] (drm_client_modeset_commit_atomic [drm]) from [<bf2f10c0>] (drm_client_modeset_commit_locked+0x6c/0x198 [drm])
[  408.162426]  r10:ffffe000 r9:00000000 r8:c31cc09c r7:c32ff800 r6:bf367020 r5:c32ff818
[  408.162436]  r4:c31cc000
[  408.162817] [<bf2f1054>] (drm_client_modeset_commit_locked [drm]) from [<bf2f1220>] (drm_client_modeset_commit+0x34/0x50 [drm])
[  408.162831]  r9:00000000 r8:c31cc09c r7:c32ff8b4 r6:bf367020 r5:c32ff800 r4:c31cc000
[  408.163119] [<bf2f11ec>] (drm_client_modeset_commit [drm]) from [<bf35c2c4>] (__drm_fb_helper_restore_fbdev_mode_unlocked+0xb0/0xd8 [drm_kms_helper])
[  408.163130]  r5:c32ff800 r4:00000000
[  408.163316] [<bf35c214>] (__drm_fb_helper_restore_fbdev_mode_unlocked [drm_kms_helper]) from [<bf35c354>] (drm_fbdev_client_restore+0x24/0x2c [drm_kms_helper])
[  408.163328]  r7:bf311058 r6:c31cc0b0 r5:c31cc000 r4:c32ff800
[  408.163614] [<bf35c330>] (drm_fbdev_client_restore [drm_kms_helper]) from [<bf2f09f4>] (drm_client_dev_restore+0x84/0xc8 [drm])
[  408.164002] [<bf2f0970>] (drm_client_dev_restore [drm]) from [<bf2c5840>] (drm_lastclose+0x58/0x5c [drm])
[  408.164016]  r9:00000000 r8:c2733770 r7:c3f62cc0 r6:c3369240 r5:c31cc074 r4:c31cc000
[  408.164401] [<bf2c57e8>] (drm_lastclose [drm]) from [<bf2c5960>] (drm_release+0x11c/0x120 [drm])
[  408.164412]  r5:c31cc074 r4:c31cc000
[  408.164614] [<bf2c5844>] (drm_release [drm]) from [<c043ddec>] (__fput+0x90/0x25c)
[  408.164627]  r7:c28409d0 r6:000a201f r5:c3177998 r4:c3f62cc0
[  408.164642] [<c043dd5c>] (__fput) from [<c043dfd0>] (____fput+0x18/0x1c)
[  408.164656]  r9:00000000 r8:c1356d6c r7:c3708000 r6:c3708554 r5:c2ed2780 r4:c3f62cc0
[  408.164674] [<c043dfb8>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
[  408.164694] [<c0242934>] (task_work_run) from [<c0225724>] (do_exit+0x498/0xb88)
[  408.164708]  r9:00000000 r8:c287fc48 r7:c4079e94 r6:ffffe000 r5:c287fc00 r4:c3708000
[  408.164725] [<c022528c>] (do_exit) from [<c0225eb0>] (do_group_exit+0x50/0xe0)
[  408.164735]  r7:c2a8b480
[  408.164754] [<c0225e60>] (do_group_exit) from [<c023393c>] (get_signal+0x1c0/0xae0)
[  408.164767]  r7:c2a8b480 r6:00000008 r5:00000000 r4:00418004
[  408.164785] [<c023377c>] (get_signal) from [<c020b6bc>] (do_work_pending+0x14c/0x5a8)
[... cut ...]
[  408.164825] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[  408.164836] Exception stack(0xc4079fb0 to 0xc4079ff8)
[  408.164849] 9fa0:                                     00000000 c00464af bebf2354 00000000
[  408.164863] 9fc0: 00000007 bebf2354 c00464af 00000036 0177250c 01775f60 eeeeeeee 00000001
[  408.164876] 9fe0: b6f9908c bebf2334 b6f7f88c b6ed251c 60000010 00000003
[  408.164890]  r10:00000036 r9:c4078000 r8:c0200204 r7:00000036 r6:c00464af r5:bebf2354
[  408.164899]  r4:00000007
[  408.164975] ---[ end trace bcc4cf783efc1df0 ]---

alanbork, 2021-04-14:
seems like it doesn't always completely crash the hardware just lock the current DRM opening app. When it doesn't crash the hardware I was able to ssh in and check the log, and got this further detail:

[   98.998235] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out
[  109.234774] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out

alanbork, 2021-04-14:
tested with 922deae6d1ff141d53cf9e703ffecd6f994e9d4e,  same problem.

6by9, 2021-04-14:
All rpi-update releases are logged in https://github.com/Hexxeh/rpi-firmware/commits/master
You can also choose to checkout any specific one of those releases by using ```sudo rpi-update <commit hash>``` on the hashes from that repo.

If you could narrow down the release that causes the issue it would be useful.
48570ba was a rework of the I2C comms for the HDMI block, with a bug fix in 16e0bdd.
f2df2a6 reworks some of the core clock tree setup.
Those would be the two most likely candidates.

alanbork, 2021-04-14:
test:  modetest -s 89:4096x2160
48570ba954a318feee348d4e642ebd2b58d9dd97:  no video out, but doesn't lock up mode test, can still ^c out
16e0bdd55fee6f61e5c17fdf9312fc8d1b05732a: same as above
7378e549a785bd486c19dc2ed737ca4a1e983351: same as above
f2df2a6a46a475eb95b66fe9659e02356432bbe9: no video out, modetest hard crashes but pi is still shh-able, with following dmesg:

[  189.432092] raspberrypi-firmware soc:firmware: Request 0x00048015 returned status 0x80000001
[  189.432121] ------------[ cut here ]------------
[  189.432221] WARNING: CPU: 1 PID: 615 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:403 vc4_plane_set_blank+0x15c/0x168 [vc4]
[  189.432237] vc4_plane_set_blank: firmware call failed. Please update your firmware
[  189.432251] Modules linked in: 8021q garp stp llc brcmfmac brcmutil vc4 v3d cec gpu_sched raspberrypi_hwmon drm_kms_helper sha256_generic cfg80211 bcm2835_codec(C) bcm2835_v4l2(C) rfkill bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig i2c_bcm2835 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 drm videobuf2_common drm_panel_orientation_quirks videodev snd_soc_core mc vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect rpivid_mem sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio nvmem_rmem i2c_dev ip_tables x_tables ipv6
[  189.432961] CPU: 1 PID: 615 Comm: modetest Tainted: G        WC        5.10.25-v7l+ #1408
[  189.432970] Hardware name: BCM2711
[  189.432980] Backtrace:
[  189.433016] [<c0b72ce8>] (dump_backtrace) from [<c0b7307c>] (show_stack+0x20/0x24)
[  189.433031]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6a3c
[  189.433051] [<c0b7305c>] (show_stack) from [<c0b7740c>] (dump_stack+0xcc/0xf8)
[  189.433071] [<c0b77340>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[  189.433087]  r10:c2f15200 r9:00000009 r8:bf653148 r7:00000193 r6:00000009 r5:bf653148
[  189.433099]  r4:bf6778bc r3:c1205094
[  189.433116] [<c0220ab0>] (__warn) from [<c0b73820>] (warn_slowpath_fmt+0xa4/0xd8)
[  189.433130]  r7:00000193 r6:bf6778bc r5:c1205048 r4:bf677884
[  189.433213] [<c0b73780>] (warn_slowpath_fmt) from [<bf653148>] (vc4_plane_set_blank+0x15c/0x168 [vc4])
[  189.433228]  r9:00000000 r8:00000000 r7:c3239000 r6:c32a5a20 r5:c1205048 r4:ffffffea
[  189.433370] [<bf652fec>] (vc4_plane_set_blank [vc4]) from [<bf6532dc>] (vc4_plane_atomic_update+0x34/0x38 [vc4])
[  189.433385]  r8:c2f14200 r7:bf672b08 r6:c32a5840 r5:00000000 r4:c2874b80
[  189.433546] [<bf6532a8>] (vc4_plane_atomic_update [vc4]) from [<bf2b8eb0>] (drm_atomic_helper_commit_planes+0xe8/0x28c [drm_kms_helper])
[  189.433706] [<bf2b8dc8>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf6559a0>] (vc4_atomic_complete_commit+0x1b0/0x680 [vc4])
[  189.433721]  r10:c3239000 r9:00000000 r8:c3239500 r7:00000000 r6:c3239000 r5:c2874b80
[  189.433731]  r4:00000001
[  189.433866] [<bf6557f0>] (vc4_atomic_complete_commit [vc4]) from [<bf655fec>] (vc4_atomic_commit+0x17c/0x1cc [vc4])
[  189.433880]  r10:c3239000 r9:00000000 r8:c3239500 r7:00000000 r6:00000000 r5:c3239000
[  189.433891]  r4:c2874b80 r3:00000001
[  189.434146] [<bf655e70>] (vc4_atomic_commit [vc4]) from [<bf23e928>] (drm_atomic_commit+0x54/0x60 [drm])
[  189.434160]  r9:c34fe9dc r8:00000001 r7:c32a5840 r6:c3239000 r5:c2874b80 r4:00000000
[  189.434524] [<bf23e8d4>] (drm_atomic_commit [drm]) from [<bf254f98>] (drm_client_modeset_commit_atomic+0x208/0x248 [drm])
[  189.434537]  r7:c32a5840 r6:00000001 r5:c32391c4 r4:c2874b80
[  189.434892] [<bf254d90>] (drm_client_modeset_commit_atomic [drm]) from [<bf2550c0>] (drm_client_modeset_commit_locked+0x6c/0x198 [drm])
[  189.434907]  r10:ffffe000 r9:00000000 r8:c323909c r7:c2bc1200 r6:bf2cb020 r5:c2bc1218
[  189.434917]  r4:c3239000
[  189.435268] [<bf255054>] (drm_client_modeset_commit_locked [drm]) from [<bf255220>] (drm_client_modeset_commit+0x34/0x50 [drm])
[  189.435282]  r9:00000000 r8:c323909c r7:c2bc12b4 r6:bf2cb020 r5:c2bc1200 r4:c3239000
[  189.435549] [<bf2551ec>] (drm_client_modeset_commit [drm]) from [<bf2c02c4>] (__drm_fb_helper_restore_fbdev_mode_unlocked+0xb0/0xd8 [drm_kms_helper])
[  189.435561]  r5:c2bc1200 r4:00000000
[  189.435737] [<bf2c0214>] (__drm_fb_helper_restore_fbdev_mode_unlocked [drm_kms_helper]) from [<bf2c0354>] (drm_fbdev_client_restore+0x24/0x2c [drm_kms_helper])
[  189.435749]  r7:bf275058 r6:c32390b0 r5:c3239000 r4:c2bc1200
[  189.436017] [<bf2c0330>] (drm_fbdev_client_restore [drm_kms_helper]) from [<bf2549f4>] (drm_client_dev_restore+0x84/0xc8 [drm])
[  189.436376] [<bf254970>] (drm_client_dev_restore [drm]) from [<bf229840>] (drm_lastclose+0x58/0x5c [drm])
[  189.436390]  r9:00000000 r8:c3883cc0 r7:c3d8ef00 r6:c34e0dc0 r5:c3239074 r4:c3239000
[  189.436748] [<bf2297e8>] (drm_lastclose [drm]) from [<bf229960>] (drm_release+0x11c/0x120 [drm])
[  189.436759]  r5:c3239074 r4:c3239000
[  189.436948] [<bf229844>] (drm_release [drm]) from [<c043dc40>] (__fput+0x90/0x25c)
[  189.436961]  r7:c2848910 r6:000a201f r5:c359a5a8 r4:c3d8ef00
[  189.436976] [<c043dbb0>] (__fput) from [<c043de24>] (____fput+0x18/0x1c)
[  189.436990]  r9:00000000 r8:c1356d6c r7:c3ec8f80 r6:c3ec94d4 r5:c3d8d9c0 r4:c3d8ef00
[  189.437009] [<c043de0c>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
[  189.437028] [<c0242934>] (task_work_run) from [<c0225724>] (do_exit+0x498/0xb88)
[  189.437042]  r9:00000000 r8:c3384248 r7:c3c6fe94 r6:ffffe000 r5:c3384200 r4:c3ec8f80
[... cut ...]
[  189.437155] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[  189.437166] Exception stack(0xc3c6ffb0 to 0xc3c6fff8)
[  189.437178] ffa0:                                     00000000 c00464af bece5354 00000000
[  189.437193] ffc0: 00000007 bece5354 c00464af 00000036 01db34d4 01db6f60 eeeeeeee 00000001
[  189.437205] ffe0: b6ef308c bece5334 b6ed988c b6e2c51c 60000010 00000003
[  189.437219]  r10:00000036 r9:c3c6e000 r8:c0200204 r7:00000036 r6:c00464af r5:bece5354
[  189.437229]  r4:00000007
[  189.437241] ---[ end trace 8db95d7b189378d2 ]---

alanbork, 2021-04-14:
I went back to 8c2ba64b27a4b6a1dfcd114203c6309f766e2ef4 which I thought had been working at 4096' but it seems my memory is wrong, as it too fails. 3840' is fine, in all these versions of the firmware. BUT: there is some kind of regression here as the locking up and dmesg stack traces reveal. But perhaps it has to do with something unusual about my TV when trying to start up 4096'.

here's the regular suspects:

root@raspberrypi:~# edidparser edid
Enabling fuzzy format match...
Parsing edid...
HDMI:EDID version 1.3, 1 extensions, screen size 95x54 cm
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is SONY_TV___00
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 23-62 Hz, horizontal is 14-136 kHz, max pixel clock is 600 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 3840x2160p @ 60 Hz (97)
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID established timing I/II bytes are 21 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x8180 A9C0 714F B300 0101 0101 0101 0101
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 1
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 2
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 3
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:0
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA format: code 97, 3840x2160p @ 60Hz
HDMI:EDID found CEA format: code 96, 3840x2160p @ 50Hz
HDMI:EDID found CEA format: code 93, 3840x2160p @ 24Hz
HDMI:EDID found CEA format: code 94, 3840x2160p @ 25Hz
HDMI:EDID found CEA format: code 95, 3840x2160p @ 30Hz
HDMI:EDID found CEA format: code 98, 4096x2160p @ 24Hz
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz
HDMI:EDID found CEA format: code 34, 1920x1080p @ 30Hz
HDMI:EDID found CEA format: code 60, 1280x720p @ 24Hz
HDMI:EDID found CEA format: code 62, 1280x720p @ 30Hz
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found CEA format: code 101, 4096x2160p @ 50Hz
HDMI:EDID found CEA format: code 102, 4096x2160p @ 60Hz
HDMI:EDID found audio format 6 channels PCM, sample rate: 32|44|48|88|96|176|192 kHz, sample size: 16|20|24 bits
HDMI:EDID found audio format 6 channels AC3, sample rate: 32|44|48 kHz, bitrate: 640 kbps
HDMI:EDID found audio format 6 channels DTS, sample rate: 32|44|48 kHz, bitrate: 1504 kbps
HDMI:EDID found audio format 8 channels Dobly Digital+, sample rate: 48 kHz, codec define: 0
HDMI:EDID found HDMI VSDB length 14
HDMI:EDID HDMI VSDB has physical address 3.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:yes 30-bit:yes DC_yuv444:yes
HDMI:EDID HDMI VSDB max TMDS clock 300 MHz
HDMI:EDID HDMI VSDB content type support: game|cinema|photo|text
HDMI:EDID HDMI VSDB supports extended resolutions 1,2,3,4
HDMI:EDID ignoring non HDMI VSDB with IEEE reg 0xc45dd8
HDMI:EDID found Video Capability DB length 2
HDMI:EDID video capability: CE:3 IT:2 PT:0 QS:3
[... 63 lines cut ...]
raspberrypi/firmware#20 1920x1080 29.97 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
raspberrypi/firmware#21 1920x1080 24.00 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
raspberrypi/firmware#22 1920x1080 23.98 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
raspberrypi/firmware#23 1680x1050 59.88 1680 1728 1760 1840 1050 1053 1059 1080 119000 flags: phsync, nvsync; type: driver
raspberrypi/firmware#24 1600x900 60.00 1600 1624 1704 1800 900 901 904 1000 108000 flags: phsync, pvsync; type: driver
raspberrypi/firmware#25 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
raspberrypi/firmware#26 1152x864 75.00 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
[... cut ...]
raspberrypi/firmware#40 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver
raspberrypi/firmware#41 720x480i 29.97 720 739 801 858 480 488 494 525 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver
raspberrypi/firmware#42 640x480 60.00 640 656 752 800 480 490 492 525 25200 flags: nhsync, nvsync; type: driver
raspberrypi/firmware#43 640x480 59.94 640 656 752 800 480 490 492 525 25175 flags: nhsync, nvsync; type: driver
#0 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: userdef, driver

alanbork, 2021-04-15:
fyi I had a chance to test the latest firmware with another 4k tv (50S423) and it also hard crashes modetest and/or locks up the pi4 when I run modetest -s 89:4096x2160. So this isn't limited to one tv. here's the edid from this tv as well:

HDMI:EDID version 1.3, 1 extensions, screen size 111x62 cm
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is 50S423
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 24-75 Hz, horizontal is 15-135 kHz, max pixel clock is 600 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found preferred CEA detail timing format: 3840x2160p @ 60 Hz (97)
HDMI:EDID found CEA detail timing format: 3840x2160p @ 30 Hz (95)
HDMI:EDID established timing I/II bytes are AD CF 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x714F 8140 8180 8100 9500 B300 D1C0 81C0
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 0
HDMI:EDID found DMT format: code 32, 1280x960p @ 60 Hz (4:3) in standard timing 1
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 2
HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 3
HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 5
HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 6
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 7
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found DMT detail timing format: 1360x768p @ 60 Hz (39)
HDMI:EDID found CEA format: code 97, 3840x2160p @ 60Hz
HDMI:EDID found CEA format: code 95, 3840x2160p @ 30Hz
HDMI:EDID found CEA format: code 93, 3840x2160p @ 24Hz
HDMI:EDID found CEA format: code 102, 4096x2160p @ 60Hz
HDMI:EDID found CEA format: code 100, 4096x2160p @ 30Hz
HDMI:EDID found CEA format: code 98, 4096x2160p @ 24Hz
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz
HDMI:EDID found CEA format: code 34, 1920x1080p @ 30Hz
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
HDMI:EDID found CEA format: code 62, 1280x720p @ 30Hz
HDMI:EDID found CEA format: code 60, 1280x720p @ 24Hz
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
HDMI:EDID found audio format 6 channels AC3, sample rate: 32|44|48 kHz, bitrate: 640 kbps
HDMI:EDID found audio format 8 channels Dobly Digital+, sample rate: 32|44|48 kHz, codec define: 0
HDMI:EDID found Video Capability DB length 2
HDMI:EDID video capability: CE:3 IT:3 PT:0 QS:3
HDMI:EDID extended data block tag 0x05 length 3 not supported
HDMI:EDID found HDMI VSDB length 13
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:yes 30-bit:yes DC_yuv444:yes
HDMI:EDID HDMI VSDB max TMDS clock 300 MHz
HDMI:EDID HDMI VSDB does not support content type
HDMI:EDID HDMI VSDB supports extended resolutions 1,3,4
HDMI:EDID ignoring non HDMI VSDB with IEEE reg 0xc45dd8
HDMI:EDID extended data block tag 0x06 length 3 not supported
HDMI:EDID extended data block YCbCr420CapMap - length 2
HDMI:EDID adding mandatory support for CEA (3) 720x480p @ 60Hz
[... 86 lines cut ...]
[  570.208313]  r9:00000000 r8:c257caa0 r7:c3c93b40 r6:c32649c0 r5:c3064874 r4:c3064800
[  570.208641] [<bf2a47e8>] (drm_lastclose [drm]) from [<bf2a4960>] (drm_release+0x11c/0x120 [drm])
[  570.208651]  r5:c3064874 r4:c3064800
[  570.208822] [<bf2a4844>] (drm_release [drm]) from [<c043ddec>] (__fput+0x90/0x25c)
[  570.208834]  r7:c28623d0 r6:000a201f r5:c3433428 r4:c3c93b40
[  570.208847] [<c043dd5c>] (__fput) from [<c043dfd0>] (____fput+0x18/0x1c)
[  570.208858]  r9:00000000 r8:c1356d6c r7:c3585d00 r6:c3586254 r5:c3c94f00 r4:c3c93b40
[  570.208874] [<c043dfb8>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
[... cut ...]
[  570.209001] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[  570.209010] Exception stack(0xc4463fb0 to 0xc4463ff8)
[  570.209021] 3fa0:                                     00000000 000ff740 00001000 00000000
[  570.209033] 3fc0: b6f13680 b6f11b1c b6f116d8 00000003 000005e8 b6f11b1c eeeeeeee 00000001
[  570.209044] 3fe0: 0000006c becb5300 b6e33acc b6e8e87c 60000010 00000000
[  570.209056]  r10:00000003 r9:c4462000 r8:c0200204 r7:00000003 r6:b6f116d8 r5:b6f11b1c
[  570.209064]  r4:b6f13680
[  570.209123] ---[ end trace 8ee602e009c3f72e ]---

alanbork, 2021-04-15:
binary copy of edid's uploaded to https://github.com/raspberrypi/linux/issues/3842

alanbork, 2021-04-16:
Add

core_freq=600
core_freq_min=600

to /boot/config.txt and selecting 4096x2160@60 works, no crashes (tested about 5x times).

alanbork, 2021-07-02:
This still causes a near hard-lock as of f29ab05611eef385d17675c77190df0b87a0d456, I thought that you had a system set up to dynamically overclock the CPU when the video mode required it?

error message the same as before when trying to load 4k60p(modetest -s 89:4096x2160-60):

[  592.431498] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out

only way to recover is to ssh in remotely and kill modetest.

popcornmix, 2023-01-24:
I believe this issue was resolved in https://github.com/raspberrypi/linux/pull/5038

closed 2023-01-24 by popcornmix