raspberrypi/linux #4274: SC16is752 SPI transfer timed out on 5.10

Aalococo, 2021-04-12:
Since I updated from Kernel 5.4.xx to 5.10.xx SC16is752 stopped working. I can find all the necessary devices in /dev/. 
Dmesg shows:
```
[    9.778466] spi0.8: ttySC1 at I/O 0x1 (irq = 199, base_baud = 921600) is a SC16IS752
```
When i start my software that uses the SC16 GPIOS i get the following in dmesg:
```
[  160.797821] sc16is7xx spi0.8: SPI transfer timed out
[  160.797851] spi_master spi0: failed to transfer one message from queue
[  163.037851] sc16is7xx spi0.8: SPI transfer timed out
[  163.037885] spi_master spi0: failed to transfer one message from queue
[  164.771783] irq 111: nobody cared (try booting with the "irqpoll" option)
```

my DT Overlay lookes like this:

```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";
        

	fragment@0 {
		target = <&spi0_cs_pins>;
		frag0: __overlay__ {
                        brcm,pins = <8 7 23 24 25 5 6 26 4>;
			brcm,function = <1>; /* out */
                        };
	};

        fragment@1 {
		target = <&spi0>;
		frag1: __overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
                        pinctrl-names = "default";
			pinctrl-0 = <&spi0_pins &spi0_cs_pins>;
			cs-gpios = <&gpio 8 1>, <&gpio 7 1>, <&gpio 23 1>, <&gpio 24 1>, <&gpio 25 1>, <&gpio 5 1>, <&gpio 6 1>, <&gpio 26 1>, <&gpio 4 1>;
			status = "okay";

		spidev@0{
                        compatible = "spidev";
                        reg = <0>;      /* CE0 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };

		spidev@1{
                        compatible = "spidev";
                        reg = <1>;      /* CE1 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };

		spidev@2{
		        compatible = "spidev";
                        reg = <2>;	/* CE2 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
	            	};

		spidev@3{
                        compatible = "spidev";
                        reg = <3>;	/* CE3 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
		        };

		spidev@4{
                        compatible = "spidev";
                        reg = <4>;      /* CE4 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };

		spidev@5{
                        compatible = "spidev";
                        reg = <5>;      /* CE5 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };

		spidev@6{
                        compatible = "spidev";
                        reg = <6>;      /* CE6 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };

		spidev@7{
                        compatible = "spidev";
                        reg = <7>;      /* CE7 */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        spi-max-frequency = <4000000>;
                        };
			
                sc16is752: sc16is752@8 {
                        compatible = "nxp,sc16is752";
                        reg = <8>; /* CE8  */
                        clocks = <&sc16is752_clk>;
                        interrupt-parent = <&gpio>;
                        interrupts = <17 2>; /* IRQ_TYPE_EDGE_FALLING */
                        gpio-controller;
                        #gpio-cells = <2>;
                        spi-max-frequency = <4000000>;
                        };
		};
	};

        

        fragment@4 {
		target-path = "/";
		__overlay__ {
			sc16is752_clk: sc16is752_spi0_8_clk {
				compatible = "fixed-clock";
				#clock-cells = <0>;
				clock-frequency = <14745600>;
			};
		};
	};

    	
	__overrides__ {
                int_pin = <&sc16is752>,"interrupts:0";
                xtal = <&sc16is752_clk>,"clock-frequency:0";
                cs0_pin  = <&frag0>,"brcm,pins:0",
			   <&frag1>,"cs-gpios:4";
		cs1_pin  = <&frag0>,"brcm,pins:4",
			   <&frag1>,"cs-gpios:16";
       
	};

	
};
```

pelwell, 2021-04-12:
What does "cat /proc/interrupts" show?

Aalococo, 2021-04-12:
111:     500000          0          0          0  ARMCTRL-level  86 Edge      3f204000.spi

pelwell, 2021-04-12:
And what's at 199?

Aalococo, 2021-04-12:
sry thats this here:
199:          0          0          0          0  pinctrl-bcm2835  17 Level     spi0.8

pelwell, 2021-04-12:
Have you tried (just as a test) with the SC16IS752 on a different CS line? Or with fewer CSs?

Aalococo, 2021-04-12:
When im using only sc16 without the other spi lanes it seems to work. With previous kernel 5.4 I had everything working together.

Aalococo, 2021-04-12:
So im just toggling a SC16 gpio. That works as long is i dont run my software. My software is using the other spi lanes aswell so it seems to me that accessing multiple spis somehow leads to the problem

pelwell, 2021-04-12:
What are you doing with the other CS lines, and at what clock rates? If something else is hogging the SPI interface then its easy to see how that could cause problems for a device.

Aalococo, 2021-04-12:
I have various chips connected to them. Spi speed is set to 500000.  The software is running without problems since a few months on kernel 5.4. 

Since the update It just crashed with the following dmesg log. I then updated the Overlay to what i posted above in this issue. Maybe the old Errors + Overlay can give a hint?

```[   91.034276] rcu: INFO: rcu_sched self-detected stall on CPU
[   91.034305] rcu: 	3-...!: (2099 ticks this GP) idle=daa/1/0x40000002 softirq=3628/3628 fqs=197 
[   91.034315] 	(t=2100 jiffies g=2409 q=306)
[   91.034327] rcu: rcu_sched kthread starved for 1703 jiffies! g2409 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=3
[   91.034334] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[   91.034339] rcu: RCU grace-period kthread stack dump:
[   91.034347] task:rcu_sched       state:R  running task     stack:    0 pid:   12 ppid:     2 flags:0x00000000
[   91.034367] Backtrace: 
[   91.034397] [<809ed8f0>] (__schedule) from [<809ee2c8>] (schedule+0x68/0xe4)
[   91.034413]  r10:00000006 r9:b6b38540 r8:b6b38540 r7:80e05008 r6:8155dee4 r5:81552e80
[   91.034421]  r4:ffffe000
[   91.034437] [<809ee260>] (schedule) from [<809f2634>] (schedule_timeout+0x1d0/0x384)
[   91.034446]  r5:80e03d00 r4:ffffa7ba
[   91.034464] [<809f2464>] (schedule_timeout) from [<8019948c>] (rcu_gp_kthread+0x4d0/0xb98)
[   91.034478]  r9:00000000 r8:00000001 r7:ffffe000 r6:80e03d00 r5:00000000 r4:80e761c0
[   91.034493] [<80198fbc>] (rcu_gp_kthread) from [<801437b8>] (kthread+0x170/0x174)
[   91.034501]  r7:8155c000
[   91.034517] [<80143648>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
[   91.034525] Exception stack(0x8155dfb0 to 0x8155dff8)
[   91.034536] dfa0:                                     00000000 00000000 00000000 00000000
[   91.034549] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   91.034560] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   91.034573]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143648
[   91.034580]  r4:814dd300
[   91.034590] NMI backtrace for cpu 3
[   91.034603] CPU: 3 PID: 1292 Comm:LBaseSoft Tainted: G         C        5.10.17-v7+ #1403
[   91.034609] Hardware name: BCM2835
[   91.034615] Backtrace: 
[   91.034637] [<809e477c>] (dump_backtrace) from [<809e4b0c>] (show_stack+0x20/0x24)
[   91.034648]  r7:ffffffff r6:00000000 r5:60000193 r4:80ee5d54
[   91.034662] [<809e4aec>] (show_stack) from [<809e8cb8>] (dump_stack+0xcc/0xf8)
[   91.034677] [<809e8bec>] (dump_stack) from [<8065a17c>] (nmi_cpu_backtrace+0xc8/0xf4)
[   91.034691]  r10:80000193 r9:80bff688 r8:80a02058 r7:80a02068 r6:00000003 r5:00000000
[   91.034699]  r4:00000003 r3:36cd1669
[   91.034713] [<8065a0b4>] (nmi_cpu_backtrace) from [<8065a2d0>] (nmi_trigger_cpumask_backtrace+0x128/0x140)
[   91.034722]  r5:80e05a9c r4:8010e024
[   91.034738] [<8065a1a8>] (nmi_trigger_cpumask_backtrace) from [<8010ef10>] (arch_trigger_cpumask_backtrace+0x20/0x24)
[   91.034750]  r7:80e761c0 r6:80e05060 r5:80e05104 r4:00000003
[   91.034764] [<8010eef0>] (arch_trigger_cpumask_backtrace) from [<809e671c>] (rcu_dump_cpu_stacks+0x10c/0x144)
[   91.034778] [<809e6610>] (rcu_dump_cpu_stacks) from [<8019c480>] (rcu_sched_clock_irq+0x7c0/0xa38)
[   91.034791]  r10:80e03d00 r9:35dd6000 r8:00000000 r7:80d92240 r6:80d92240 r5:b6b68240
[   91.034798]  r4:80e761c0
[   91.034814] [<8019bcc0>] (rcu_sched_clock_irq) from [<801a7704>] (update_process_times+0x70/0x9c)
[   91.034828]  r10:801bb460 r9:b6b61640 r8:ffffe000 r7:00000015 r6:7d4dacc4 r5:00000000
[   91.034835]  r4:ffffe000
[   91.034849] [<801a7694>] (update_process_times) from [<801bad4c>] (tick_sched_handle+0x64/0x70)
[   91.034861]  r7:00000015 r6:31cf9e2d r5:87f3d9a0 r4:b6b61978
[   91.034873] [<801bace8>] (tick_sched_handle) from [<801bb4bc>] (tick_sched_timer+0x5c/0xb8)
[   91.034887] [<801bb460>] (tick_sched_timer) from [<801a8220>] (__hrtimer_run_queues+0xe8/0x384)
[   91.034898]  r7:b6b61680 r6:80e05058 r5:b6b61640 r4:b6b61978
[   91.034913] [<801a8138>] (__hrtimer_run_queues) from [<801a90f8>] (hrtimer_interrupt+0x130/0x2c0)
[   91.034927]  r10:b6b61790 r9:b6b617c0 r8:ffffffff r7:7fffffff r6:00000003 r5:20000193
[   91.034934]  r4:b6b61640
[   91.034953] [<801a8fc8>] (hrtimer_interrupt) from [<80821d7c>] (arch_timer_handler_phys+0x38/0x40)
[   91.034966]  r10:7f0c8000 r9:87f3c000 r8:8147e000 r7:000000c3 r6:814c19c0 r5:80ee60e8
[   91.034973]  r4:814cef00
[   91.034990] [<80821d44>] (arch_timer_handler_phys) from [<8018b8d0>] (handle_percpu_devid_irq+0x8c/0x21c)
[   91.035006] [<8018b844>] (handle_percpu_devid_irq) from [<80184b44>] (generic_handle_irq+0x44/0x54)
[   91.035017]  r7:00000001 r6:00000000 r5:00000000 r4:80d90d10
[   91.035032] [<80184b00>] (generic_handle_irq) from [<801852a4>] (__handle_domain_irq+0x6c/0xc4)
[   91.035048] [<80185238>] (__handle_domain_irq) from [<801012c8>] (bcm2836_arm_irqchip_handle_irq+0x60/0x64)
[   91.035062]  r9:87f3c000 r8:8374e400 r7:87f3d9d4 r6:ffffffff r5:a0000013 r4:7f0c5d04
[   91.035076] [<80101268>] (bcm2836_arm_irqchip_handle_irq) from [<80100abc>] (__irq_svc+0x5c/0x7c)
[   91.035083] Exception stack(0x87f3d9a0 to 0x87f3d9e8)
[   91.035097] d9a0: 0000000d 000002f6 9461b6c0 00000055 835f1b80 87f3db9c 80e05008 835f1800
[   91.035110] d9c0: 8374e400 ed08ae82 7f0c8000 87f3da3c 85df6c80 87f3d9f0 7f0c595c 7f0c5d04
[... 168 lines cut ...]
                                };
    	                };
	};

        fragment@2 {
		target-path = "/";
		__overlay__ {
			sc16is752_clk: sc16is752_spi0_0_clk {
				compatible = "fixed-clock";
				#clock-cells = <0>;
				clock-frequency = <14745600>;
			};
		};
	};

    	
	__overrides__ {
        int_pin = <&sc16is752>,"interrupts:0";
        xtal = <&sc16is752_clk>,"clock-frequency:0";
        
	};

	
};
```

pelwell, 2021-04-20:
I have some insight on this problem now. The BCM2835/2711 SPI driver doesn't cope with more than 4 CS lines; exceeding that number will not be rejected but the driver may start to malfunction. There is an ongoing discussion in this Pull Request (https://github.com/raspberrypi/linux/pull/4281) and in an email exchange with the upstream devs.

Aalococo, 2021-04-20:
thanks alot for that info. Im following (#4281) now

Aalococo, 2021-05-17:
as far is a can tell the spi limit to 24 chip selects already got merged into 5.10 ? im still having this problem ?