raspberrypi/linux #4287: rpi-5.12.y: BUG: KASAN: slab-out-of-bounds in vc4_fkms_crtc_duplicate_state+0x70/0xa0 [vc4]

yetanotheraccount, 2021-04-17:
Greetings,

[dmesg.txt](https://github.com/raspberrypi/linux/files/6328722/dmesg.txt)

Rev info is the dmesg output itself.  To reproduce, enable KASAN, build/boot on any pi4b should do it.

yetanotheraccount, 2021-04-20:
[dmesg2.txt](https://github.com/raspberrypi/linux/files/6343206/dmesg2.txt)

Silencing $subject led to a second KASAN gripe, see dmesg2.txt.

Applying the below silences both, and the kernel now runs KASAN gripe free.  Note that this comes with no claim of correctness, I'm only showing it at all because might it _might_ be useful.

---
 drivers/gpu/drm/vc4/vc4_firmware_kms.c |   13 ++++++++++---
 1 file changed, 10 insertions(+), 3 deletions(-)

--- a/drivers/gpu/drm/vc4/vc4_firmware_kms.c
+++ b/drivers/gpu/drm/vc4/vc4_firmware_kms.c
@@ -1241,12 +1241,19 @@ vc4_fkms_crtc_duplicate_state(struct drm
 static void
 vc4_fkms_crtc_reset(struct drm_crtc *crtc)
 {
+       struct vc4_crtc_state *vc4_state;
+
        if (crtc->state)
                __drm_atomic_helper_crtc_destroy_state(crtc->state);
 
-       crtc->state = kzalloc(sizeof(*crtc->state), GFP_KERNEL);
-       if (crtc->state)
+       kfree(to_vc4_crtc_state(crtc->state));
+       crtc->state = NULL;
+
+       vc4_state = kzalloc(sizeof(*vc4_state), GFP_KERNEL);
+       if (vc4_state) {
+               __drm_atomic_helper_crtc_reset(crtc, &vc4_state->base);
                crtc->state->crtc = crtc;
+       }
 }
 
 static int vc4_fkms_enable_vblank(struct drm_crtc *crtc)
@@ -1482,7 +1489,7 @@ vc4_connector_duplicate_state(struct drm
 {
        struct vc4_fkms_connector_state *state;
 
-       state = kmemdup(connector->state, sizeof(*state), GFP_KERNEL);
+       state = kzalloc(sizeof(*state), GFP_KERNEL);
        if (!state)
                return NULL;

yetanotheraccount, 2021-04-20:
[xx.txt](https://github.com/raspberrypi/linux/files/6343283/xx.txt)

Hohum, pasting inline seems to have made a mess, attaching.

yetanotheraccount, 2021-04-20:
Hasta la vista

pelwell, 2021-04-20:
Use the triple-backtick escape (` ``` `) for blocks of code, and triple-backtick-diff (`  ```diff`) for diffs.

6by9, 2021-04-20:
The second fragment looks like it does the wrong thing - it fails to copy any of the specific members of ```struct vc4_fkms_connector_state```, although ```__drm_atomic_helper_connector_duplicate_state``` should deal with the rest.

The first one is making me scratch my head a bit as to what it's trying to do. Was this copied from somewhere else?

Comparing against vc4_crtc.c, a correction to use ```drm_atomic_helper_crtc_destroy_state``` instead of ```__drm_atomic_helper_crtc_destroy_state``` is in order - the latter doesn't actually free the state so we have a memory leak.
Contrary, with your hunk there is no need for the ```crtc->state->crtc = crtc;``` as it's done by ```__drm_atomic_helper_crtc_state_reset``` called from ```__drm_atomic_helper_crtc_reset```

yetanotheraccount, 2021-04-20:
The first one is making me scratch my head a bit as to what it's trying to do. Was this copied from somewhere else?

More or less, after changing the allocation, I rummaged a bit and decided a little monkey see monkey do was in order.  (eep eep:)

yetanotheraccount, 2021-04-21:
Aha, rummaging a bit more, I think I see what's causing the connector_reset() KASAN gripe, namely this:

git@homer:..git/raspberrypi-kernel> git grep vc4_hdmi_connector_reset
drivers/gpu/drm/vc4/vc4_firmware_kms.c:static void vc4_hdmi_connector_reset(struct drm_connector *connector)
drivers/gpu/drm/vc4/vc4_firmware_kms.c:extern void vc4_hdmi_connector_reset(struct drm_connector *connector);
drivers/gpu/drm/vc4/vc4_firmware_kms.c: .reset = vc4_hdmi_connector_reset,
drivers/gpu/drm/vc4/vc4_hdmi.c:static void vc4_hdmi_connector_reset(struct drm_connector *connector)
drivers/gpu/drm/vc4/vc4_hdmi.c: .reset = vc4_hdmi_connector_reset,

The one in vc4_hdmi.c seems to do the right thing, but has not been called (yet?).  The allocation was done by the one in vc4_firmware_kms.c...

[   18.815613] Allocated by task 175:
[   18.815630]  kasan_save_stack+0x28/0x60
[   18.815670]  __kasan_kmalloc+0x88/0xb0
[   18.815703]  kmem_cache_alloc_trace+0x1f8/0x3e0
[   18.815738]  drm_atomic_helper_connector_reset+0x48/0xa0 [drm_kms_helper]
[   18.816405]  vc4_hdmi_connector_reset+0x24/0x40 [vc4]
[   18.816820]  drm_mode_config_reset+0x214/0x220 [drm]
[   18.818162]  vc4_kms_load+0x2b8/0x380 [vc4]

..and drm_atomic_helper_connector_reset() allocated a struct drm_connector_state. 

Methinks perhaps the kmemdup() belongs in the vc4_hdmi.c .atomic_duplicate_state function, instead of the vc4_firmware_kms.c function. 

--- a/drivers/gpu/drm/vc4/vc4_firmware_kms.c
+++ b/drivers/gpu/drm/vc4/vc4_firmware_kms.c
@@ -1482,7 +1489,7 @@ vc4_connector_duplicate_state(struct drm
 {
        struct vc4_fkms_connector_state *state;
 
-       state = kmemdup(connector->state, sizeof(*state), GFP_KERNEL);
+       state = kzalloc(sizeof(*state), GFP_KERNEL);
        if (!state)
                return NULL;
 
--- a/drivers/gpu/drm/vc4/vc4_hdmi.c
+++ b/drivers/gpu/drm/vc4/vc4_hdmi.c
@@ -596,7 +596,7 @@ vc4_hdmi_connector_duplicate_state(struc
        struct vc4_hdmi_connector_state *vc4_state = conn_state_to_vc4_hdmi_conn_state(conn_state);
        struct vc4_hdmi_connector_state *new_state;
 
-       new_state = kzalloc(sizeof(*new_state), GFP_KERNEL);
+       new_state = kmemdup(connector->state, sizeof(*new_state), GFP_KERNEL);
        if (!new_state)
                return NULL;

KASAN is happy with than, as is box, but then box hasn't noticed either issue.

yetanotheraccount, 2021-04-21:
Ok, vc4_hdmi.c:: .reset/.atomic_duplicate_state are never called in my box, but they do appear to be compatible with each other, whereas the vc4_firmware_kms.c variants are definitely not.  With that, I'll get out of your hair.  For my purposes, what I've done locally silences what I've reported, leaving box+KASAN ready should something else pop up before these are properly fixed.

yetanotheraccount, 2021-04-22:
Sigh, "ignore 'em, they'll get fixed up properly in good time" didn't work.  I believe the below contains the proper fixes.  Hopefully, I can now stop caring about these little toothless buglets and really really  walk away ;-)

``---
 drivers/gpu/drm/vc4/vc4_firmware_kms.c |   35 +++++++++++++++++++++++++++------
 1 file changed, 29 insertions(+), 6 deletions(-)

--- a/drivers/gpu/drm/vc4/vc4_firmware_kms.c
+++ b/drivers/gpu/drm/vc4/vc4_firmware_kms.c
@@ -1241,12 +1241,19 @@ vc4_fkms_crtc_duplicate_state(struct drm
 static void
 vc4_fkms_crtc_reset(struct drm_crtc *crtc)
 {
+       struct vc4_crtc_state *vc4_state;
+
        if (crtc->state)
                __drm_atomic_helper_crtc_destroy_state(crtc->state);
 
-       crtc->state = kzalloc(sizeof(*crtc->state), GFP_KERNEL);
-       if (crtc->state)
-               crtc->state->crtc = crtc;
+       kfree(to_vc4_crtc_state(crtc->state));
+       crtc->state = NULL;
+
+       vc4_state = kzalloc(sizeof(*vc4_state), GFP_KERNEL);
+       if (!vc4_state)
+               return;
+
+       __drm_atomic_helper_crtc_reset(crtc, &vc4_state->base);
 }
 
 static int vc4_fkms_enable_vblank(struct drm_crtc *crtc)
@@ -1480,12 +1487,16 @@ static void vc4_fkms_connector_destroy(s
 struct drm_connector_state *
 vc4_connector_duplicate_state(struct drm_connector *connector)
 {
-       struct vc4_fkms_connector_state *state;
+       struct vc4_fkms_connector_state *state, *old_state;
 
-       state = kmemdup(connector->state, sizeof(*state), GFP_KERNEL);
+       state = kzalloc(sizeof(*state), GFP_KERNEL);
        if (!state)
                return NULL;
 
+       /* copy the sole digital connector specific element first */
+       old_state = to_vc4_fkms_connector_state(connector->state);
+       state->broadcast_rgb = old_state->broadcast_rgb;
+
        __drm_atomic_helper_connector_duplicate_state(connector, &state->base);
        return &state->base;
 }
@@ -1551,7 +1562,19 @@ int vc4_connector_atomic_set_property(st
 
 static void vc4_hdmi_connector_reset(struct drm_connector *connector)
 {
-       drm_atomic_helper_connector_reset(connector);
+       struct vc4_fkms_connector_state *conn_state;
+
+       if (connector->state)
+               __drm_atomic_helper_connector_destroy_state(connector->state);
+
+       kfree(connector->state);
+       connector->state = NULL;
+
+       conn_state = kzalloc(sizeof(*conn_state), GFP_KERNEL);
+       if (!conn_state)
+               return;
+
+       __drm_atomic_helper_connector_reset(connector, &conn_state->base);
        drm_atomic_helper_connector_tv_reset(connector);
 }
``

yetanotheraccount, 2021-04-22:
[xx.txt](https://github.com/raspberrypi/linux/files/6355563/xx.txt)

Either the user interface here is hottible, or I'm a hopeless case at web based crap, likely the later :) Poking 'Insert code' widget helped not at all, pasting code made another mess.

Ciao

pelwell, 2021-04-22:
One needs to be able to count to 3 (backticks on line of their own).