raspberrypi/linux #4290: Missing spi-rtc-ds3234-overlay.dts ?

Exaga, 2021-04-17:
While searching in the Linux kernel source for a DTB overlay, or DT source to compile, for installing a DS3234-based RTC on a Raspberry Pi (using the SPI interface) I soon realised this RTC wasn't currently supported. So I created one.

May I suggesting that it could become a standard overlay?

spi-rtc-ds3234-overlay.dts code:

`/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			ds3234@0 {
				compatible = "ds3234";
				spi-max-frequency = <5000000>;
				reg = <0>;
			};
		};
	};
};
`

pelwell, 2021-04-19:
Rather than adding a dedicated overlay, I'm proposing to extend the existing spi-rtc overlay. See this pull request: https://github.com/raspberrypi/linux/pull/4292

Exaga, 2021-04-19:
> 
> 
> Rather than adding a dedicated overlay, I'm proposing to extend the existing spi-rtc overlay. See this pull request: #4292

Sounds good. However a DS3234 DT overlay is implemented would be a step forward for sure. Thanks.

pelwell, 2021-04-21:
#4292 has been merged, and the updated overlay will be in future kernel releases.

Exaga, 2021-04-24:
> 
> 
> #4292 has been merged, and the updated overlay will be in future kernel releases.

Thanks PhilE.

Infinitely easier for all users when the DS3234 is supported by you guys in the Device Tree.

closed 2021-04-24 by Exaga