raspberrypi/linux #4299: RPi 2 doesn't boot on linux-raspberrypi 5.10.31-2 when connected with HDMI

nightBulb, 2021-04-20:
**Describe the bug**
Raspberry Pi 2 fails to boot when using linux-raspberrypi **5.10.31-2** with HDMI connected.

**To reproduce**
Reboot `R Pi 2` after update to linux-raspberrypi 5.10.31-2 
with the the RPi 2 connected via HDMI to a screen

**Expected behaviour**
System to boot

**Actual behaviour**
Raspberry Pi 2 fails to boot when using linux-raspberrypi `5.10.31-2` with RPi 2 connected to a screen with HDMI.

**System**

* Which model of Raspberry Pi? 
`R Pi 2`

* Which OS and version (`cat /etc/rpi-issue`)?
```
Arch Linux ARM (latest)
```
 (Don't have rpi-issue file)

* Which firmware version (`vcgencmd version`)?
```
Apr 12 2021 15:54:06 
Copyright (c) 2012 Broadcom
version 0227a0840ee8c94294cf3b2c4b2d42031a58f05a (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
 ```
Linux rpi 5.10.31-1-ARCH #1 SMP Mon Apr 19 13:33:01 UTC 2021 armv7l GNU/Linux

```

**Logs**
(Unable to get dmesg when boot fails)

**Additional context**
```
Linux 5.10.31-1 boots successfully with and without HDMI connected.

Linux 5.10.31-2 boot fails when HDMI screen is connected.
Linux 5.10.31-2 boot succeeds when HDMI is **disconnected**.
```

nightBulb, 2021-04-20:
dmesg output:

1. When boot succeeds with bad kernel ( DISCONNECTED HDMI ) linux 5.10.31-2
[dmesg-2021-04-21-bad-kernel-log.txt](https://github.com/raspberrypi/linux/files/6347124/dmesg-2021-04-21-bad-kernel-log.txt)

2. When boot succeeds with good kernel Linux ( CONNECTED HDMI ) 5.10.31-1
[dmesg-2021-04-21-good-kernel-log.txt](https://github.com/raspberrypi/linux/files/6347127/dmesg-2021-04-21-good-kernel-log.txt)

3. Couldn't get the failure ( linux 5.10.31-2, HDMI Connected)  `dmesg` output as no login prompt

<hr />
Colored variants of above logs

```
$ dmesg --color=always
```

[dmesg-2021-04-21-bad-kernel-log-colored.txt](https://github.com/raspberrypi/linux/files/6347126/dmesg-2021-04-21-bad-kernel-log-colored.txt)

[dmesg-2021-04-21-good-kernel-log-colored.txt](https://github.com/raspberrypi/linux/files/6347128/dmesg-2021-04-21-good-kernel-log-colored.txt)

nightBulb, 2021-04-21:
This is the last thing that happens when booting with Linux 5.10.31-2 
(bad kernel) connected via HDMI to the screen.

[image]

henkv1, 2021-04-21:
Same issue here on RPi3.
The system crashes when kms refreshes the screen. The issue does not occur when fkms is used. So this is probably a bug in the new vc4 code.

pelwell, 2021-04-21:
Can one of you provide a dmesg log for the good kernel in the disconnected state?

popcornmix, 2021-04-21:
I can reproduce this (on Pi3+). No panic out of uart when it hangs. Presumably related to clock changes.

popcornmix, 2021-04-21:
Yes, 9e081a7bf69c75d5e9cb05add07eaf4cb4ee4dc9 seems to be first commit which hangs. It's okay on Pi4.

nightBulb, 2021-04-21:
@pelwell This is the `dmesg` for good kernel (Linux 5.10.31-1 ) in HDMI disconnected state
[dmesg-2021-04-21-3-good-kernel-disconnected.txt](https://github.com/raspberrypi/linux/files/6350397/dmesg-2021-04-21-3-good-kernel-disconnected.txt)

<hr />

Same dmesg with `dmesg --color=always`

[dmesg-2021-04-21-3-good-kernel-disconnected-colored.txt](https://github.com/raspberrypi/linux/files/6350398/dmesg-2021-04-21-3-good-kernel-disconnected-colored.txt)

pelwell, 2021-04-21:
Thanks. Now that popcornmix has reproduced the problem we shouldn't need any more info.

popcornmix, 2021-04-21:
I've pushed a new kernel to rpi-update that should hopefully resolve this. Can you test?

nightBulb, 2021-04-21:
@popcornmix Problem still persists.

Unable to complete boot with linux 5.10.31-3 while HDMI is connected.

Boot only succeeds with it disconnected.

popcornmix, 2021-04-21:
With hdmi disconnected what does `uname -a` report?

nightBulb, 2021-04-21:
@popcornmix  Linux rpi 5.10.31-3-ARCH #1 SMP Wed Apr 21 14:04:34 UTC 2021 armv7l GNU/Linux

popcornmix, 2021-04-21:
You aren't using rpi-update kernel, so I'm unsure what commits are in there.
From the timestamp of your kernel, it doesn't look like you'll have the commit that fixes it.
```
commit 01a6d727b407ed064bdd72e3a6611101102a2228
Author: Dom Cobley <popcornmix@gmail.com>
Date:   Wed Apr 21 15:15:42 2021 +0100

    vc4/drm: hdmi: Handle case when bvb clock is null
    
    Pi2/3 have no bvb clock but want the other clocks to remain enabled here
    
    See: https://github.com/raspberrypi/linux/issues/4299
    Signed-off-by: Dom Cobley <popcornmix@gmail.com>
```

popcornmix, 2021-04-21:
This is booting with hdmi output on a pi3 for me:
```
pi@b3plus:~ $ uname -a
Linux domnfs 5.10.31-v7+ #62 SMP Wed Apr 21 15:41:57 BST 2021 armv7l GNU/Linux
```

nightBulb, 2021-04-21:
@graysky2 could you please update the linux kernel to commit 01a6d727b407ed064bdd72e3a6611101102a2228

graysky2, 2021-04-21:
@nightBulb - Building https://github.com/raspberrypi/linux/commit/203f70487548714de96bd3f530c8f8a6d558971e and will test/push if all is well.

nightBulb, 2021-04-21:
@graysky2 Thanks  :smile: :+1:

graysky2, 2021-04-21:
https://github.com/archlinuxarm/PKGBUILDs/commit/c9ee7290871028d2135656ae9ce4e06ada0d0e49

nightBulb, 2021-04-21:
I just tested `Linux rpi 5.10.31-4-ARCH #1 SMP Wed Apr 21 17:54:17 UTC 2021 armv7l GNU/Linux`

and can confirm the problem is fixed, and the system is booting with HDMI connected. :smile:  :+1: 

Thanks :+1: :+1: and Kudos :1st_place_medal:   everyone. 

I can now recommend we close this bug.

closed 2021-04-21 by nightBulb