raspberrypi/linux #4301: By default, the Wifi interface seems to not working on Raspberry 4

yangsongx, 2021-04-21:
I have Raspberry Pi 4,  5.10 kernel version,

but when I *ifconfig -a*, I didn't find the wlan0 interface, should I need to config some one in the kernel defconfig file?

pelwell, 2021-04-21:
If it is Raspberry Pi OS image you will need to set a WiFi country code before you can use the interface - it is blocked with rfkill until then. `sudo raspi-config` can set the country code (also available as the Configuration utility in the GUI), or you can manually add `country=GB` (for example) to `/etc/wpa_supplicant/wpa_supplicant.conf`.

yutayu, 2021-04-30:
https://raspberrypi.stackexchange.com/questions/124306/cannot-access-headless-raspberry-pi-due-to-wifi

me, too. 
suddnly , wifi stalls.
an0: STA 00:ed:b8:xx:xx:xx IEEE 802.11: disassociated

yutayu, 2021-05-01:
[   23.039570] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   23.039729] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   23.040629] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd

yutayu, 2021-05-01:
https://lists.infradead.org/pipermail/linux-rpi-kernel/2021-February/007994.html

yutayu, 2021-05-01:
Excuse me , I disable dnscrypt-proxy and reboot pi, it fixed.
Thanks.