raspberrypi/linux #4304: New I2S module in order to use a generic I2S microphone

linuxchris12, 2021-04-22:
I found that there is no way to connect a generic I2S device to the Raspberry Pi without re-compiling the kernel.

Situation: I2S is a standardized bus which is defined by a clock rate, frame-size and some other parameters, very similar to I2C. It can be used to transmit or receive digital audio from the Raspberry Pi. Unlike I2C, there currently seems to be no way to establish a new I2S connection and route a I2S signal to the ALSA driver (aplay, arecord) from a generic source (there seem to be only strongly hardware specific modules right now).

Problem: Some hardware is not supported in the current kernel, even though it would only require to set the I2S master/secondary status, clock-rate and frame settings correctly to stream the I2S data and use it with aplay/arecord. Creating a new hardware specific module would be kind of over-kill for this. And asking users to compile their own kernel modules can be tricky.

Solution: We can create a generic I2S module, which can be adjusted through device tree overlay parameters. This way, the I2S can be used in a generic way by simply defining the bus parameters within the /boot/config.txt file.
I will attach a first, minimal example.
This allows to setup a new I2S device by calling something similar to:
```
sudo dtoverlay generic-i2s-mic.dtbo snd_soc_daifmt_i2s=true snd_soc_daifmt_nb_nf=true snd_soc_daifmt_cbm_cfm=true
```

linuxchris12, 2021-04-22:
First try can be found here:
https://github.com/linuxchris12/linux/commit/5c82f2660155bcb04fab7f2c55d9338d5871b012

pelwell, 2021-04-22:
Is the simple-audio-card driver not sufficient?

linuxchris12, 2021-04-23:
That could work - but I would still need to create the right overlay file .dtbo. So I am not sure how that would help with respect to the objective to get this working without having to change and build the kernel. 

I could not find a way how to set the parameters of the simple-audio-card directly. But that could also just be my ignorance?

pelwell, 2021-04-23:
The simple-audio-card module is already included with Raspberry Pi kernels and used by a few audio overlays - search arch/arm/boot/dts/overlays for "simple".

linuxchris12, 2021-04-26:
Thanks @pelwell for your great support. The issue is solved for me.

I have created an overlay file which is based on the dmic codec and the simple-audio-card module.
It still needs compiling of the kernel, but the created .dtbo file is pretty independent from the kernel code and should work for a wide range of kernel versions, so I can distribute this file directly to any users.

For anyone interested, here is the overlay:

```
// Definitions for a generic I2S MIC
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&i2s>;
        __overlay__ {
            status = "okay";
        };
    };

    fragment@1 {
        target-path = "/";
        __overlay__ {
                dmic_codec: dmic-codec {
                #sound-dai-cells = <0>;
                compatible = "dmic-codec";
                status = "okay";
            };
        };
    };

    fragment@2 {
        target = <&sound>;
            sound_overlay: __overlay__ {
            compatible = "simple-audio-card";
            simple-audio-card,format = "i2s";
            simple-audio-card,name = "dabboard";
            simple-audio-card,bitclock-master = <&dailink0_slave>;
            simple-audio-card,frame-master = <&dailink0_slave>;
            simple-audio-card,widgets =
                    "Microphone", "Microphone Jack";
            status = "okay";
            simple-audio-card,cpu {
                sound-dai = <&i2s>;
            };
            dailink0_slave: simple-audio-card,codec {
                sound-dai = <&dmic_codec>;
            };
        };
    };

    __overrides__ {
        card-name = <&sound_overlay>,"simple-audio-card,name";
    };
};
```

closed 2021-04-26 by linuxchris12

pelwell, 2021-04-26:
If you give it a name and a few words of description I'll add it to the others in this repo. And a Signed-off-by: line for the commit, if you want.

linuxchris12, 2021-04-27:
Yes sure, that would be great!

I have these lines as description:

`
This is a simple overlay based on the simple-audio-card and the dmic 
codec. It has the speciality that it is configured to use the codec 
as a master I2S device. It works for example with the Si468x DAB 
receiver on the uGreen DABBoard.
`

and created a PR:
https://github.com/raspberrypi/linux/pull/4310