raspberrypi/linux #4306: kms+pi4 without x11: 1-pixel wide bar line at left edge of screen flickers on and off once per flip

alanbork, 2021-04-22:
with the 2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91 firmware, I'm occasionally seeing a 1-pixel wide line running the height of the screen, which flickers on and off with each call to swap buffers (off might = black, but I haven't tried on a white background to see if it's actually being drawn on those frames). Line is on the left edge of the screen.

if you pause swap buffer it stops toggling on and off.

this only happens about 1 out every 10 times I switch to 3840x2160 (from 1080p). If it does show up, it's there until I switch the video mode back. because it's intermittent, I haven't tested exhaustively to determine if it ever happens outside 3840, but I've yet to see it elsewhere. same binary does not exhibit this issue prior to 2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91.

popcornmix, 2021-04-23:
Duplicate of https://github.com/raspberrypi/linux/issues/4187

It can affect all hdmi modes and has always been present on kms, but it's timing related so sometime appears to occur more or less frequently.

alanbork, 2021-05-11:
well it seems I don't have permission to reopen this one, but it's not a duplicate of #4187, as this line is not a wrap-around from the other side of the screen. There's no single stripe of white pixels that runs the entire height of the screen, yet that's what I see.  Actually the behavior is more chaotic than that - sometimes what I see looks like an inversion of the frame buffer from before (ie 1-lastbuffer). And it's not always on the left, at least once I saw it show up on the right-most side.  I only see it rarely so I don't have a video of this to demonstrate.  In any case it's kms related as I've never seen anything like this with fkms or the dispmanx firmware, and it still seems to be mostly an issue at 3840x2160.