raspberrypi/linux #4307: new kms driver doesn't detect 480i

alanbork, 2021-04-23:
as of 2ac4de4eaac5c1d1b25acec4a5e0a9fdb16f0c91, the kms driver does not detect 480i.

kms: modetest | grep 480i
[nothing]
fkms: modetest | grep 480i
  #31 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver
  #32 720x480i 29.97 720 739 801 858 480 488 494 525 13500 flags: nhsync, nvsync, interlace, dblclk; type: driver

there's a bug in modetest that shows the refresh rate as half if the mode is interlaced, those really are 60hz/ntsc.

perhaps relatedly, kms fails if I force 480i by using the same modeline detected by fkms:

720x480i-60.00hz(60): 13.514000 720 739 801 858 480 488 494 525 -hsync -vsync interlace dblclk

it initializes kms/drm without errors but swapbuffer returns in .28ms and nothing is shown on screen. 

it works if I instead I generate a modeline using cvt:

13.50  720 744 808 896  480 483 493 512 interlace -hsync +vsync

so maybe the issue is dblclk?

alanbork, 2021-04-23:
seems this exact problem was fixed for fkms a while ago: #3725.

6by9, 2021-04-28:
DRM works in frames not fields. 480i is 30 frames (60 fields) per second, and 33ms would therefore be correct for vsync timings (I'm assuming .28ms was a typo for 28ms).
You can not update the display between fields.

Your listed timing is totally different from the porch timings of the one parsed from the EDID, so I'm not sure of the relevance.

As per the [docs](https://www.raspberrypi.org/documentation/configuration/config-txt/pi4-hdmi.md), due to the 2-pixels/clock architecture, odd values in horizontal timings can not be supported.
```720x480i 30.00 720 739 801 858 480 488 494 525 13514``` and ```720x480i 29.97 720 739 801 858 480 488 494 525 13500``` will have been pruned due to this.  See https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L1444

Fkms has a [specific conditional to allow such modes with DBLCLK](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L1093).
Looking at [vc4_crtc.c](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_crtc.c#L340), all the horizontal pixel valve timings get doubled in DBLCLK mode, so there is no way they can be odd, so those modes can be supported. The filtering out is erroneous then.

alanbork, 2021-04-28:
> DRM works in frames not fields. 480i is 30 frames (60 fields) per second, and 33ms would therefore be correct for vsync timings (I'm assuming .28ms was a typo for 28ms).
> You can not update the display between fields.

not a typo, it's returning after less than a millisecond.  Note these are only inferred timings as they are based on when the swapbuffer call returns.  Are you saying that 480i60 can only be updated at 30hz?

>Your listed timing is totally different from the porch timings of the one parsed from the EDID, so I'm not sure of the relevance.

you must be talking about the timing generated with the cvt command line, which does work even with the kms driver release I tested under.  my point of including that was that kms interlaced at this resolution works fine if you bypass the edid.  To be clear, I'll restate:

mode as revealed under fkms via modetest:

#31 720x480i 30.00 720 739 801 858 480 488 494 525 13514 flags: nhsync, nvsync, interlace, dblclk; type: driver

is the same as this xorg compatible modeline:

720x480i-60.00hz(60): 13.514000 720 739 801 858 480 488 494 525 -hsync -vsync interlace dblclk

which works with fkms but not kms. 

as soon as a new firmware is pushed out I'll test it but since you can bypass the filtering by jamming custom values into the mode struct before initializing kms/drm I'm worried that it still won't work. At least I assume that bypasses the filtering.  In any case no reason to pursue this further until the new firmware is out and ready to test, I just wanted to close the loop for posterity.

6by9, 2021-04-28:
> Are you saying that 480i60 can only be updated at 30hz?

Yes. Always has been the case if you're actually updating the display lists rather than scrawling over a live buffer.

> as soon as a new firmware is pushed out I'll test it

Kernel, not firmware.

> I'm worried that it still won't work

I've tested this with an HDMI analyser. It reports the timings exactly as expected.

alanbork, 2021-04-30:
> > Are you saying that 480i60 can only be updated at 30hz?
> 
> Yes. Always has been the case if you're actually updating the display lists rather than scrawling over a live buffer.

I guess I'm "scrawling over a live buffer" then. Times printed below are between calls to  eglSwapBuffers(*display, *surface), and alternating full black and white frames are displayed correctly.

```
 kmsmodeline 40
using /dev/dri/card0
720x480i-60.00hz(60): 13.514000 720 739 801 858 480 488 494 525 -hsync -vsync interlace dblclk
Initialized EGL version: 1.4
GL Viewport size: 720x480
16.70 ms        59.89 hz
16.64 ms        60.11 hz
16.70 ms        59.89 hz
16.66 ms        60.02 hz
16.67 ms        59.98 hz
16.64 ms        60.11 hz
16.70 ms        59.89 hz
16.64 ms        60.11 hz
16.70 ms        59.89 hz
```

(code:
````
    eglSwapBuffers(*display, *surface);
    struct gbm_bo *bo = gbm_surface_lock_front_buffer(gbmSurface);
    uint32_t handle = gbm_bo_get_handle(bo).u32;
    uint32_t pitch = gbm_bo_get_stride(bo);
    uint32_t fb;
    drmModeAddFB(device, mode->hdisplay, mode->vdisplay, 24, 32, pitch, handle, &fb);
    drmModeSetCrtc(device, crtc->crtc_id, fb, 0, 0, &connectorId, 1, mode);
````

in fact none of the code I've used seems to update at 32ms intervals, though I suspect strongly all are based on this example above. From a technical viewpoint I would have thought "scrawling over a live buffer" meant writing directly to the front buffer (or as we called it 10+ years ago, the framebuffer).

> > as soon as a new firmware is pushed out I'll test it
> 
> Kernel, not firmware.

as of aad6d2e912201703446691bb5efeca8e97dbaa9b I can confirm that 480i is detected and works properly with similar timing latency between kms/fkms and the old pi0..3 api as well.

alanbork, 2021-04-30:
oh goodness, I'm very sorry. that post above was with the new rpi-update but with fkms enabled not kms. 480i is still not detected under kms.  I'm going to update the post above.

6by9, 2021-04-30:
You could check the commit history yourself
https://github.com/Hexxeh/rpi-firmware/commits/master

https://github.com/Hexxeh/rpi-firmware/commit/aad6d2e912201703446691bb5efeca8e97dbaa9b
userland: Rebuild with man pages
See: raspberrypi/userland#663

https://github.com/Hexxeh/rpi-firmware/commit/1340be471e2764cca8b9f8266e23347d1288919e
 kernel: Bump to 5.10.33

kernel: spi: bcm2835: Increase the CS limit to 24
See: raspberrypi/linux#4281

kernel: ARM: dts: bcm2711-rpi-400: Limit MDIO clock speed

kernel: Single overlay file to configure the Si4682 I2S
See: raspberrypi/linux#4310

kernel: drm/vc4: Allow DBLCLK modes even if horz timing is odd.
See: raspberrypi/linux#4312

firmware: Pi400: Reduce MII clock freq when probing ethernet PHY

alanbork, 2021-04-30:
Yes, sorry that was sloppy of me, I thought it was working so I didn't bother to check the commits.  But I can report that it is working as of 1340be471e2764cca8b9f8266e23347d1288919e: the mode is now reported as existing via modetest, and it can be initialized too.

There is some oddness of the timing, with  eglSwapBuffers(*display, *surface); returning at a semi-random time after vsync , which means that sometimes you have a lot less than a full frame period to render your next frame.  This is not a problem with fkms @ 480i despite it's reputation for poor vsync timing.

alanbork, 2021-04-30:
after extensive testing I'm convinced that 480i is working right except for the timing issue. that seems to be generally a problem with interlaced modes not just 480i so I'm marking this as closed, although I am still very interested in resolving what 6by9 meant by interlaced modes only updating at 1/2 the field rate (that is 60hz updating at 30hz, sorry if I have the terms wrong).

closed 2021-04-30 by alanbork

popcornmix, 2021-05-05:
>  I am still very interested in resolving what 6by9 meant by interlaced modes only updating at 1/2 the field rate (that is 60hz updating at 30hz, sorry if I have the terms wrong).

There are two ways that may make sense for presenting frames on a 1080i50 display.
You could present a framebuffer of 1920x540 50 times a second with alternating odd and even fields.
You could present a framebuffer of 1920x1080 25 times a second which contains both odd and even fields interleaved.

I believe @6by9 is saying the latter is the scheme used.

6by9, 2021-05-05:
I believe the latter is used, but could be wrong. It depends on exactly when VSYNC is signalled, and when the hardware allows the display list to be updated.

The Pi HVS always generates a full progressive frame, and then takes only the odd or even lines from it for interlaced output. Whether it signals a VSYNC and allows the display list to be updated between the paired fields would need to be checked.

popcornmix, 2021-05-05:
Actually:
```
pi@pi4d:~ $ modetest -M vc4 -v -s 32:1920x1080i-25
setting mode 1920x1080i-25.00Hz on connectors 32, crtc 80
failed to set gamma: Invalid argument
freq: 50.64Hz
freq: 50.09Hz
freq: 50.09Hz
freq: 50.09Hz
freq: 50.09Hz
```
Suggests that vsync is triggering 50 times a second. Assuming the dlist is also allowed to change at that rate then you may be able to switch at 50Hz.

alanbork, 2021-05-05:
Thanks for the clarification, though it's not clear what the conclusion is here, actually.

The buffer submit rate is definitely 60hz for 480i (ie calling eglSwapBuffers() at full speed asymptotes at 16.6ms between function returns, except for the somewhat regular glitch where it's long by about 2ms). 

```
root@raspberrypi:~# kmsmodeline 40
/dev/dri/card0 does not have DRM resources, using card1
720x480i-60.00hz(60): 13.514000 720 739 801 858 480 488 494 525 -hsync -vsync interlace dblclk
Initialized EGL version: 1.4
GL Viewport size: 720x480
16.59 ms        60.28 hz
16.61 ms        60.19 hz
16.57 ms        60.35 hz
16.63 ms        60.14 hz
16.57 ms        60.34 hz
16.63 ms        60.12 hz
16.57 ms        60.36 hz
16.64 ms        60.10 hz

```

Full frame flicker summitted at 60hz looks roughly to be happening at 60hz too, but honestly that's hard to judge by eye, it might be at 30hz. I can set up a slow motion camera and check if that's the only way to know.  I'm not sure how a 60hz submit rate could work out to 30hz, though. I'd expect that it would just drop frames which would give no flicker at all. 

The timing of when display vsync happens relative to eglSwapBuffers() is pretty wacky (drifting slowly and jumping wildly), as documented in https://github.com/raspberrypi/linux/issues/4315.  it doesn't look like it's dropping frames, but it's possible.

alanbork, 2021-05-05:
> I believe the latter is used, but could be wrong. It depends on exactly when VSYNC is signalled, and when the hardware allows the display list to be updated.

FYI the hardware definitely does support 60hz updates in in 480i60 using fkms (accepts them at 60hz and updates the odd and even fields from the most recently submitted buffer with no lag).  perhaps the two different video drivers use the hardware in very different ways though?