raspberrypi/linux #4314: USB audio crackling on Rasperry Pi Compute Module 4

skateskate, 2021-04-29:
**Describe the bug**
I have really bad crackling sound when I try to reproduce 44100Hz - 24bit audio format on a C-Media USB sound card connected to a RasPi CM4 + CM4 I/O board. The other audio formats supported by the sound card (included 44100Hz - 16bit) reproduce correctly. If I connect the same sound card to a RasPi 2 with the same RasPiOS, all audio formats reproduce correctly. I have also tried to use Volumio (kernel 4.x.x) on the CM4 but the same problem persists. I think there could be a problem in the dwc2 usb controller of the CM4.

**To reproduce**
I run the following command to reproduce the issue:
`speaker-test -c 2 -r 44100 -F S24_3LE -D hw:1,0`
where `hw:1,0` is the index of the usb sound card read from `aplay -l`. 

**Expected behaviour**
To reproduce correctly also the 44100Hz-24bit audio format.

**Actual behaviour**
The audio crackles a lot, the white noise reproduced from `speaker-test` is really distorted. Same happens with a 44100Hz-24bit .wav file reproduced with `aplay`.

**System**
* Which model of Raspberry Pi? PiCM4 + CM4 I/O
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Feb 25 2021 12:10:40
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```
**Logs**
The first time I run the `speaker-test -c 2 -r 44100 -F S24_3LE -D hw:1,0` command, `dmesg` shows the following lines:
```
[  266.174814] ------------[ cut here ]------------
[  266.174889] WARNING: CPU: 0 PID: 561 at drivers/usb/dwc2/hcd.c:2683 dwc2_assign_and_init_hc+0x8a8/0x954 [dwc2]
[  266.174899] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc joydev snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device brcmfmac brcmutil sha256_generic cfg80211 rfkill v3d gpu_sched vc4 cec drm_kms_helper bcm2835_v4l2(C) raspberrypi_hwmon drm dwc2 videobuf2_vmalloc roles drm_panel_orientation_quirks bcm2835_isp(C) snd_soc_core bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm videodev vc_sm_cma(C) snd_timer mc snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio ip_tables x_tables ipv6
[  266.175433] CPU: 0 PID: 561 Comm: speaker-test Tainted: G         C        5.10.17-v7l+ #1403
[  266.175443] Hardware name: BCM2711
[  266.175453] Backtrace:
[  266.175494] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
[  266.175510]  r7:ffffffff r6:00000000 r5:60000193 r4:c12e69fc
[  266.175529] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
[  266.175548] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[  266.175564]  r10:c33fdef0 r9:00000009 r8:bf2f41e4 r7:00000a7b r6:00000009 r5:bf2f41e4
[  266.175574]  r4:bf30ac4c r3:c1205094
[  266.175592] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
[  266.175605]  r7:00000a7b r6:bf30ac4c r5:c1205048 r4:00000000
[  266.175657] [<c0b63ae8>] (warn_slowpath_fmt) from [<bf2f41e4>] (dwc2_assign_and_init_hc+0x8a8/0x954 [dwc2])
[  266.175671]  r9:00000000 r8:ded48000 r7:c3f67be4 r6:c30a4e00 r5:c3453040 r4:c30aa044
[  266.175751] [<bf2f393c>] (dwc2_assign_and_init_hc [dwc2]) from [<bf2f59a0>] (dwc2_hcd_select_transactions+0xb0/0x1a4 [dwc2])
[  266.175765]  r10:c33fdef0 r9:c34534e0 r8:c3453458 r7:c3453450 r6:00000001 r5:c30a4e84
[  266.175775]  r4:c3453040
[  266.175849] [<bf2f58f0>] (dwc2_hcd_select_transactions [dwc2]) from [<bf2faebc>] (dwc2_handle_hcd_intr+0x240/0x69c [dwc2])
[  266.175863]  r10:c33fdef0 r9:ffffe000 r8:00000008 r7:c3453170 r6:00000001 r5:c3453450
[  266.175873]  r4:c3453040 r3:c3453448
[  266.175946] [<bf2fac7c>] (dwc2_handle_hcd_intr [dwc2]) from [<bf2f2d1c>] (_dwc2_hcd_irq+0x1c/0x20 [dwc2])
[  266.175960]  r10:c33fdef0 r9:ffffe000 r8:00000025 r7:c1a3ac00 r6:00000000 r5:00000000
[  266.175969]  r4:c309e380
[  266.176022] [<bf2f2d00>] (_dwc2_hcd_irq [dwc2]) from [<c08d7e24>] (usb_hcd_irq+0x34/0x48)
[  266.176043] [<c08d7df0>] (usb_hcd_irq) from [<c0287e20>] (__handle_irq_event_percpu+0xa8/0x268)
[  266.176061] [<c0287d78>] (__handle_irq_event_percpu) from [<c028801c>] (handle_irq_event_percpu+0x3c/0x90)
[  266.176075]  r10:00000000 r9:00026008 r8:c1810800 r7:00000001 r6:c1a3ac00 r5:c1a3ac6c
[  266.176084]  r4:c1205048
[  266.176102] [<c0287fe0>] (handle_irq_event_percpu) from [<c02880c4>] (handle_irq_event+0x54/0x78)
[  266.176113]  r6:00000000 r5:c1a3ac6c r4:c1a3ac00
[  266.176131] [<c0288070>] (handle_irq_event) from [<c028ce60>] (handle_fasteoi_irq+0xd4/0x18c)
[  266.176143]  r7:00000001 r6:00000000 r5:c1205b3c r4:c1a3ac00
[  266.176159] [<c028cd8c>] (handle_fasteoi_irq) from [<c0286ddc>] (generic_handle_irq+0x44/0x54)
[  266.176169]  r5:00000000 r4:c1094d10
[... 177 lines cut ...]
    Data packet interval: 1000 us
    Bits: 24
    Channel map: FL FR

```

`lsusb` output:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 003: ID 0d8c:0171 C-Media Electronics, Inc.
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 1, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 4, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 5, Class=Human Interface Device, Driver=usbhid, 480M
```

lsusb verbose on the C-Media:
[verbose.txt](https://github.com/raspberrypi/linux/files/6401614/verbose.txt)

P33M, 2021-05-07:
dwc2 doesn't support FIQ acceleration. dwc_otg is the default host driver used on Pi 2 (deprecated on BCM2711 ) and its FIQ code prevents disruption of isochronous scheduling from system interrupt contention.

I would recommend setting `otg_mode=1` in config.txt to use the built-in xhci controller with the USB2 OTG port.

skateskate, 2021-05-08:
Thank you very much, substituting `dtoverlay=dwc2,dr_mode=host` (used to enable the CM4 USBs as suggested in the datasheet) with `otg_mode=1` in the config.txt solved my issue!

closed 2021-05-13 by skateskate