raspberrypi/linux #4315: pi4/kms: high variability between eglSwapBuffers return and vsync start with interlaced modes

alanbork, 2021-04-30:
testing with kms I find that interlaced modes (480i and 1080i) have a very high variability between when vsync occurs on the display and the return of  eglSwapBuffers(), with the time spent in eglSwapBuffers() sometimes taking much more than an entire frame. This is inefficient for high performance rendering, since it leaves less time for useful computation.

testing with fkms I see no such issue. Although fkms used to be very bad about vsync timing, it seems to be much better now., (though still fragile in other ways that make me desire to use kms).

Aside from the efficiently issue, this personally effects me because I use the time between when the monitor responds and eglSwapBuffers() to measure monitor input lag. 

example data (time between return from eglSwapBuffers() and monitor's response, measured with a photodiode) on a display with ~27ms of input lag using Fkms: 

27.1
27.2
27.3
27.3
27.3
27.3
27.3
27.4
27.4
27.4
27.4
27.4
27.4
27.4
27.4
27.4
27.4
27.5
27.5
27.5
27.5
27.5
27.6
27.6
27.6
(times are milliseconds; data is randomly sampled and sorted).

same setup but with kms:
17.0
17.3
19.7
20.8
21.2
21.3
21.5
21.7
23.5
23.8
23.9
24.5
25.3
25.4
25.6
25.6
25.6
25.7
25.8
25.8
25.9
26.6
27.4
27.9
31.4

This may be related to the report from 6by9 that interlaced modes can only be updated at 30hz:

https://github.com/raspberrypi/linux/issues/4307#issuecomment-830244108

although if it were exactly that issue, I would expect timing to alternate between being too short and too long, rather than the slow drift + random variability each time the display is initialized.  

All testing with release 1340be471e2764cca8b9f8266e23347d1288919e.

alanbork, 2021-05-10:
This seems to be somewhat TV dependent, which is really strange. I just ran the same exact code on a TCL 40S325, and the variability issue doesn't replicate using kms (or fkms). This is across lots of reboots, and repeat runs.  The bug report above was based on results from a Sony X800D. It's as though the Sony is slowly adjusting it's own vsync to match that of the pi, taking a few seconds to do so (indeed, this is what it does in non-interlaced modes, successfully), but for some reason the signal put out by the pi breaks this synchronization. 

I'm also wondering if this might be related to this other issue I opened where a Panasonic TV seems to be confused by which lines are odd and even: https://github.com/raspberrypi/linux/issues/4331 - which seems a bit like an interlacing timing issue since the odd lines are being rendered like they are even, and vs versa.

alanbork, 2021-05-10:
sorry if I was unclear. I know it's a one-way thing.  What I'm suggesting here is that there's some variably in the pi's vsync which is confusing the TV.  The sony TV adjusts it's own redraw start to match pi's vsync over multiple frames rather than just waiting for the first vsync and latching to that.  But jitter in vsync is just a guess as to why these two TVs have significant (but very different) issues with the interlaced video signal put out by kms (but not fkms). if you have an alternate hypothesis from your position of greater knowledge about HDMI video signals and the pi(4) I'm all ears.

popcornmix, 2021-05-11:
In terms of the timing signals the TV sees, that is all generated by hardware and won't have any jitter.
Vsync callbacks in software will have some additional variability.
With kms, I'd expect that to be very low for something simple like `modetest -v`, but adding something like GL to the mix will likely increase it.
With fkms, the software vsync callback is flawed, so there are no guarantees.

alanbork, 2022-05-11:
kms has gotten way better timing in the days since I opened this bug, so I'm closing it now. Not sure when the change occurred, but some time in 2021. Thanks guys!

closed 2022-05-11 by alanbork