raspberrypi/linux #4316: Latest stable kernel 5.10.17+ #1414 linux source commit id missing

MrEngman, 2021-05-01:
The latest stable kernel **5.10.17xxx+ #1414** appeared in Hexxeh/rpi-firmware and raspberrypi/firmware yesterday 30 April.

Tried compiling my out of tree wifi drivers and the kernel commit id could not be found. In Hexxeh/rpi-firmware and raspberrypi/firmware **git-hash=3a33f11c48572b9dd0fecac164b3990fc9234da8**, however the commit ID does not appear to exist.

MrEngman

XECDesign, 2021-05-01:
https://github.com/raspberrypi/linux/commit/3a33f11c48572b9dd0fecac164b3990fc9234da8

I had to add `git fetch --tags` to work resolve it in my scripts.

MrEngman, 2021-05-01:
Thank you. Never seen that mentioned before but ran the command in the linux src cloned directory and the compiles worked just fine.

MrEngman