raspberrypi/linux #4317: Raspberry Pi 4B 8G, Blank Sceen, Invalid EDID on 5.10.33 but not on 5.10.27

SteveFosdick, 2021-05-01:
**Is this the right place for my bug report?**
I suspect this is a Linux kernel bug because the bug is seen with a later version of the kernel (5.10.33) and not with an earlier one (5.10.27)

**Describe the bug**
The HDMI output is initially driven with a signal my TV can understand but this ceases - the TV complains of no signal - when I would expect the login prompt to appear.  There is a message in the system log about the EDID being invalid.

**To reproduce**
This issue does seem to depend on both the kernel version and on the particular TV/monitor connected.  For me, I attached a Celcus brand TV to the first HDMI port, selected the HDMI input from within the TV's input menu and turn on the Raspberry Pi having previously updated to linux 5.10.33-1-ARCH (inux-raspberrypi4-5.10.33-1).

**Expected behaviour**
I expected the Raspberry Pi to complete the boot process and present a login prompt in a virtual console.

**Actual behaviour**
An early boot screen briefly flashed on the screen, followed by the rainbow pattern, then the screen with four raspberries at the top and the kernel and systemd messages appeared.  Once the boot processed seemed almost complete the TV complained "No HDMI signal".  Tapping keys on the keyboard did not cause the TV to see an HDMI signal.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4B 8Gb.
* Which OS and version (`cat /etc/rpi-issue`)?
Arch Linux ARM aarch64
* Which firmware version (`vcgencmd version`)?
Apr 30 2021 15:54:12 
Copyright (c) 2012 Broadcom
version 3e104c5ed69dc796755b7195973a8d00482d76dd (clean) (release) (start)
* Which kernel version (`uname -a`)?

**Logs**
Apr 16 16:12:03 alarm systemd[1]: Starting Create Volatile Files and Directories...
Apr 16 16:12:03 alarm systemd-udevd[242]: cfg80211: Process '/usr/bin/set-wireless-regdom' failed with exit code >
Apr 16 16:12:03 alarm kernel: Bluetooth: Core ver 2.22
Apr 16 16:12:03 alarm kernel: [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
Apr 16 16:12:03 alarm kernel: NET: Registered protocol family 31
Apr 16 16:12:03 alarm kernel: Bluetooth: HCI device and connection manager initialized
Apr 16 16:12:03 alarm kernel: Bluetooth: HCI socket layer initialized
Apr 16 16:12:03 alarm kernel: Bluetooth: L2CAP socket layer initialized
Apr 16 16:12:03 alarm kernel: Bluetooth: SCO socket layer initialized
Apr 16 16:12:03 alarm kernel: vc4-drm gpu: [drm] HDMI-A-1: EDID is invalid:
Apr 16 16:12:03 alarm kernel:         [00] BAD  00 ff ff ff ff ff ff 00 3c ad 00 37 00 00 00 00
Apr 16 16:12:03 alarm kernel:         [00] BAD  0a 12 01 03 80 10 09 78 0a 68 37 a4 54 48 9a 25
Apr 16 16:12:03 alarm kernel:         [00] BAD  0c 4a 4c 3f ef 80 d1 00 d1 c0 81 19 90 59 01 01
Apr 16 16:12:03 alarm kernel:         [00] BAD  01 01 01 01 01 01 01 1d 80 18 71 1c 16 20 58 2c
Apr 16 16:12:03 alarm kernel:         [00] BAD  25 00 10 09 00 00 00 9e 01 1d 00 72 51 d0 1e 20
Apr 16 16:12:03 alarm kernel:         [00] BAD  6e 28 55 00 10 09 00 00 00 1e 00 00 00 fc 00 32
Apr 16 16:12:03 alarm kernel:         [00] BAD  32 57 5f 4c 43 44 5f 54 56 0a 20 20 00 00 00 fd
Apr 16 16:12:03 alarm kernel:         [00] BAD  00 31 3d 0f 2e 08 00 0a 20 20 20 20 20 20 01 2c
Apr 16 16:12:03 alarm systemd[1]: Finished Create Volatile Files and Directories.

**Additional context**
As mentioned earlier, this problem does not exist with the same TV but kernel version 5.10.27.  When booting with 5.10.27 the message about about invalid EDID is not in the journal and the login prompt appears as expected.

Because this is TV/monitor specific I have attached the EDID file for the TV which has the problem and the decoded version (from edid-decode) is:
edid-decode (hex):

00 ff ff ff ff ff ff 00 3c ad 00 37 00 00 00 00
0a 12 01 03 80 10 09 78 0a 68 37 a4 54 48 9a 25
0c 4a 4c 3f ef 80 d1 00 d1 c0 81 19 90 59 01 01
01 01 01 01 01 01 01 1d 80 18 71 1c 16 20 58 2c
25 00 10 09 00 00 00 9e 01 1d 00 72 51 d0 1e 20
6e 28 55 00 10 09 00 00 00 1e 00 00 00 fc 00 32
32 57 5f 4c 43 44 5f 54 56 0a 20 20 00 00 00 fd
00 31 3d 0f 2e 08 00 0a 20 20 20 20 20 20 01 2c

02 03 24 71 50 01 02 04 85 06 11 13 14 15 03 07
12 16 1f 20 10 23 09 07 01 83 01 00 00 66 03 0c
00 10 00 80 8c 0a d0 8a 20 e0 2d 10 10 3e 96 00
04 03 00 00 00 18 8c 0a a0 14 51 f0 16 00 26 7c
43 00 04 03 00 00 00 99 01 1d 80 d0 72 1c 16 20
10 2c 25 80 10 09 00 00 00 9f 01 1d 00 bc 52 d0
[... 83 lines cut ...]
      Max channels: 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 16
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 1.0.0.0
    Supports_AI
  Detailed Timing Descriptors:
    DTD 3:   720x480    59.940 Hz   3:2    31.469 kHz   27.000 MHz (4 mm x 3 mm)
                 Hfront   16 Hsync  62 Hback  60 Hpol N
                 Vfront    9 Vsync   6 Vback  30 Vpol N
    DTD 4:  1440x480i   59.940 Hz   3:1    15.734 kHz   27.000 MHz (4 mm x 3 mm)
                 Hfront   38 Hsync 124 Hback 114 Hpol N
                 Vfront    4 Vsync   3 Vback  15 Vpol N Vfront +0.5 Odd Field
                 Vfront    4 Vsync   3 Vback  15 Vpol N Vback  +0.5 Even Field
    DTD 5:  1920x1080i  50.000 Hz  16:9    28.125 kHz   74.250 MHz (16 mm x 9 mm)
                 Hfront  528 Hsync  44 Hback 148 Hpol P
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback  15 Vpol P Vback  +0.5 Even Field
    DTD 6:  1280x720    50.000 Hz  16:9    37.500 kHz   74.250 MHz (16 mm x 9 mm)
                 Hfront  440 Hsync  40 Hback 220 Hpol P
                 Vfront    5 Vsync   5 Vback  20 Vpol P
Checksum: 0xf2
[edid.gz](https://github.com/raspberrypi/linux/files/6410496/edid.gz)

user13571113, 2021-05-08:
**Is this the right place for my bug report?**
As reported by SteveFosdick, I have no HDMI output with Xorg following the kernel upgrade 5.10.27 -> 5.10.31 on ArchLinuxARM 64bit.

**Describe the bug**
With the 5.10.27-4 kernel (and any prior version), I can boot up my Raspberrpi 4B 8G, see the boot log on my screen (cf. "Additional context" below for the screen model information), log in, run `startx`, and enter a working LXQt environment.

With the 5.10.31-5 kernel (and any future version), I can boot up my Raspberrpi 4B 8G, see the boot log on my screen, log in, but when running `startx`, my screen goes blank saying "No HDMI input signal". This also occurs when I enable a display manager (e.g., LightDM).

**To reproduce**
```
[alarm@alarm]$ cat /boot/config.txt
# See /boot/overlays/README for all available options

over_voltage=5
arm_freq=2000

gpu_mem=256
dtoverlay=vc4-kms-v3d
disable_overscan=1

initramfs initramfs-linux.img followkernel
```
Use an up-to-date ArchlinuxARM image, install xorg-server, xorg-xinit, xf86-video-fbdev, and any desktop environment; create an appropriate ".xinitrc" file (mine contains `exec startlxqt`); connect the screen to the HDMI 0 input, power on the device, log in, and run `startx`.

**Expected behaviour**
A working X session displayed on my screen.

**Actual behaviour**
For a brief moment, I see the startup messages of the X server, then the screen goes blank and displays "No HDMI input signal". The X server, however, is running and I can switch TTYs; via CTRL+ALT+F5, I return to the console (displayed properly on the screen) and run `pkill Xorg`. No errors are reported in the Xorg log file (see logs below). Thus I suspect that the problem lies with the vc4 driver and not with the X server (note also that many commits were made to vc4 driver during the 5.10.27 -> 5.10.31 transition.)

**System**
* Which model of Raspberry Pi? Pi4B 8G
* Which OS and version? ArchLinux ARM 64bit running _this_ kernel (https://archlinuxarm.org/packages/aarch64/linux-raspberrypi4)
* Which firmware version (`vcgencmd version`)?
Apr 30 2021 15:54:12 
Copyright (c) 2012 Broadcom
version 3e104c5ed69dc796755b7195973a8d00482d76dd (clean) (release) (start)
* Which kernel version (`uname -a`)? 5.10.35-1 (corresponds to https://github.com/raspberrypi/linux/archive/53a5ac4935c500d32bfc465551cc5107e091c09c.tar.gz)

**Logs**
Working X session with kernel 5.10.27-4 (see logs of the failing session below):

dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.27-4-ARCH (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35) #1 SMP PREEMPT Fri Apr 16 15:57:09 MDT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e800000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 2015232
[    0.000000]   DMA zone: 3072 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12288 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s90904 r8192 d31976 u131072
[    0.000000] pcpu-alloc: s90904 r8192 d31976 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[... 1915 lines cut ...]
Vendor: Ancor Communications Inc
Serial Number: E3LMTF171793
Display Size: 60cm x 34cm

**Custom HDMI modes**

As seen in the EDID entry in the output of `dmesg`, my native screen resolution is 2560x1440 @ 60Hz. Changing the contents of /boot/config.txt to

```
over_voltage=5
arm_freq=2000

gpu_mem=256
dtoverlay=vc4-kms-v3d
disable_overscan=1

hdmi_cvt=2560 1440 60
hdmi_group=2
hdmi_mode=87
hdmi_drive=2

initramfs initramfs-linux.img followkernel
```

does not change the erroneous behavior (I observe exactly the same problem; logs related to the above settings can be provided if requested.)

user13571113, 2021-06-20:
As replied [here](https://github.com/raspberrypi/linux/issues/4339#issuecomment-864587587), the issue is resolved for me as of June 20, 2021.

closed 2021-09-01 by SteveFosdick