raspberrypi/linux #4318: RFC: Future of VCHIQ function tracing

lategoodbye, 2021-05-02:
While checking/cleaning up the vchiq sources with checkpatch.pl the script complains that some trace calls provide something which is already handled by the common kernel function called [ftrace](https://www.kernel.org/doc/Documentation/trace/ftrace.txt).

Example:
WARNING: Unnecessary ftrace-like logging - prefer using ftrace
FILE: drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c:
+	vchiq_log_trace(vchiq_core_log_level, "%s called", __func__);

So the question is how to improve this situation? Currently i see the following options:
a) drop all function entry traces, but keep the return traces
b) drop all function entry traces and trace only the error cases of return

Any comments or better ideas?
@pelwell @popcornmix @vianpl

pelwell, 2021-05-04:
There's a value to having the function tracing integrated with the more data-oriented tracing (rather than having to stitch the two together after the fact), but if necessary it could be removed or limited to error cases-only. Fortunately, most of the functions in vchiq_arm.c result in calls into vchiq_core.c which contains more event-oriented tracing, so in practise nothing much should be lost.

lategoodbye, 2021-05-04:
Okay, do you prefer one of the options above?

pelwell, 2021-05-04:
I marginally prefer option a.

lategoodbye, 2021-05-20:
Just for reference:
https://lore.kernel.org/linux-staging/1621105859-30215-10-git-send-email-stefan.wahren@i2se.com/T/#u

Thanks

closed 2021-05-20 by lategoodbye