raspberrypi/linux #4320: vc4 mainline 5.12 kernel regression with flip_done timed

warpme, 2021-05-03:
**Describe the bug**
After upgrade of mainline kernel from 5.11.16 to 5.12 i started to see regression in drm synsytem on rpi4 (rpi3 works ok). It manifests as flip_done timed out at playback exit. 
dmesg with drm log is like this: https://pastebin.com/rHqYdAnL 

By Slack conversation (https://libreelec.slack.com/archives/CAAMLHZ1U/p1620040466129700) - on 5.12.1 mainline: revert of f3c420fe19f8 not helps. but revert f3c420fe19f8 then 82faa3276012 fixes issue. 

**System**

* Which model of Raspberry Pi: RPI4
* Which OS and version: MiniMyth2
* Which firmware version: g20081d8e86119
* Which kernel version: 5.12.1 mainline

**Logs**
https://pastebin.com/rHqYdAnL

popcornmix, 2021-05-03:
Confirming @mripard requested issue created here to avoid it getting lost.

mripard, 2021-06-10:
I have made a fix here:
https://github.com/mripard/rpi-linux/tree/rpi/fix-async-commits

I made a test program that was mimicking your logs and it's fixed, but it would be great if you could test it with a real workload

warpme, 2021-06-14:
Maxime,
I applied https://github.com/mripard/rpi-linux/commit/bdb339dfc173cb8236157a17b9eed3e07191b217 on 5.12.10 and indeed it looks issue gone. I'll test it a bit more - but so far it works well :-) Great work!

closed 2021-06-14 by warpme