raspberrypi/linux #4321: Mailbox driver timeout prevents longer-running code execution on QPU

doe300, 2021-05-03:
The mailbox interface has a hard-coded timeout in the kernel ([here](https://github.com/raspberrypi/linux/blob/7fb9d006d3ff3baf2e205e0c85c4e4fd0a64fcd0/drivers/firmware/raspberrypi.c#L60)) set to 1 seconds introduced in https://github.com/raspberrypi/linux/commit/0829187b17d8019f9bb40c56d47f37b1991a2189.

This makes it impossible to run any command which takes longer than a second. I do get that for most commands, 1s is a lifetime and a safe choice, but for execution of code on the QPUs (mailbox property tag 0x00030011), this is a very narrow limit which can be reached by a correct and responsive code execution. Since the mailbox property tag 0x00030011 already has a timeout argument (see [here](https://github.com/raspberrypi/userland/blob/45a0022ac64b4d0788def3c5230c972430f6fc23/host_applications/linux/apps/hello_pi/hello_fft/mailbox.c#L238)), can't the kernel reuse this one for this special case?

BTW, is there any particular reason the mailbox property tag 0x00030011 to execute code on the QPUs is not documented in https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface ?

lategoodbye, 2021-05-03:
As author of the mentioned patch, i wanted to known is a module parameter to configure the timeout a solution for you?

doe300, 2021-05-03:
I would need this configuration to be doable from userspace at runtime (e.g. passing in to the ioctl). If we can make the module parameter dynamically configurable at runtime without elevated privileges, then this would also be okay (assuming this can be done at all).
For running the software as root, I already directly poke the V3D registers to run code on the QPUs which circumvents this timeout.

If the module parameter can be set (with/without elevated privileges) as a one-time setup, this might also be okay depending on the complexity of the initial setup.

doe300, 2021-05-13:
As mentioned in the linked commit, I managed to get the VCHIQ GPU service working to execute code on the QPUs.

Since th GPU service has a very similar interface for executing code on both VPU and QPUs and any other mailbox call should not take nearly a second, I guess there is a working way and no need to change anything on the mailbox driver.

closed 2021-05-13 by doe300

dernasherbrezon, 2021-11-17:
It looks like ```mbox_send_message``` will wait for timeout:
https://github.com/raspberrypi/linux/blob/0829187b17d8019f9bb40c56d47f37b1991a2189/drivers/mailbox/mailbox.c#L275

So no need for additional timeout in the firmware (not tested!)

lategoodbye, 2021-11-18:
The code path isn't reached. But yes, the BCM2835 mailbox driver could be improved.

dernasherbrezon, 2021-11-18:
I’m trying to narrow down the issue with early return from the blocking qpu execution. If early returns happens then the next attempt always fails and corrupts something , so only hard reset helps.

Double check for timeout on the same completion object looks suspicious. As well as two spin locks: one for mailbox , the other one is for bcm mailbox controller.