raspberrypi/linux #4324: with RT patch, DWC2 eat a lot of CPUS on CM4

andygaof, 2021-05-05:
**Describe the bug**
Have tried 5.10.14 and 5.10.17. Both of the kernels with RT Patch. build with kvm off and full RT.
after plug a USB device. the CPU will be around 17% for DWC2 and FE9800000

**To reproduce**
just install the kernel with RT patch. make with default BCM2711_DEFCONFIG, disable kvm, select full RT. after install the kernel to CM4, after cm4 bootup, plugin a usb device. you will see very high CPU usage. about 17% for DWC2.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?  computer module 4
* Which OS and version (`cat /etc/rpi-issue`)? pios and arch linux both is the same issue
* Which firmware version (`vcgencmd version`)? tried serveral version. same issue
* Which kernel version (`uname -a`)? Linux PNA 5.10.14-rt27-v8+ #1 SMP PREEMPT_RT Wed May 5 02:04:00 CST 2021 aarch64 GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.14-rt27-v8+ (root@CelAudio-Builder) (aarch64-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1 SMP PREEMPT_RT Wed May 5 02:04:00 CST 2021
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Zeroed struct page in unavailable ranges: 512 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 517632
[    0.000000]   DMA zone: 4096 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 255488 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 27 pages/cpu s72576 r8192 d29824 u110592
[    0.000000] pcpu-alloc: s72576 r8192 d29824 u110592 alloc=27*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 509440
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:FE:6F:2C vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rw rootwait rootfstype=ext4 dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 smsc95xx.turbo_mode=N fsck.repair=yes
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000003a600000-0x000000003e600000] (64MB)
[    0.000000] Memory: 1877132K/2070528K available (10816K kernel code, 1822K rwdata, 3792K rodata, 3456K init, 1257K bss, 127860K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33595 entries in 132 pages
[    0.000000] ftrace: allocated 132 pages with 2 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] rcu: 	RCU priority boosting: priority 1 delay 500 ms.
[    0.000000] rcu: 	RCU_SOFTIRQ processing moved to rcuc kthreads.
[    0.000000] 	No expedited grace period (rcu_normal_after_boot).
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[... 331 lines cut ...]
[   13.723050] audit: type=1334 audit(1620162483.476:52): prog-id=20 op=LOAD
[   19.905971] kauditd_printk_skb: 39 callbacks suppressed
[   19.905998] audit: type=1130 audit(1620162489.658:92): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-udev-trigger comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   21.164420] asoc-simple-card soc:sound: ASoC: no DMI vendor name!
[   21.296290] audit: type=1131 audit(1620162491.049:93): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=minidlna comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   21.662839] audit: type=1130 audit(1620162491.415:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=minidlna comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   23.804194] audit: type=1130 audit(1620162493.556:95): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=upnprender comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[   25.850310] audit: type=1130 audit(1620162495.603:96): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=roonbridge comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   26.276530] audit: type=1130 audit(1620162496.029:97): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=networkaudio comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   27.479662] audit: type=1131 audit(1620162497.232:98): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=roonbridge comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   28.711497] audit: type=1130 audit(1620162498.464:99): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=roonbridge comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.161971] audit: type=1130 audit(1620217945.161:100): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=avahi-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.257514] audit: type=1130 audit(1620217945.256:101): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=shairport-sync comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[   31.500810] audit: type=1130 audit(1620217945.500:102): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=upnprenderop comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.559482] audit: type=1130 audit(1620217945.559:103): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=networkaudioop comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.633585] audit: type=1130 audit(1620217945.633:104): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=minidlnaop comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.666320] audit: type=1130 audit(1620217945.666:105): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=roonbridgeop comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   31.703844] audit: type=1131 audit(1620217945.703:106): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=minidlnaop comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[   32.303640] audit: type=1130 audit(1620217946.303:107): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=cron comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   32.709411] audit: type=1131 audit(1620217946.709:108): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=sshd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   32.824352] audit: type=1131 audit(1620217946.824:109): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-timesyncd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   34.961876] sched: RT throttling activated

**Additional context**
Add any other relevant context for the problem.

pelwell, 2021-05-05:
I'm happy for this issue to act as a place to discuss the problem (no me-toos, please - just helpful suggestions and insights), but be aware that the RT patches are not supported here - but we will host a -rt kernel, when someone is prepared to support it.

andygaof, 2021-05-05:
got it. then May I get some information about how to enable FIQ?

andygaof, 2021-05-05:
I mean dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 in the cmdline.txt. will work?

pelwell, 2021-05-05:
Getting FIQ to work with the RT kernel was always (from what I remember) the stumbling block. Using dwc2 was the workaround because it isn't FIQ-aware.

Given that you're on CM4 (a BCM2711-based platform) and aren't trying to be a gadget, why not use the more-capable onboard XHCI controller instead? Add `otg_mode=1` to config.txt.

andygaof, 2021-05-05:
May I get more details for this? I do not have any documentations on this. does it mean the config.txt like this? Thanks

gpu_mem=16
kernel=kernel8.img
dtoverlay=dwc2
dr_mode=host
otg_mod=1
dtparam=i2c=on,i2s=on,spi=on

l

andygaof, 2021-05-05:
Or, just like this

 gpu_mem=16
kernel=kernel8.img
otg_mod=1
dtparam=i2c=on,i2s=on,spi=on

pelwell, 2021-05-05:
Closer to the latter (i.e. no dwc2), but be careful to spell the setting correctly - it's `otg_mode=1` (with an `e`).

kdoren, 2021-05-16:
@andygaof Don't know if it will help on CM4, but I just posted a 32-bit release of PREEMPT_RT kernel 5.10.35-rt39 that I built by applying the same dwc_otg patches that were used in the rpi-4.19.y-rt kernel (in addition to standard PREEMPT_RT patch set).  These additional patches were required to get it working on Pi3.

Also FYI I suggest using an OS build with latest firmware.  There have been some recently-fixed firmware bugs that could be easily confused with kernel problems.

https://github.com/kdoren/linux/releases/tag/5.10.35-rt39

P33M, 2021-05-25:
dwc_otg plus FIQ code is incompatible with an -RT kernel on any model of Pi. Threaded interrupt handlers just don't work, the FIQ code requires that spinlocks are honoured.

dwc_otg is effectively deprecated on all BCM2711-based Pi products, and users should either use dwc2 (for OTG or device mode) or the on-board xhci controller (for host mode). Details on how to select the preferred host-mode controller are given above.

closed 2021-05-25 by P33M