raspberrypi/linux #4325: Raspberry PI CM4 not booting from USB

havok13888, 2021-05-05:
**Describe the bug**
Raspberry PI CM4 not booting from USB. We have a carrier with the USB 2.0 lane connected to an sd card slot. If we boot into the OS using EMMC we can see the sd card mounted to sda. But we cannot use this sd card to boot from if the EMMC is blank. We are using yocto gatesgarth to build the image.

**To reproduce**
1. Build Yocto gatesgarth Image. 
2. Erase emmc. 
3. Burn the image to an SD card and try to boot from it.

**Expected behaviour**
CM4 should be able to boot from SD card using USB 2.0

**Actual behaviour**
config.txt is loaded -> errors (attached logs) -> rainbow screen

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW - Raspberry PI CM4
* Which OS and version (`cat /etc/rpi-issue`)? Yocto Poky Gatesgarth
* Which firmware version (`vcgencmd version`)? 5.10.17, issue exists on 5.10.31 too
* Which kernel version (`uname -a`)? Linux localhost 5.10.17-v7l #1 SMP Mon Mar 1 09:44:55 UTC 2021 armv7l GNU/Linux

**Logs**
[usb-boot-xhci-debug.log](https://github.com/raspberrypi/linux/files/6430051/usb-boot-xhci-debug.log)
[usb-boot-xhci-debug2.log](https://github.com/raspberrypi/linux/files/6430052/usb-boot-xhci-debug2.log)
[usb-boot-using-start_debug.log](https://github.com/raspberrypi/linux/files/6430053/usb-boot-using-start_debug.log)
[cmdline.txt](https://github.com/raspberrypi/linux/files/6430109/cmdline.txt)
[config.txt](https://github.com/raspberrypi/linux/files/6430110/config.txt)

**Additional context**
Fails with u-boot too. It never reaches that stage. I can provide an image if needed.

pelwell, 2021-05-05:
You're bootloader EEPROM image is too old to take advantage of the onboard XHCI controller. From https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md:
> 2021-04-19 - Promote 2021-03-18 from LATEST to DEFAULT - DEFAULT
> ...
> Add support for the BCM2711 XHCI controller - BOOT_ORDER 0x5

havok13888, 2021-05-06:
Same problem using pieeprom-2021-04-29.bin

timg236, 2021-05-07:
All of the logs seem to show that start.elf is loaded correctly but the first USB command where it reads the MBR in start.elf fails. This suggests that this USB hub device is sensitive to something in the early firmware init code and stops responding. 

Can you reproduce this on a CM4 IO board i.e. with a hub using that chipset?

havok13888, 2021-05-07:
Tried just the CM4 IO board, no hardware changes. Installed Raspbian buster to a usb stick and put the stick in the USB 2.0 slot. It still wouldn't boot. Log attached
[puttyOut1.txt](https://github.com/raspberrypi/linux/files/6444272/puttyOut1.txt)