raspberrypi/linux #4328: RPI4 failing to create context for opengl without X with proper sdl2

Askmewho, 2021-05-06:
**Is this the right place for my bug report?**
I do think so.

**Describe the bug**
I am on rpi os 32 bit, kernel  5.10.17-v7l+ . I compiled sdl2 master with kms (and opengl) support just in case, the same that works perfectly fine to draw stuff without X on RK3399 under panfrost. for some reason, on RPI4, it wont draw anything, the kmscube works just fine, but thats just gles, and it's GL what I need.

**To reproduce**
just got to gbm and try to run opengl desktop profile workloads.

**Expected behaviour**
to work

**Actual behaviour**
it is not able to create the window

**System**

```
pi@raspberrypi:~ $ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2

Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
Revision	: c03111
Serial		: 10000000efa53f4a
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x50000
Camera          : supported=0 detected=0

Videocore information
---------------------

Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
S.ficheros     bloques de 1K    Usados Disponibles Uso% Montado en
/dev/root           61105308   8579876    50610556  15% /
devtmpfs             1801156         0     1801156   0% /dev
tmpfs                1933252     31652     1901600   2% /dev/shm
tmpfs                1933252     10088     1923164   1% /run
tmpfs                   5120         4        5116   1% /run/lock
tmpfs                1933252         0     1933252   0% /sys/fs/cgroup
/dev/mmcblk0p1        258095     48783      209312  19% /boot
tmpfs                 386648        28      386620   1% /run/user/1000
/dev/sda1          640733180 602902792    37830388  95% /media/pi/Externo
/dev/sda2          329673224 305415164     7441932  98% /media/pi/Linux

Filename				Type		Size		Used	Priority
/var/swap                               file		102396		0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Instalados: (ninguno)
[... 554 lines cut ...]
[  146.226888] ......ready
[  151.427649] sd 0:0:0:0: [sda] 1953458176 512-byte logical blocks: (1.00 TB/931 GiB)
[  151.428258] sd 0:0:0:0: [sda] Write Protect is off
[  151.428266] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[  151.428864] sd 0:0:0:0: [sda] No Caching mode page found
[  151.428872] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  151.528945]  sda: sda1 sda2
[  151.534361] sd 0:0:0:0: [sda] Attached SCSI disk
[  153.182609] fuse: init (API version 7.32)
[  158.947941] Under-voltage detected! (0x00050005)
[  165.188141] Voltage normalised (0x00000000)
[  192.230284] Under-voltage detected! (0x00050005)
[  198.469446] Voltage normalised (0x00000000)
[  248.003833] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
[  294.518366] process 'usr/local/bin/box86' started with executable stack
[  374.611110] v3d fec00000.v3d: MMU error from client L2T (0) at 0xdaa1000, pte invalid
```
**Logs**
maybe..
```
[  294.518366] process 'usr/local/bin/box86' started with executable stack
[  374.611110] v3d fec00000.v3d: MMU error from client L2T (0) at 0xdaa1000, pte invalid
```
**Additional context**
none

popcornmix, 2021-05-08:
This feels more like an SDL issue that a kernel issue.
What happen if you switch from fkms driver to kms?

Askmewho, 2021-05-08:
Sadly, nothing.

vanfanel, 2021-05-09:
@Askmewho I did the recent KMSDRM backend work on SDL2. And I can say for sure that GLES2/GL2 both work on SDL2 on a Pi3 and Pi4.
I don't know what's wrong on your side, but there are several components involved here:

-The Linux kernel
-libDRM (you must build latest libDRM version always)
-MESA (libGBM is currently part if this)
-libGLVND (needed only if you use OpenGL without X11 along with GLES)

NOTHING X11 is involved here. The fact that something works inside X11 is important and it doesn't mean anything, since I am not interested on X11 at all.

Also, you MUST use **dtoverlay=vc4-kms-v3d**
Forget about fkms or other half-inventions that partially use the old Broadcom stack,

I am writing here some fast instructions to build everything you need:

1) Remove libDRM, MESA and SDL2 from your system. I don't know what libDRM/MESA/SDL2 version does Raspberry Pi OS nor do I care because it's a heavily outdated GNU/Linux distro. These instructions can be used with any Debian based GNU/Linux distro. So if you are not using Raspberry Pi OS, it's even better.
So, just do:
`apt-get purge libdrm* libgl* libgbm* libsdl*`
Also, remove the /opt/vc directory altogether: we don't want libSDL2 to be confused with the closed, unneeded Broadcom stuff:
`sudo rm -R /opt/vc`
You may also remove everything related to X11. Again, not my business, but I would simply do:
`sudo apt-get purge libx11*`

2) For libDRM, install these dependencies first:`
sudo apt-get install xsltproc libpciaccess-dev xutils-dev libtool make automake pkg-config gcc g++ meson --no-install-recommends`
Now download the latest libDRM from:
`https://dri.freedesktop.org/libdrm/`
Create a build directory, configure it with:
`CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dudev=true -Dvc4=true -Dintel=false -Dvmwgfx=false -Dradeon=false -Damdgpu=false -Dnouveau=false -Dfreedreno=false -Dinstall-test-programs=true ..`
...build it, and install it:
```
ninja -j4
sudo ninja install
```
**NOW, BEFORE YOU GO ANY FURTHER, PLEASE TEST THAT THIS TEST IS SUCCESSFUL:**
modetest -s 32:1360x768
(Where 32 is a valid connector number and 1360x768 is a valid resolution for that connector: you can get this info from `modetest|more` once you have installed libDRM as per these instructions)

3) Install libGLVND for OpenGL support without stinky X11 or stinky X11 deps:
```
git clone --depth 1 https://gitlab.freedesktop.org/glvnd/libglvnd.git
cd  libglvnd
mkdir build
cd build
CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dglx=disabled -Dx11=disabled ..
ninja -4
sudo ninja install
```

4) Install these dependencies for building MESA:
```
sudo apt-get install --no-install-recommends flex bison python3-mako python3-setuptools libexpat1-dev libudev-dev gettext ca-certificates xz-utils zlib1g-dev pkg-config meson
```

And these for VULKAN support in MESA:`
sudo apt-get install libvulkan-dev libvulkan1 vulkan-tools --no-install-recommends`

Get the latest MESA stable version (21.1.0 as of today) from here:
https://mesa.freedesktop.org/archive/

Now configure, build and install MESA:
```
cd mesa
mkdir build
cd build
CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dglx=disabled -Dplatforms= -Dllvm=disabled -Dvulkan-drivers=broadcom -Ddri-drivers='' -Dgallium-drivers=v3d,vc4,kmsro -Dglvnd=true -Dbuildtype=release ..

ninja -j4
sudo ninja install
```
5) **VERY IMPORTANT: If you are using Raspberry Pi OS, do this before progressing:**
`sudo adduser pi render`
And now run `vulkaninfo`, and check that it's giving you the info about the Vulkan driver without any major errors.

6) Install these previous deps for SDL2:`
sudo apt-get install libudev-dev libasound2-dev libvorbis-dev libflac-dev libmpg123-dev libmodplug-dev libpng-dev libjpeg-dev libfreetype6-dev --no-install-recommends` 
Now clone the SDL2 git, configure and build it like this:
`git clone --depth 1 https://github.com/libsdl-org/SDL.git
cd SDL
CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72 -DEGL_NO_X11" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72 -DEGL_NO_X11" ./configure --enable-video-kmsdrm --disable-video-x11 --disable-dbus --disable-diskaudio --disable-oss --disable-pulseaudio --disable-video-rpi --disable-dummyaudio --disable-video-dummy --enable-video-opengles --enable-video-opengl --enable-video-vulkan --enable-libudev --disable-ime --disable-libsamplerate
make -j4
sudo make install
`
Now you can enter the "test" directory inside SDL2 sources, configure and build that, and run testgl2 and testgles2. They both should work: they do work here for sure.

Askmewho, 2021-05-09:
Hahaha yeah I am salva. Well, I am wrong at some point so I asked to close the issue. The thing is that some opengl workloads work withour X on rpi4 with my sdl2 so that seems not the problem.

I think therr is another component involved. I dont know exactly how this is handled on ubuntu 20.04 but on rk3399 I just compiled sdl2 with kms and opengl and then everything just works.

When I tested it on rpi4, for example, shovel knight, it crash and inside the log there are request to X with null results. Thats logical.. but somehow they are succesfully handled on rk3399 without X and they arent on rpi4 with rpi OS.

Running stuff without x11 would be the fastest option on rpi4 and also the easiest way to port stuff to retropie. Its just an easy script and you wre good to go. 

I leaved the channel. Its too attached to rpi4, a board I dont like that much. I preferred to leave it to my friend grayduck and created my own one. I am an AMR linux enthusiast. I need to promote something better than broadcom hardware... abd thats rockchip.

vanfanel, 2021-05-09:
@Askmewho I don't quite understand what you mean, but follow these instructions and OpenGL will work on your Pi4 without X, really.

Askmewho, 2021-05-09:
Ok, I will test it. Thanks for your detailed guide.

pelwell, 2021-05-09:
As requested.

closed 2021-05-09 by pelwell

Askmewho, 2021-05-09:
@vanfanel It does work. It can create gl context. It doesnt work on most cases for other reasons we dont know. Thats why i think there is another issue related to rpi os. But the main problem isnt kn context without x. Thats why it should be closed.

vanfanel, 2021-05-09:
@Askmewho I am not paid, don't worry, so I can use whatever I want to.
Are you saying that JDK games work outside of X11 on SDL2? :o: 
I am always awed at your experiments, really. 
Does this work for you on the Pi4 without X11? --> https://www.abadiadelcrimenextensum.com/en/the-game-2/

Askmewho, 2021-05-09:
Sure @vanfanel . So if you are that skilled you may find out whats the component that doesnt work properly, sdl2 isnt. I am not loosing my time on rpi4 for now.

Yes, on open jedy night works wothout X, there are tons of stuff thst runs without x. I just realized this recently. Its awesome bc it open many possibilities.

vanfanel, 2021-05-10:
@Askmewho I didn't ask about Jedi Knight, but I understand that you meant Open Jedi Knight when you said OpenJDK.

Askmewho, 2021-05-10:
@vanfabel yes, my mistake. A typo. about the topic. Very few stuff running on the pi without x. But of course, there is something missing on my setup. You may test for example shovel knight. It should run fine without x bit crash on my side like mostly everything.

Askmewho, 2021-05-26:
@vanfanel It's not rpi4 fault it seems. I can reproduce that for some reason, I can reproduce my success on another rk3399, for some reason, it doesnt work. I get shovel knight working on 64 bit without x, but not on 32 bit. I obviously compiled both sdl2 32 and 64 bits with kms. but for some reason, glew is crashing. I need your help, please, join the pi labs discord. If i need to run box86 games on the potato 4 at least I need to discard X.

TimPearson, 2021-05-26:
@vanfanel - apologies for my stupid question on this dead thread - i am an old programmer - out of my depth in this world a bit - anyway trying to make SDL work on a pi4 without X, for use in a production environment - is the Vulkan support an option in your instructions or a requirement? i am a bit nervous of depending on Vulkan support given their work in progress nature?

MakerMatrix, 2021-08-08:
I am currently stuck in a catch-22 because the great guide put here by @vanfanel (bookmarked!  Thank you, this is exactly what I have been searching for, for weeks!) clearly states this requires the real GL driver.  However, I am trying to drive a non-standard display with a custom HDMI timing and this appears to not work with the real GL driver (#4516).  A real pity.  I feel I have reached an impasse and have to give up.

vanfanel, 2021-08-12:
@TimPearson Sorry, I just saw your question today!

Vulkan is not a requeriment, it's an option. You can have SDL2 working without X11, and you don't need Vulkan for that AT ALL :)
Any other questions are welcome!

padremayi, 2022-04-06:
> @Askmewho It's doable on Raspberry Pi OS with some modifications. @popcornmix This one is mine, forget about this issue, I will handle it. I did the recent KMSDRM backend work on SDL2. And I can say for sure that GLES2/GL2 both work on SDL2 on a Pi3 and Pi4. I don't know what's wrong on your side, but there are several components involved here:
> 
> -The Linux kernel -libDRM (you must build latest libDRM version always) -MESA (libGBM is currently part if this) -libGLVND (needed only if you use OpenGL without X11 along with GLES)
> 
> NOTHING X11 is involved here. The fact that something works inside X11 is not important and it doesn't mean anything, since I am not interested on X11 at all (I just wish X11 is soon forgotten in favor of Wayland because X11 is old, buggy, bloated and a plague for GNU/Linux).
> 
> Also, you MUST use **dtoverlay=vc4-kms-v3d** Forget about fkms or other half-inventions that partially use the old Broadcom stack,
> 
> I am writing here some fast instructions to build everything you need:
> 
>     1. Remove libDRM, MESA and SDL2 from your system. I don't know what libDRM/MESA/SDL2 version does Raspberry Pi OS nor do I care because it's a heavily outdated GNU/Linux distro (they don't ship an SDL2 version capable of X11-less rendering, even if this feature has been there since 2016. Surreal). These instructions can be used with any Debian based GNU/Linux distro. So if you are not using Raspberry Pi OS, it's even better.
>        So, just do:
>        `apt-get purge libdrm* libgl* libgbm* libsdl*`
>        Also, remove the /opt/vc directory altogether: we don't want libSDL2 to be confused with the closed, unneeded Broadcom stuff:
>        `sudo rm -R /opt/vc`
>        You may also remove everything related to X11. Again, not my business, but I would simply do:
>        `sudo apt-get purge libx11*`
> 
>     2. For libDRM, install these dependencies first:` sudo apt-get install xsltproc libpciaccess-dev xutils-dev libtool make automake pkg-config gcc g++ meson --no-install-recommends`
>        Now download the latest libDRM from:
>        `https://dri.freedesktop.org/libdrm/`
>        Create a build directory, configure it with:
>        `CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dudev=true -Dvc4=true -Dintel=false -Dvmwgfx=false -Dradeon=false -Damdgpu=false -Dnouveau=false -Dfreedreno=false -Dinstall-test-programs=true ..`
>        ...build it, and install it:
> 
> 
> ```
> ninja -j4
> sudo ninja install
> ```
> 
> **NOW, BEFORE YOU GO ANY FURTHER, PLEASE TEST THAT THIS TEST IS SUCCESSFUL:** `modetest -s 32:1360x768` (Where 32 is a valid connector number and 1360x768 is a valid resolution for that connector: you can get this info from `modetest|more` once you have installed libDRM as per these instructions)
> 
>     3. Install libGLVND for OpenGL support without stinky X11 or stinky X11 deps:
> 
> 
> ```
> git clone --depth 1 https://gitlab.freedesktop.org/glvnd/libglvnd.git
> cd  libglvnd
> mkdir build
> cd build
> CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dglx=disabled -Dx11=disabled ..
> ninja -4
> sudo ninja install
> ```
> 
>     4. Install these dependencies for building MESA:
> 
> 
> ```
> sudo apt-get install --no-install-recommends flex bison python3-mako python3-setuptools libexpat1-dev libudev-dev gettext ca-certificates xz-utils zlib1g-dev pkg-config meson
> ```
> 
> And these for VULKAN support in MESA:` sudo apt-get install libvulkan-dev libvulkan1 vulkan-tools --no-install-recommends`
> 
> Get the latest MESA stable version (21.1.0 as of today) from here: https://mesa.freedesktop.org/archive/
> 
> Now configure, build and install MESA:
> 
> ```
> cd mesa
> mkdir build
> cd build
> 
> CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72" meson -Dglx=disabled -Dplatforms= -Dllvm=disabled -Dvulkan-drivers=broadcom -Ddri-drivers='' -Dgallium-drivers=v3d,vc4,kmsro -Dglvnd=true -Dbuildtype=release ..
> 
> ninja -j4
> sudo ninja install
> ```
> 
[... 4 lines cut ...]
>     6. Install these previous deps for SDL2:` sudo apt-get install libudev-dev libasound2-dev libvorbis-dev libflac-dev libmpg123-dev libmodplug-dev libpng-dev libjpeg-dev libfreetype6-dev --no-install-recommends`
>        Now clone the SDL2 git, configure and build it like this:
> 
> 
> ```
> git clone --depth 1 https://github.com/libsdl-org/SDL.git
> 
> cd SDL
> 
> CFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72 -DEGL_NO_X11" CXXFLAGS="-O2 -march=armv8-a+crc+simd -mtune=cortex-a72 -DEGL_NO_X11" ./configure --enable-video-kmsdrm --disable-video-x11 --disable-dbus --disable-diskaudio --disable-oss --disable-pulseaudio --disable-video-rpi --disable-dummyaudio --disable-video-dummy --enable-video-opengles --enable-video-opengl --enable-video-vulkan --enable-libudev --disable-ime --disable-libsamplerate
> 
> 
> make -j4
> sudo make install
> ```
> 
> Now you can enter the "test" directory inside SDL2 sources, configure and build that, and run testgl2 and testgles2. They both should work: they do work here for sure.
> 
> PD: Are you Salva from PiLabs? If so, we have met on youtube already! Good job on all those experiments you do to run games on the Pi4! :)
> 
> PD2: In a nutshell, having OpenGL (NOT GLES) on SDL2 without stinky X11 involves libGLVND and rebuilding MESA with support for that, and then SDL2 with support for that again (libGLVND produces libOpenGL.so instead of libGL.so: but SDL2 already handles that.. had a hard time figuring out all this crazy stuff!)
> 
> PD3: You were very lucky that I read this here. This is far from the right place for this problem, I guess an issue on SDL2 github would be the right place. But well, you are in good hands now dancing_men

Followed your instuctions and all my SDL2 application can't initialize video driver