raspberrypi/linux #4332: Jack connector audio on Pi4 with vc4hdmi ALSA device

vanfanel, 2021-05-09:
Hi,

Using Pi4 with the KMS overlay:
`dtoverlay=vc4-kms-v3d`

..and the vc4hdmi ALSA device (ie: NO bcm2835 audio module loaded):
`#dtparam=audio=on` <---Commented out line.

I have very good audio (and low-latency capable) via the vc4hdmi ALSA device.

So, my question is very simple:
**Is it possible to have JACK connector audio without the BCM2835 module? In other words, is it possible to have JACK connector audio using the arm-side audio stack, be it via vc4hdmi or with another method?**
If so, how?

Thanks!

pelwell, 2021-05-09:
The short answer is no - you need the bcm2835 audio module in order to send the streams to the VPU. One _could_ drive the PWM output using a custom driver running on the ARMs, but doing so would lose out on the SDM magic the firmware performs, and you really don't want that.

vanfanel, 2021-05-09:
@pelwell Ah, thanks. Clear and good response, just what I needed :)

closed 2021-05-09 by vanfanel