raspberrypi/linux #4338: Pi4 4GB drops USB

mhzawadi, 2021-05-12:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
when downloading from the pi, the usb seems to get reset

**To reproduce**
1. make a 3GB file
2. download that file over ssh or other method (rclone seems to kill it)

**Expected behaviour**
the file download would complete, with the USB disk dropping off

**Actual behaviour**
the download would error and all the USB devices would drop off, none are listed with lsusb and all the disks are gone from lsblk

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:
-bash: raspinfo: command not found

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
  *  Pi4 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
  * Raspberry Pi reference 2020-02-13
  * Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
* Which firmware version (`vcgencmd version`)? 
  * version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start_cd)
* Which kernel version (`uname -a`)? 
  * Linux pi-file3 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
dmesg:
```
[   51.312753] Bluetooth: HCI device and connection manager initialized
[   51.312768] Bluetooth: HCI socket layer initialized
[   51.312779] Bluetooth: L2CAP socket layer initialized
[   51.312795] Bluetooth: SCO socket layer initialized
[   51.324151] Bluetooth: HCI UART driver ver 2.3
[   51.324162] Bluetooth: HCI UART protocol H4 registered
[   51.324215] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   51.324361] Bluetooth: HCI UART protocol Broadcom registered
[   51.516730] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   51.516743] Bluetooth: BNEP filters: protocol multicast
[   51.516760] Bluetooth: BNEP socket layer initialized
[74184.926993] sd 0:0:0:0: [sda] tag#22 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=11s
[74184.927036] sd 0:0:0:0: [sda] tag#22 Sense Key : 0xb [current]
[74184.927071] sd 0:0:0:0: [sda] tag#22 ASC=0x4b ASCQ=0x0
[74184.927105] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x28 28 00 6e 7c e9 78 00 04 00 00
[74184.927138] blk_update_request: I/O error, dev sda, sector 1853679992 op 0x0:(READ) flags 0x84700 phys_seg 128 prio class 0
[74203.775817] sd 0:0:0:0: [sda] tag#26 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD IN
[74203.775838] sd 0:0:0:0: [sda] tag#26 CDB: opcode=0x28 28 00 6f cd 60 90 00 04 00 00
[74203.780271] sd 0:0:0:0: [sda] tag#20 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD IN
[74203.780291] sd 0:0:0:0: [sda] tag#20 CDB: opcode=0x28 28 00 6f cd 40 c8 00 04 00 00
[74203.781753] sd 0:0:0:0: [sda] tag#18 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD IN
[74203.781771] sd 0:0:0:0: [sda] tag#18 CDB: opcode=0x28 28 00 6f 22 ec 00 00 04 00 00
[74203.782656] sd 0:0:0:0: [sda] tag#17 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD IN
[74203.782672] sd 0:0:0:0: [sda] tag#17 CDB: opcode=0x28 28 00 6f 22 e8 00 00 04 00 00
[74203.783557] sd 0:0:0:0: [sda] tag#16 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD IN
```
syslog:
```
May 12 02:30:45 pi-file3 rngd[648]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
May 12 02:40:01 pi-file3 CRON[18491]: (root) CMD (/root/bin/backblaze_user_sync)
May 12 02:58:27 pi-file3 smartd[480]: Device: /dev/sdc [USB JMicron], SMART Usage Attribute: 190 Airflow_Temperature_Cel changed from 66 to 74
May 12 03:00:01 pi-file3 CRON[18716]: (root) CMD (/root/bin/backblaze_user_sync)
May 12 03:01:01 pi-file3 CRON[18800]: (root) CMD (/root/bin/backblaze_backup)
May 12 03:06:22 pi-file3 kernel: [74184.926993] sd 0:0:0:0: [sda] tag#22 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=11s
May 12 03:06:22 pi-file3 kernel: [74184.927036] sd 0:0:0:0: [sda] tag#22 Sense Key : 0xb [current]
May 12 03:06:22 pi-file3 kernel: [74184.927071] sd 0:0:0:0: [sda] tag#22 ASC=0x4b ASCQ=0x0
May 12 03:06:22 pi-file3 kernel: [74184.927105] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x28 28 00 6e 7c e9 78 00 04 00 00
May 12 03:06:22 pi-file3 kernel: [74184.927138] blk_update_request: I/O error, dev sda, sector 1853679992 op 0x0:(READ) flags 0x84700 phys_seg 128 prio class 0
```
other log files you may need - https://owncloud.horwood.biz/s/dL5T93YwkdxtPdf

**Additional context**
at about 0205 I have backups from other pis start, they seem to upload to this pi fine.
at about 0305 I have an rclone sync run to Backblaze, syncs about 222GB

P33M, 2021-05-12:
Please follow the steps in this thread : https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

mhzawadi, 2021-05-12:
I have made the change suggested, will see in the morning if that stops my issue.

mhzawadi, 2021-05-13:
that worked, didnt need to logon at 06:30 this morning to fix my file server.

Could have done with a notice that the USB storage driver was going to change

closed 2021-05-13 by mhzawadi