raspberrypi/linux #4339: kms/RPi4: 4096x2160-30 output broken since clock request changes

HiassofT, 2021-05-12:
**Describe the bug**
Trying to use 4096x2160-30 on RPi4 with kms and latest rpi-update kernel results in no signal output and "flip_done timed out" messages in dmesg unless `force_turbo=1` is used in config.txt

This mode worked fine until rpi-update version https://github.com/Hexxeh/rpi-firmware/commit/5fa7013a2cce73e03d400514e10f0893a3eee824

**To reproduce**
run `modetest -s 32:4096x2160-30`` on a RPi4 with plain config.txt and only vc4.-kms-v3d overlay added

**Expected behaviour**
Video output works, eg modetest shows test pattern.

**Actual behaviour**
No signal on TV and modetest hangs.

**System**
RPi4 4GB running RPiOS with all updates and latest rpi-update kernel https://github.com/Hexxeh/rpi-firmware/commit/c1737c7a3bd5c0e401c3303ba454d9b9663a2765

Also LibreELEC master with kernel 5.10.33 (and newer) is affected - see https://github.com/LibreELEC/LibreELEC.tv/pull/5348#issuecomment-830673202

**Additional context**
Checking through kernel commits showed that 8b664487bde45b137c6511fc91afaf993b30016c is the first bad version, 9e081a7bf69c75d5e9cb05add07eaf4cb4ee4dc9 is fine.

Comparing clk_summary taken while modetest was running/hanging showed that in the non-working state the core clock was running at 200 MHz while in the working state it was at 500MHz
```
root@raspberrypi:~# diff -u clk-err clk-ok 
--- clk-err	2021-05-12 15:06:09.735748650 +0100
+++ clk-ok	2021-05-12 15:08:28.068190366 +0100
@@ -5,7 +5,7 @@
  fw-clk-m2mc                          1        1        0   299970000          0     0  50000
  fw-clk-hevc                          0        0        0   250000000          0     0  50000
  fw-clk-v3d                           0        0        0   250000000          0     0  50000
- fw-clk-core                          1        1        0   200000000          0     0  50000
+ fw-clk-core                          1        1        0   500000000          0     0  50000
  fw-clk-arm                           0        0        0   600000000          0     0  50000
  108MHz-clock                         0        0        0   108000000          0     0  50000
     hdmi1-108MHz                      0        0        0   108000000          0     0  50000
```

ping @popcornmix

popcornmix, 2021-05-12:
So, the firmware has two boost requests for core frequency that relate to hdmi.
It requests max core frequency at start of the mode set, that is removed when it completes. 
And it requests a frequency roughly proportional to the pixel clock (but it is somewhat complicated) that persists until the next mode change.

I believe the vc4 driver only does the former. And before the "clock request changes" commit, a bug meant the former boost was never removed.

The correct solution is to add the latter request.
FYI @mripard (I'll have to explain the firmware side logic when I understand it myself).

user13571113, 2021-05-12:
The problem description and the fact that it emerged after the upgrade to kernel 5.10.31 leads me to suspect that this problem is the same encountered by me: https://github.com/raspberrypi/linux/issues/4317

In fact, I can confirm that adding `force_turbo=1` to /boot/config.txt results in a working display for me, wheres the monitor reports "No HDMI input signal" without the option set.

AlexanderStein, 2021-05-25:
I hit this problem too. Bisected to [`8b66448`](https://github.com/raspberrypi/linux/commit/8b664487bde45b137c6511fc91afaf993b30016c) as well.

My use case is playing a 1080p Youtube video in kodi. After the offending commit the video desynchronizes with the audio and gets reset after a while, while the desynchro starts again. Local playback of mp4 files is not affected though.

As mentioned before `force_turbo=1` to /boot/config.txt also works here.

mripard, 2021-05-26:
https://github.com/raspberrypi/linux/pull/4365 should fix this issue. Can you test and report if it works?

Thanks!

AlexanderStein, 2021-05-27:
I applied both the patches to commit c2f585ae894081445b7f1ab8d0b6334061be9965, but unfortunately they don't fix my problem on youtube playback. The `fw-clk-core` is set to 500MHz, but using `grep fw-clk-core /sys/kernel/debug/clk/clk_summary`. I noticed that it falls back to 200MHz once in a while, about 2-3 every 10s using watch. Not very accurate but the clock switch can be detected this way. Dunno what the effect actually is.

There is also another difference in the clocks which might explain my problem with the video decoding:
```diff
--- clk_good    2021-05-27 18:41:46.964726722 +0200
+++ clk_fixed   2021-05-27 18:45:57.006687269 +0200
@@ -35,8 +35,8 @@
        plla_ccp2                      0        0        0    11718750          0     0  50000
        plla_dsi0                      0        0        0    11718750          0     0  50000
        plla_core                      2        2        0   499999998          0     0  50000
-          h264                        0        0        0   444444442          0     0  50000
-          isp                         0        0        0   444444442          0     0  50000
+          h264                        0        0        0   415584413          0     0  50000
+          isp                         0        0        0   415584413          0     0  50000
           vpu                         3        3        0   500000000          0     0  50000
              aux_spi2                 0        0        0   500000000          0     0  50000
              aux_spi1                 0        0        0   500000000          0     0  50000
```

mripard, 2021-06-04:
I updated the PR with a few fixes, can you test this new version?

AlexanderStein, 2021-06-05:
Thanks for the update. Unfortunately none of my issues has changed. The fw-clk-core still alternates bewteen 20 and 500 MHz during playback. My youtube playback desyncs on video and audio. And the h264/isp clocks still have a lower frequency than before.

mripard, 2021-06-07:
It looks like a separate issue than the one that started this bug report. Can you open a new one? Thanks.

user13571113, 2021-06-12:
After upgrading to kernel 5.10.42 (pertaining to commit https://github.com/raspberrypi/linux/commit/da6937fd9c7416745a2bff9bb81a8804d4907514), [my issue](https://github.com/raspberrypi/linux/issues/4317) still persists: Without `force_turbo=1` in /boot/config.txt, my monitor reports "No HDMI input signal" when starting X; it works with `force_turbo=1`.

mripard, 2021-06-16:
Did you test with #4365 ?

popcornmix, 2021-06-16:
#4365 will be present in today's rpi-update firmwware.

user13571113, 2021-06-20:
The issue is resolved (X launches without `force_turbo=1`) with kernel 5.10.44 (commit 2b13c54592135b6fab269517ed687fa9f80bf8e5) and EEPROM

```
BOOTLOADER: up to date
   CURRENT: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
    LATEST: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
   RELEASE: critical (/lib/firmware/raspberrypi/bootloader/critical)

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

Thank you very much for these updates.

popcornmix, 2021-06-23:
@HiassofT is this okay to close?

closed 2021-06-23 by HiassofT