raspberrypi/linux #4341: Auto negotiation bug on RasPi 3b+

shawaj, 2021-05-12:
**Describe the bug**
There seems to be a bug on raspberry pi 3b+ with lan9515 chips where auto negotiation on the Ethernet fails to work after some random period.

**To reproduce**
We are using the raspberry pi 3b+ as a test rig for Raspberry pi cm3+ based boards. We plug it in via Ethernet connection. Cm3 carrier has a maxlinear/exar xr22800 Eth chip.

One test rig will test a device, device unplugged, new one plugged in and tested... And then repeat.

This will work for the first 1 or 2 times (sometimes more, it seems random) and then eventually will fail to work anymore.

Putting an Ethernet switch between the two and it works flawlessly.

This is repeatable on 10 test rigs we have set up.

**Expected behaviour**
Auto negotiation should work at all times

**Actual behaviour**
Auto negotiation fails

**System**
 
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Pi3b+

* Which OS and version (`cat /etc/rpi-issue`)?

Latest raspios

* Which firmware version (`vcgencmd version`)?

Latest
* Which kernel version (`uname -a`)?

Latest

Copying in @ryanteck as well as he may know more about the versions and setup.

6by9, 2021-05-12:
Pi's up to 3B have LAN9514 (LAN9512 on the original Model B).
Pi3B+ has LAN7800 packaged with a hub chip as LAN7515.

AFAIK LAN9515 does not exist. Please confirm you mean 3B+ and LAN7515.

When asked for versions, please provide actual version numbers. "Latest" could be latest rpi-update kernel/firmware, or that packaged through apt.

shawaj, 2021-05-12:
Sorry that was a typo! Should have said 7515!

@ryanteck can you provide the version info from the test image?