raspberrypi/linux #4345: Raspbian 32 bit distro binaries contain illegal instructions on Raspberry Pi Zero W causing crashes

dhdurgee, 2021-05-13:
Here is the output from my pi Zero W:

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
pi@raspberrypi:~ $ 

I encountered a problem with vlc as provided with the distro.  When I asked for help from vlc support they had me run it with gdb.  This yielded:

(gdb) bt
#0  0xa22e9004 in ?? () from /usr/lib/arm-linux-gnueabihf/vlc/plugins/mmal/libmmal_codec_plugin.so
#1  0xa22d9314 in ?? () from /usr/lib/arm-linux-gnueabihf/vlc/plugins/mmal/libmmal_codec_plugin.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

and

   ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
  >│0xa22e9004      vdup.8 d7, r2                                                                                                   │
   │0xa22e9008      subs   r3, r3, #8                                                                                               │
   │0xa22e900c      vmov.i8        d6, #255        ; 0xff                                                                           │
   │0xa22e9010      blt    0xa22e9098                                                                                               │
   │0xa22e9014      cmp    r3, #8                                                                                                   │
   │0xa22e9018      mov    r4, r3                                                                                                   │
   │0xa22e901c      ble    0xa22e9034                                                                                               │
   │0xa22e9020      rsb    r3, r0, #0                                                                                               │
   │0xa22e9024                     ; <UNDEFINED> instruction: 0xe7e23153                                                            │
   │0xa22e9028      cmp    r3, #0                                                                                                   │
   │0xa22e902c      blne   0xa22e90a4                                                                                               │
   │0xa22e9030      sub    r3, r4, r3                                                                                               │
   │0xa22e9034      vld4.8 {d16-d19}, [r1]                                                                                          │
   │0xa22e9038      vmull.u8       q15, d19, d7                                                                                     │
   │0xa22e903c      vld4.8 {d20-d23}, [r0]                                                                                          │
   │0xa22e9040      vsra.u16       q15, q15, #8                                                                                     │
   │0xa22e9044      subs   r3, r3, #8                                                                                               │
   │0xa22e9048      vrshrn.i16     d31, q15, #8                                                                                     │
   │0xa22e904c      vsub.i8        d30, d6, d31                                                                                     │
   │0xa22e9050      vmull.u8       q12, d16, d31                                                                                    │
   │0xa22e9054      vmull.u8       q13, d17, d31                                                                                    │
   │0xa22e9058      vmull.u8       q14, d18, d31                                                                                    │
   │0xa22e905c      addge  r1, r1, #32                                                                                              │
   └────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
multi-thre Thread 0xa1e0e3e0 ( In:                                                                              L??   PC: 0xa22e9004 
(gdb) 

The vlc debug log shows:

[9bbdadd0] main blend debug: looking for video blending module matching "any": 3 candidates
[9bbdadd0] mmal_codec blend debug: OpenBlendNeon: (blend) RGBA/RGBA,273x15 [(0,0) 273x15]->RV32/RGBA,288x192 [(0,0) 288x192]
[9bbdadd0] main blend debug: using video blending module "mmal_codec"

aptcache showpkg vlc yields:

Package: vlc
Versions: 
3.0.12-0+deb10u1+rpt1 (/var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_main_binary-armhf_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_main_binary-armhf_Packages
                  MD5: d96da6b991aa50928533b16fb11b3136
 Description Language: 
                 File: /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_main_binary-armhf_Packages
                  MD5: d96da6b991aa50928533b16fb11b3136

3.0.12-0+deb10u1 (/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_main_binary-armhf_Packages)
 Description Language: 
                 File: /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_main_binary-armhf_Packages
                  MD5: d96da6b991aa50928533b16fb11b3136
 Description Language: 
                 File: /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_main_binary-armhf_Packages
                  MD5: d96da6b991aa50928533b16fb11b3136

Processor details are:

pi@raspberrypi:~ $ cat  /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000005bae4c7f
Model		: Raspberry Pi Zero W Rev 1.1
pi@raspberrypi:~ $

I am told the problem is that the binaries are being built including neon instructions, which are not available on my CPU.

I have opened a thread on the troubleshooting forums as well:

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=307312

6by9, 2021-05-13:
Duplicate of https://github.com/RPi-Distro/vlc/issues/42 by the looks of it.

dhdurgee, 2021-05-13:
True. I decided to post here as well after I realized that vlc is likely not the only place that neon instructions would show up and crash something else on the Pi Zero and others with this CPU. My nightmare would be if they are hiding somewhere in the kernel waiting to trigger a panic.

dhdurgee, 2021-05-13:
When you locked the thread on the troubleshooting forum you said "If you have an demonstrable bug, then please raise an issue on the appropriate Github repo. Forum threads are not always seen by the appropriate people."  Please let me know what information beyond what I have provided here and in the RPi_Distro forum you need and where you would like me to provide it.

Given how I encountered this bug I would expect that all that would be necessary to duplicate it would be to install the current Raspbian Buster on a Pi with the same CPU as mine, install the current vlc, download an mp3/cdg Karaoke track and attempt to play it.  I will be happy to provide the one I was encountering the problem with if you cannot duplicate it with another track, subject to the understanding the material is copyrighted by the publisher and cannot be released and will only be used to duplicate the problem.

As the problem is with neon instructions being built into binaries to be used on units without those instructions I would think there would be little need for duplication, as there will be a crash as soon as the CPU encounters the instruction.

pelwell, 2021-05-14:
https://github.com/RPi-Distro/vlc is a better home for this report.

closed 2021-05-14 by pelwell