raspberrypi/linux #4347: Kernel 1:1.20210430-2 not tagged/released

Ve2mrx, 2021-05-15:
**Describe the bug**
The kernel 1:1.20210430-2 isn't tagged or released on Github and isn't downloadable/compilable

**To reproduce**
N/A

**Expected behaviour**
Have the kernel release equivalent to raspberrypi-kernel 1:1.20210430-2 to be tagged or in the release section

**Actual behaviour**
Missing from the release and Tag sections of Github

**System**
N/A

**Logs**
N/A

**Additional context**
N/A

popcornmix, 2021-05-16:
Possibly duplicate of https://github.com/raspberrypi/linux/issues/4316

XECDesign, 2021-05-16:
That tag really doesn't exist, since git wouldn't allow it (I'm guessing the colon character is invalid). I am not able to push anything right now and I'm not sure what the alternative name should be.

Technically, only the "upstream" part of the Debian package version is relevant, so maybe it should just be that (1.20210430).

closed 2021-08-07 by Ve2mrx