raspberrypi/linux #4349: Xbox One DVB-T/T2/C tuner -  device was not an MTP device

zangaby, 2021-05-15:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Unable to use Xbox One DVB-T/T2/C tuner 

**To reproduce**
Plug an Xbox tuner

**Expected behaviour**
Load the firmware and use with tvheadend

**Actual behaviour**
Tuner is now being properly loaded.

**System**
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage2

Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
Revision        : a02082
Serial          : 000000008f28749b
Model           : Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Apr 30 2021 13:47:07
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       15020844 1887604  12490288  14% /
devtmpfs          439916       0    439916   0% /dev
tmpfs             473196       0    473196   0% /dev/shm
tmpfs             473196    6336    466860   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             473196       0    473196   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48783    209313  19% /boot
log2ram            40960   12900     28060  32% /var/log
tmpfs              94636       0     94636   0% /run/user/1000

Filename                                Type            Size            Used            Priority
/var/swap                               file            102396          0               -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
  Installed: (none)
[... 620 lines cut ...]
May 15 13:47:23 raspberrypi kernel: [  393.673060] usb 1-1.5: Product: Xbox USB Tuner
May 15 13:47:23 raspberrypi kernel: [  393.673075] usb 1-1.5: Manufacturer: Microsoft Corp.
May 15 13:47:23 raspberrypi kernel: [  393.673090] usb 1-1.5: SerialNumber: 006327240115
May 15 13:47:23 raspberrypi kernel: [  393.676843] dvb-usb: found a 'Microsoft Xbox One Digital TV Tuner' in cold state, will try to load a firmware
May 15 13:47:23 raspberrypi kernel: [  393.677094] dvb-usb: downloading firmware from file 'dvb-usb-dib0700-1.20.fw'
May 15 13:47:23 raspberrypi kernel: [  393.757284] dib0700: firmware started successfully.
May 15 13:47:24 raspberrypi kernel: [  394.272374] dvb-usb: found a 'Microsoft Xbox One Digital TV Tuner' in warm state.
May 15 13:47:24 raspberrypi kernel: [  394.272813] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
May 15 13:47:24 raspberrypi kernel: [  394.273427] dvbdev: DVB: registering new adapter (Microsoft Xbox One Digital TV Tuner)
May 15 13:47:24 raspberrypi kernel: [  394.273444] usb 1-1.5: media controller created
May 15 13:47:24 raspberrypi kernel: [  394.274931] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
May 15 13:47:24 raspberrypi kernel: [  394.540081] mn88472 11-0018: Panasonic MN88472 successfully identified
May 15 13:47:24 raspberrypi kernel: [  394.556298] tda18250 11-0060: NXP TDA18250BHN/M successfully identified
May 15 13:47:24 raspberrypi kernel: [  394.558843] usb 1-1.5: DVB: registering adapter 0 frontend 0 (Panasonic MN88472)...
May 15 13:47:24 raspberrypi kernel: [  394.558880] dvbdev: dvb_create_media_entity: media entity 'Panasonic MN88472' registered.
May 15 13:47:24 raspberrypi kernel: [  394.561879] dvb-usb: Microsoft Xbox One Digital TV Tuner successfully initialized and connected.
May 15 13:47:24 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
May 15 13:47:24 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
May 15 13:47:24 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
May 15 13:47:24 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
May 15 13:47:24 raspberrypi kernel: [  394.677216] mn88472 11-0018: downloading firmware from file 'dvb-demod-mn88472-02.fw'
May 15 13:47:24 raspberrypi kernel: [  394.844225] usb 1-1.5: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)

**Additional context**
Add any other relevant context for the problem.

6by9, 2021-05-15:
> **Actual behaviour**
Tuner is now being properly loaded.

So it's being loaded fine and works, but you get the error line?

MTP is Media Transfer Protocol which is generally only used for transferring files to/from media players or mobile phones. I'd suspect that the device has a small setup partition, but that's irrelevant on a Linux system that has drivers already present.

zangaby, 2021-05-15:
Thank you for the clarification.

Indeed you are right, I was mislead by the mtp message, the tuner works fine.
In order to enable it I had to create a network for it in the tvheadend web gui.

closed 2021-05-15 by zangaby