raspberrypi/linux #4350: glmark2-drm fails, glmark2-es2-drm passes

jcdutton, 2021-05-15:
**Describe the bug**
From the Raspberry 4 command line (without X)
glmark2-es2-drm  works. It completes all the tests.
glmark2-drm fails with a segfault after a few tests.

Compiled RPI-5.12.x kernel
Mesa 20.3

Add a clear and concise description of what you think the bug is.
I think their might be a bug in Mesa or the kernel causing this problem.

**To reproduce**
glmark2-es2-drm  works. It completes all the tests.
glmark2-drm fails with a segfault after a few tests.

**Expected behaviour**
Both glmark2-es2-drm and glmark2-drm should complete all tests.

**Actual behaviour**
glmark2-es2-drm  works. It completes all the tests.
glmark2-drm fails with a segfault after a few tests.

**System**
Compiled RPI-5.12.x kernel
Mesa 20.3
Raspberry CM4 on CM4 IO board.

popcornmix, 2021-05-16:
@naushir is this something for Igalia?

schnitzeltony, 2021-05-17:
I think I asked similar at mesa a while back - maybe last post in [1] helps.

[1] https://gitlab.freedesktop.org/mesa/mesa/-/issues/2626

naushir, 2021-05-17:
@itoral could you provide some guidance on this issue please?

itoral, 2021-05-18:
> I think I asked similar at mesa a while back - maybe last post in [1] helps.
> 
> [1] https://gitlab.freedesktop.org/mesa/mesa/-/issues/2626

Exactly, the issue is explained there by Neil, I quote his analysis here:

"It crashes because glmark2 tries to call glGenerateMipmap. It is using a library called glad which loads the GL function pointers into global variables. However, glGenerateMipmap is a GL3.0 feature and glad doesn’t try to get the pointer because v3d only advertises GL2.1.

If I run glmark2 with MESA_GL_VERSION_OVERRIDE=3.0 then the suite runs successfully. I guess that means we actually do support enough features for it to run. So I guess we can consider the crash a bug with glmark2 because it assumes GL3.0 without checking for it and doesn’t fall back to checking for extensions to provide the features it needs."

So basically, the problem is that the non-ES version requires GL 3.0, which we don't support.